SignalK/specification #48: Multiple sources for the same value

rob42, 2015-01-07:
It is quite possible for a key value to come from more than one device. eg position (lat/lon) could come from several gps enabled devices, and multiple depth sounders are not uncommon. We need a consistent way to handle this.

All the incoming values may well be valid in their own context, and it is feasible that all of them may be wanted, for instance, dispalying depth under each hull on a catamaran.

Hence discarding or averaging is not a solution, and since the signalk is unable to derive the best way to handle multiple values it must always fall to a default action, with human over-ride when needed. 

I propose we should simply store all the options in the tree, and have the main 'source' reference the options?

Then simple rules can apply:
1) If its the first value for the key, it becomes the first option and the default value.
2) If another value with different source arrives, we add to options - if its our preferred source (from persistent config) we auto-switch to it, otherwise we just record it
3) Users can then view the source options for a key and select from list for a specific display. They might select courseOverGroundTrue.options.ttyUSB1 as the source for a specific need or they may select an average/mean/min/max etc. (This facility is not part of the spec, but implementation specific)
4) maybe needs a rule to trim unreferenced options (or all options) on output.

{"vessels":
    {"self":
        {"navigation":
            {"courseOverGroundTrue":
                {
                 "value": 102.29,
                 "source": "options.actisense",
                 "options":
                    {"ttyUSB1":  **just a unique name, have to generate somehow from incoming
                        {  
                            "value": 99.2900009155,
                            "source": "/dev/ttyUSB1"
                            "timestamp": "2014-08-15-16:00:00.081",
                        },
                     "actisense":
                        {
                            "value": 102.29,
                            "source": "/dev/actisense",

                            "timestamp": "2014-08-15-16:00:00.081",
                            "src": "115",
                            "pgn": "128267"
                        }
                    }
                }
            }
        }
    }
}

tkurki, 2015-01-08:
Thanks @rob42  for kicking this off.

I'll pitch in with my TDD hat on and commit a test for the new format. I modified your example a little:
* pgn was wrong
* added complexity: two nmea0183 sources with different sentences and two n2k sources on one bus with different pgns

I think that the way the identifer is constructed should be specified:
* n2k: producerid-pgn-sourceid (producer id from server configuration, others from n2k data)
* nmea0183: producerid-talkerid-sentence (like n2k)

BTW we have neglected to include nmea0183 talker id, it should be added to the schema as well.

The next step would be to fix the schema so that the test passes for the sample. You can run the test after `npm install` with `npm test`.

PS. The test code assumes that the input is without the `vessels.123456789` preamble, that ought to be fixed as well.

tkurki, 2015-01-08:
I created added Travis configuration for this and opened pull request #49 for this so now we have CI integration as well.

rob42, 2015-01-08:
Also need the extra rule to make it clear that we have options object:
4) If there is an 'options' object, then source = options.name
Or we could always have 'options', but that would be verbose for most cases...

tkurki, 2015-01-08:
Where should the additional documentation like rules 1-4 go?

rob42, 2015-01-09:
I would put it as a comment in relevant code, and I will write up a page for the web site under 'developers'. 
Wonder if I can add it to a readme.md in the specification repo?

tkurki, 2015-01-09:
The test I mentioned earlier: https://github.com/SignalK/specification/blob/multiple-values/test/multiple_values.js

pod909, 2015-01-25:
The prefered source of data probably belongs in a manifest (seperate structure defining the resources available from a vessle). Such a manifest would list the readings available from a vessel, a list of devices and contain the preferred mapping between them. If a date stamp was also included that would remove the need to litter the basic data with souce and timing data

By declairing the devices (with readings available) in a manifest and specifying a suitable URI structure for SignalK you would give the user the choice to query the individual device for data using the existing SignalK schema with out the need to add additional "option" tags.

tkurki, 2015-01-25:
We have previously discussed the need for a _mapping layer_ in case there is need for server side processing of for example derived values. This would be part of the server configuration.

But I think that the simplest case should be supported with minimun a priori knowledge, eg. the server should just need minimum configuration to connect the producers on board (n2k bus, nmea sources, other sources) and then be able to pass that data on.

With the current implementations there is a de facto registry of all available data in the REST api contents.

There are some ideas about this  type of metadata in https://github.com/SignalK/specification/blob/master/schemas/groups/sensors.json.

About timestamp: I think it belongs right next to the value as it is metadata about that particular value. It is also part of the streaming data: 'new value is x with timestamp y'.

tkurki, 2015-01-25:
More stuff available at SignalK/signalk.github.io#17

pod909, 2015-01-25:
Appologies if this is going over old ground. I suspect I'm coming at this from a different angle..

Because this moves SignalK towards being essentially just a super transport of other messaging a user is going to need considerable prior knowlage of the exact configuration of the vessel -- including knowlage of whcih components are reliably calibrated and the way it's going the structure of the none SignalK messaging its self -- to use SignalK sucessfully.

Coming to the vessel from the outside with none of that knowlage a menu of possable readings for each value is going to be confusing at best. So either some work is done on the 'server' to pick the right source or there's a need for a manifest to tell me which is the best reading to use from the range provided.

Achitecture wise it's also pushing SignalK towards a particular and more expensive implementation model (all the work done at the end applicaiton) that doesn't allow for work to be distributed over the network. Where as a good scheme, or one that it's hoped will have a wide take up, should leave these architecture questions open. Otherwise there's a danger SignalK is going to end up tied to the implementation found here and go no further.

My interest is in using SignalK end-2-end from sensor to display via processing. Then from the boat to a nework of boats and to the internet. With a large number of derived values being generated in the chain, some of which go back in the opposit dirrection. With the right SignalK schema I'd be quite happy to translate to SignalK at the device where it's not implemented natively.

I'd like to see a single data structure to represent all the data available from a single location-device i.e. a single data structure that can be used to pass data from a device to intra location processing AND once processed out to the outside world. As start point the sensor data object should have the same structure as the vessel object, rather than the vessel object (which may not be my frame of reference for makign a query) containing all sensor values.

Intelegent multiplexing (with an explaination of how it was done and the alternative sources that can be seperatly RESTed if wanted) of the data enables me to expose my knowlage of the local system set-up in the data. The way you're taking this the use of SignalK from the device is prety unatractive and if I have to pay for N2K from device to multiplexer then I might as well stick with it e2e.

Time stamp

For me the timestamp should be abstracted to the sensor object with all the values coming from that sensior referencing the sensor (which I'd do in the manifest as the mapping is prety static). That way all the data can be left in the vessel object rather than being seperated out into different sensor payloads that I have to process back together again.

Timestamping for delta data belongs in the packet header no?

canboat, 2015-01-25:
@pod909, I can't really make out what you are saying. On an abstract level, and on a detailed technical level. Maybe it's just me though.

tkurki, 2015-01-25:
@pod909 Maybe a real life practical example would help clarify your viewpoint? 

As for "a menu of possable readings for each value is going to be confusing at best": on my boat the menu would contain items like 
- navigation.position
- navigation.courseOverGroundTrue
- navigation.speedThroughWater
- navigation.speedOverGround
- enviroment.wind.speedApparent
- enviroment.wind.angleApparent
- enviroment.depth.belowTransducer

Often there is only one source for each data item. All this can be gleaned from the delta messages in a minute or so or fetched from the server when the display starts.

I fail to see that list as confusing to pretty much anybody with a little navigation background.

pod909, 2015-01-25:
no problem with that list if the data was at that level but the there's alread value, source, timestamp below that level and the propsal is to add multiple option values as well isn't it?

tkurki, 2015-01-25:
@pod909 The user interface does not need to show all the information that it has. 

For example https://github.com/SignalK/instrumentpanel/ populates the grid as it receives new data items, creating cells when it sees new paths it hasn't seen before. Even if the information it receives has timestamp and source it only shows the name of data item, for exaple navigation.courseOverGround, and the value.

Please take also a look at Delta messages at http://signalk.org/dev/messageFormat.html 

This addition to the spec is about having multiple values for a given data item, say navigation.courseOverGround, in the REST data model. Current delta message format already allows updates to one data item from multiple sources, for example from multiple GPS's.

rob42, 2015-08-06:
The handling of multiple values is superceeded by the method detailed in https://github.com/SignalK/specification/wiki/Multiple-Values-Handling, so closing this one

faceless2, 2015-08-18:
Adding multiple values as described at https://github.com/SignalK/specification/wiki/Multiple-Values-Handling doesn’t seem to work: here’s a section:

```
"self": {
      "navigation": {
        "courseOverGroundTrue": {
          "timestamp": "2014-08-15-16: 00: 01.083",
          "value": 102.29,
          "source": "vessels.self.sources.n2k.actisense-115-129026"
        }
      },
      "sources": {
        "n2k": {
          "actisense-115-129026": {
            "value": 102.29,
            "bus": "/dev/actisense",
            "timestamp": "2014-08-15-16: 00: 01.083",
            "src": "115",
            "pgn": "129026"
          },
```

The trouble is the value 102.29 in the “source” isn’t linked to the key “courseOverGroundTrue”. If that source also provided us with other values, I’d have no way of distinguishing them.

I presume this is a solved problem and that page just hasn’t been updated (and if by chance it’s not, changing the “value” in the source to a tree of values (eg : `{ “self”: { “navigation”: { “courseOverGroundTrue”: 102.29 } } }`) would do it… although this doesn’t allow for different timestamps for different values)

(this is an edited repost of my comments on Slack here, for reference)

tkurki, 2015-08-18:
I noticed the same thing a few days ago when I resurrected my work on this. 

The problem isn't really solved, but imho the solution presented at the beginning of this issue is the best if not perfect option and I'm going to move in that direction in the node ref implementation.

The wiki needs updating, but I would like to verify the thinking with some working software before a real update. I'll throw in a label that the wiki version is not normative and has problems.

fabdrol, 2015-08-25:
Just a thought; we could approach this differently:

- we consider the value of a property to be the primary value. Which one to put there is inferred by server using the priority of the source (e.g. taken from `sensors` or `sources` group). 
- we add a optional key called `alternatives` or something similar which lists alternative values by source id, ordered by priority. 

Example:
```
"01010101": {
  "navigation": {
    "courseOverGroundTrue": {
      "timestamp": "...",
      "value": 102.29
      "source": {
        "label": "...",
        "type": "...",
        "id": "sensor1"
      },
      "alternatives": {
        "sensor2": {
          "timestamp": "...",
          "value": 100.11,
          "source": {...}
        },
        "sensor2": {
          "timestamp": "...",
          "value": 101.2,
          "source": {...}
        }
      }   
    }
  }
}
```

faceless2, 2015-08-26:
JavaScript maps are unordered :-) But of course if you needed to order them you could add a priority field, or you could leave them unprioritised - realistically determining priority beyond the primary might not be much use: I have 5 GPS sources on my boat, I know which one I want to use, if that's not working, well I suppose it's whichever one has a signal. So long as it's consistent.

Personally, I quite like having the alternatives bundled off in a separate section of the datamodel. There are two reasons for this.

1. I haven't investigated the syntax for subscription yet, but if I subscribe to  navigation.courseOverGround I probably don't care that about the 8 alternatives, their timestamps and labels. It's easier to subscribe to "navigation/*" than "navigation/*[not(options)]", or however it is the syntax goes - my xpath is a bit rusty, but you get the idea.

2. Sources will typically provide more than one data value, which means if you "interleave" the various options for each data value into the main model, you're going to repeat a lot of information unless you have some sort of "sources" object and refer to the source by ID. You also won't be able to easily identify all values from a particular source.

I can think of three broad ways to handle this from a datamodel point of view, I'll sketch them out here. All the models below present the same information - course and speed from two different local sources (I'm assuming that sources for data can come from other vessels - kind of the point of SignalK after all - so sources are under "vessels.self" as in the original example in this issue).

### Option 1
First, the "primary values only stored in the main datamodel, alternatives kept with their sources" option. This is the one I'd advocate.
```javascript
{
    "navigation": {
        "courseOverGround": {
            "value": 109.2,
            "timestamp": "2015-08-25T08:12:17.824Z",
            "source": "vessels.self.dev1"
        },
        "speedOverGround": {
            "value": 4.2,
            "timestamp": "2015-08-25T08:12:17.824Z",
            "source": "vessels.self.dev1"
        }
    },
    "sources": {
        "vessels": {
            "self": {
                "dev1": {
                    "label": "Masthead GPS",
                    "timestamp": "2015-08-25T08:12:17.824Z",
                    "bus": "nmea2000:actisense",
                    "values": {
                        "navigation": {
                            "courseOverGround": 107.9,
                            "speedOverGround": 4.2
                        }
                    }
                },
                "dev2": {
                    "label": "VHF Internal GPS",
                    "timestamp": "2015-08-25T08:12:17.824Z",
                    "bus": "nmea1083:/dev/ttyUSB4",
                    "values": {
                        "navigation": {
                            "courseOverGround": 108.7,
                            "speedOverGround": 4.3
                        }
                    }
                }
            }
        }
    }
}
```
Advantages: primary data (eg navigation, environment) is kept concise (which should simplify subscribing to just the values and not the alternatives). It's easy to determine all data from a particular source should you wish to. Disadvantages, there's no easy way to reflect two updates from an individual source with different timestamps (e.g if GPS position was updated once a second, but course/speed only every five seconds). I'm unsure if this matters.

### Option 2
The model describe by fabdroi above, where the options are included inline with the primary datamodel. I've added a "primary" key to each option to indicate which value was chosen, but you could indicate the chosen alternative in any number of ways.
```javascript
{
    "navigation": {
        "courseOverGround": {
            "value": 109.2,
            "sources": {
                "vessels": {
                    "self": {
                        "dev1": {
                            "label": "Masthead GPS",
[... 79 lines cut ...]
                        }
                    }
                }
            }
        }
    },
    "sources": {
        "vessels": {
            "self": {
                "dev1": {
                    "label": "Masthead GPS",
                    "timestamp": "2015-08-25T08:12:17.824Z",
                    "bus": "nmea2000:actisense"
                },
                "dev2": {
                    "label": "VHF Internal GPS",
                    "timestamp": "2015-08-25T08:12:17.824Z",
                    "bus": "nmea1083:/dev/ttyUSB4"
                }
            }
        }
    }
}
```
Advantages: data is not repeated, all alternative values are in the main section of the datamodel. and one source can set data values with different timestamps. Disadvantages: not as brief as option 1, syntax for subscribing to just the primary values is not obvious, and you still can't identify all values from a particular source.

tkurki, 2015-08-31:
Thanks for the input. You raised an issue at least I haven't thought about before: there is currently no way to subscribe to values from a specific source.

tkurki, 2015-08-31:
Imho the values should be available logically near each other. 

In your first example: how you do discover the alternative values for a data item without going through all the sources - and I assume that the sources would have _all_ the sources on the vessel, so you would essentially have a list of all the values in a rather awkward structure.

In option 2/3 I don't get the point for the `vessels.self`part under sources (btw one is sources, other is options?). Skipping vessels.self down there it is pretty much the same as in the beginning of this thread.

The sources section in 3 is not very informative - what does the timestamp stand for there, since there is no value present? Last update - but what update?

tkurki, 2015-08-31:
I think we should also think about this as a REST api and the urls.

`.../navigation/speedThroughWater` should give me "best" value and metadata about it (like that there are optional/alternative values).
`.../navigation/speedThroughWater/options` should give me all the values and metadata about how the server considers the different options. The response should also give me a way to address a specific value in case I want just that one.

Both responses should point to other resources, like sensors and say their related calibration values, if it is not logical or convenient to represent that data as part of this resource.

faceless2, 2015-08-31:
1. Yes, you're right - in option one, there's no easy way to identify the list of alternatives. I've been thinking about it from a server POV until now - I have the alternatives available, I want somewhere to store them in the model. But I have to admit I'm still not clear why a client would ever need to use an alternative value rather than the specified primary value. The only example I recall where alternatives would be of use was two depth sounders, one under each hull of a cat. But for this, and all other cases I can conceive of, the client knows exactly what field it's requesting, *and from what sensor it's requesting it*

2. The explicit reference to vessels.self is in all of the options because a) I wanted to make sure they all represented exactly the same data, and b) I've presumed that it's possible (in theory at least) to get data values from another vessel - e.g. I sail past a boat/buoy/marina that's broadcasting a SignalK model and has a barometer, and I do not - now I can incorporate their data. If this isn't needed then it can be dropped, of course. It was "sources" in the original example and "options" in fabdroi's example - wasn't worried about that as these were just sketching out ideas for structure, not intended as final suggestions.

3. Yes you're right about timestamp in option 3, pointless - should be dropped.

4. I'm not familiar with the API for subscribing to parts of the model - I've presumed there is one and that 99% of the time you won't care about the options or where it came from, just the primary value for that field. If that's not done yet, then yes I'd agree it probably needs fleshing out before this goes much further - perhaps with something like json-path (or some other xpath variation for JSON) rather than reinventing the wheel.

tkurki, 2015-09-01:
The assumption that there is always one primary value for each data item is not valid. 

A Signal K consuming client, like some kind of electronic dashboard, must be able to provide a list of the alternative values. After that it may know exactly what it wants, but discovery and exploration of the available items should be easy.

For example in a case of malfunction (for example a stuck log wheel) you may want to switch to another source.

Imho data from another entity should not be present under vessels.self.

We need to support the use case of a gauge that wants to subscribe to updates from the port hull sounder.

See http://signalk.org/developers/subscription_protocol.html

fabdrol, 2015-09-01:
If we drop the assumption that there always is a primary data source, we could simply drop the idea of having to specify a "primary" or a priority. That would make the schema simpler:

*1. one sensor, one value, indicated by using `value`:*
```javascript
{
  "navigation": {
    "speedThroughWater": {
      "timestamp": "2015-08-25T08:12:17.824Z",
      "value": 9.5,
      "source": {
        "label": "Airmar DST-800",
        "uuid": "22E63B84-749D-489B-A872-8B0D62507C54",
        "type": "NMEA0183",
        "capabilities": ["DBT", "DPT", "VHW", "VLW", "MTW"]
      }
    }
  }
}
```

*2. Multiple sensors, multiple values. Consumers are responsible for choosing one, either by UI or (if a simpler consumer) simply selecting the first one:*
```javascript
{
  "environmental": {
    "depth": {
      "belowTransducer": {
        "values": {
          "starboard-hull": {
            "timestamp": "2015-08-25T08:12:17.824Z",
            "value": 9.5,
            "source": {
              "label": "Airmar DST-800",
              "uuid": "22E63B84-749D-489B-A872-8B0D62507C54",
              "type": "NMEA0183",
              "capabilities": ["DBT", "DPT", "VHW", "VLW", "MTW"]
            }
          },

          "port-hull": {
            "timestamp": "2015-08-25T08:12:17.824Z",
            "value": 6.5,
            "source": {
              "label": "Airmar DST-800",
              "uuid": "22E63B84-749D-489B-A872-8B0D62507C54",
              "type": "NMEA0183",
              "capabilities": ["DBT", "DPT", "VHW", "VLW", "MTW"]
            }
          }
        }
      }
    }
  }
}
```

In case of option 2 and subscriptions, if a user subscribes to `~/environmental/depth/belowTransducer` the user simply receives updates for both values (not that different from when a user subscribes to a higher up path like `~/environmental/depth`, where he would receive updates for `belowTransducer`, `belowKeel` etc). 
REST is not an issue either: when requesting `/signalk/v1/api/self/environmental/depth/belowTransducer` the user simply receives the contents of the values map:

```javascript
"starboard-hull": {
  "timestamp": "2015-08-25T08:12:17.824Z",
  "value": 9.5,
  "source": {
    "label": "Airmar DST-800",
    "uuid": "22E63B84-749D-489B-A872-8B0D62507C54",
    "type": "NMEA0183",
    "capabilities": ["DBT", "DPT", "VHW", "VLW", "MTW"]
  }
},

"port-hull": {
  "timestamp": "2015-08-25T08:12:17.824Z",
  "value": 6.5,
  "source": {
    "label": "Airmar DST-800",
    "uuid": "22E63B84-749D-489B-A872-8B0D62507C54",
    "type": "NMEA0183",
    "capabilities": ["DBT", "DPT", "VHW", "VLW", "MTW"]
  }
}
```

rob42, 2015-09-01:
SInce a single source (like GPS RMC message) can have several diverse values, we should support the idea of the `source` being a reference to more data. So we can use the same rule as elsewhere, if `source` is a string its a reference to another location, if its an object it holds the source data directly.
Then we can benefit from a many-to-one type of simplification.
```
{
    "environmental": {
        "depth": {
            "belowTransducer": {
                "timestamp": "2015-08-25T08:12:17.824Z",
                "value": 6.5,
                "source": "sources.n2k.some-unique-name",
                "values": {
                    "dst800": {
                        "timestamp": "2015-08-25T08:12:17.824Z",
                        "value": 9.5,
                        "source": "sources.n2k.some-other-unique-name"
                    }
                }
            }
        }
    },
    "sources": {
        "n2k": {
            "some-unique-name": {
                "label": "StarboardDST-800",
                "uuid": "9801FBF0-76A3-47DD-ACA6-D76DC405F95C",
                "type": "NMEA2000",
                "capabilities": [....]
            },
            "some-other-unique-name": {
                "label": "PortDST-800",
                "uuid": "7701FBF0-5546-47DD-ACA6-D76DC405F95C",
                "type": "NMEA2000",
                "capabilities": [....]
            }
        }
    }
}
```
That also means we can store config and lots of misc data like version, manufacturer etc without polluting the model. And its still easily available when needed.

In practice I think that multiple values will be the minority, but still important. So I like the idea of a simple `value` key for a simple value. 

For the multple values I think we can then use  `values` is a map of alternative values. This means they are discoverable, and using source refs, not too verbose. So the rules:
1) if there is a value key, thats the only or 'official' value. 
2) If there is a `values` key, that holds alternatives
3) If there is no `value` key, pick your choice  from `values`

faceless2, 2015-09-01:
1. @tkurki , your point about vessels.self has just dawned on me - I'd overlooked that the "navigation" entry is already under vessels.self.

2. @fabdrol, I'd strongly suggest that the server needs to be able to specify a primary value for some fields. I have two electronic compasses, one is +/- about 5 degrees, one is much more accurate. And one of my water temperature sensors keeps adding about 5° to the ambient temperature - sadly, I have to prefer the other one. The server can be told this in its configuration, the client can't, so I don't want to rely on whichever one happens to be parsed first in the data structure.

  Remember, the implied order that you see when the data is serialised is not necessarily the order the keys will be stored in the client environment. The order is not only undefined, it's unstable - add another key to the map and the order may change, which means a display that just chooses the "first" value may oscillate unpredictable between multiple sources.

3. I really like the "value" vs "values" idea (and accompanying example) - I think it covers all possibilities without being too verbose. The only other thing I'd suggest is as well as (or instead of)  "capabilities", which is hardware/source/bus-dependent, perhaps a "provides" key to identify the keys in the model that this source has provided values for - e.g. `provides: ["environment.depth.belowTransducer"]`. This would allow you to find all the values in the model provided by a particular source without having to walk the entire tree (aside: I see arrays have been downgraded from evil to tolerated...)

fabdrol, 2015-09-01:
@rob42 I like your approach, maybe a small change: there should always be a `value` key *unless* we drop the idea that some sources are more important than others (which, as @faceless2 rightly points out, not something we should want). 

So, this changes the rules a little bit:
1) if there is a value key, thats the primary value. A consumer can always rely on this key being present.  
2) If there are more than one values, there is a `values` key with alternatives. The `values` key holds all alternatives including the one in `value` (a consumer can identify which by source).
3) `source` definitions will be changed to a string, which references a `source` object in a `sources` list by JSONPath (?)

Your example would then look like this:
```javascript
{
    "environmental": {
        "depth": {
            "belowTransducer": {
                "timestamp": "2015-08-25T08:12:17.824Z",
                "value": 6.5,
                "source": "~/sources/port-dst800",
                "values": {
                    "port-dst800": {
                        "timestamp": "2015-08-25T08:12:17.824Z",
                        "value": 6.5,
                        "source": "~/sources/port-dst800"
                    },
                    "starboard-dst800": {
                        "timestamp": "2015-08-25T08:12:17.824Z",
                        "value": 9.5,
                        "source": "~/sources/starboard-dst800"
                    }
                }
            }
        }
    },
    "sources": {
        "port-dst800": {
            "label": "Starboard DST-800",
            "type": "NMEA2000",
            "provides": [....],
            // etc
        },
        "starboard-dst800": {
            "label": "Port DST-800",
            "type": "NMEA2000",
            "provides": [....],
            // etc
        }
    }
}
```
I made some small changes to the sources list, but that's a discussion for a different issue. Just for clarification: I removed UUID, if we use a unique key and JSONPath this is not required anymore; I copied @faceless2 idea to drop "capabilities" for "provides" (with a list of paths). Much more useful information, let the parsers worry about N2K and NMEA0183. The "type" might not even be required. Finally I removed the "category" (n2k parent), as SK shouldn't really care where data is coming from IMHO.

fabdrol, 2015-09-01:
P.s.: I think JSONPath is more appropriate than using dot-notation, as JSONPath is standardised and there are many libraries available for many languages.. making it an easy way to work with nested JSON data.

timmathews, 2015-09-01:
So much going on here. I think we're quite close to an appropriate solution, allow me to throw my hat in the ring.

Assumptions/Requirements:
1. A user may have multiple sources for the same data.
2. A user may order those sources by preference on some external criteria (e.g. accuracy, refresh rate, etc.)
3. A user may wish to compare two or more sources, therefore all values must be present in the model and available to consumers.

Unaddressed:
1. What happens when the currently consumed source for a datum goes offline when there are multiple sources for that datum? Is this handled server-side or client-side?

Given the above, I propose the format below. It addresses the need for multiple sources for the same data, presenting a preferred source to the clients along with a list of alternate sources, an ordered ranking of the sources, references to the physical device which provides the data and references from the physical device to the node in the model where its values can be found.

I disagree with the idea that there should be a top level `value` field. It _duplicates_ data which is never a good thing and adds a burden to the server to generate two keys for the same value.

`currentSource` tells the consumer which value it should use for the datum. It serves the same purpose here as `self` does at the top of the model.

`sourceRank` is the "order of succession." If `sourceRank[0]` is not available, use `sourceRank[1]` and so on.

I'm not 100% on the source names, but they're sufficient for this example. I do agree with @fabdrol that we don't need the hierarchy in `sources`.

`$source` is a pointer to the source device which generates the datum.

I'm also not 100% on how the keys in `provides` are named. If we use a path, how much of the path do we need to use?

`$ref` is a pointer back to the place in the model where the value for a specific datum is provided.

Note that I use `$` to identify keys which are pointers to other parts of the model. This is consistent with the usage in JSON-schema and a lot of libraries which marshal to JSON.

One final point, since we're using a NMEA2000 DST800 as our example, it provides water temperature via three different PGNs. Now, two of those are deprecated and may be turned off at the device, I have included them as an array passed to `pgn` to indicate that the water temperature could come from any of them (it makes no difference which, they'll always be the same value).

```js
{
  "environmental": {
    "depth": {
      "belowTransducer": {
        "currentSource": "port-dst800",
        "sourceRank": ["port-dst800", "starboard-dst800"],
        "values": {
          "port-dst800": {
            "timestamp": "2015-08-25T08:12:17.824Z",
            "value": 6.5,
            "$source": "~/sources/port-dst800"
          },
          "starboard-dst800": {
            "timestamp": "2015-08-25T08:12:17.824Z",
            "value": 9.5,
            "$source": "~/sources/starboard-dst800"
          }
        }
      }
    }
  },
  "sources": {
    "port-dst800": {
      "label": "Port DST-800",
      "type": "NMEA2000",
      "provides": {
        "speed/waterReferenced": {
          "$ref": "~/environment/speed/values/port-dst800",
          "pgn": 128259
        },
        "depth/belowTransducer": {
          "$ref": "~/environment/depth/values/port-dst800",
          "pgn": 128267
        },
        "waterTemp": {
          "$ref": "~/environment/waterTemp/values/port-dst800",
          "pgn": [130310, 130311, 130312]
        },
      }
    },
    "starboard-dst800": {
      "label": "Starboard DST-800",
      "type": "NMEA2000",
      "provides": {}
    }
  }
}
```

BTW @fabdrol, JSONPath uses dot notation (http://goessner.net/articles/JsonPath/). Unfortunately, it doesn't have the concept of `~`, that's an extension on our part to mean `$.vessels[self]` in JSONPath notation.

That said, I prefer `/` to `.` because it translates well to URLs which we can use with a HTTP-based API. However, I am fine with using either.

fabdrol, 2015-09-01:
@timmathews my mistake, I was confused and my comment was misguided. Sorry @rob42! 

Tim, I like your approach. My two comments would be: lose the currentSource and simply use the sourceRanking, where sourceRanking[0] is the current source and lose the `pgn` stuff inside capabilities. I really do not see the use case in that, as the translation is already handled somewhere upstream by a parser. But, that last bit is a different discussion in an issue about the `sources` group.

pod909, 2015-09-01:
By what mechanism does the publisher tell the consumer about the accuracy
of the readings available in the consolidated SignalK message, allowing
them to make a decision on prioritizing consumption?

e.g. there may be 4 values for heading, only 1 of which may be accurately
calibrated

tkurki, 2015-09-01:
@timmathews @fabdrol If we lose `value` and just use `values` why not use an array instead of a map with non-functional keys (eg. no special functionality really attached to them) plus a separate order-indicating array? 

If the primary value is always the first one then why not always use an array for the value? Even with just source the item the value would be `values[0]`, `values[0].value for primitive values.

If we go with sources as refs then that should be universal, not related or driven by multiple values.

faceless2, 2015-09-01:
I was about to reply on the same lines - so it would look like this then?

```javascript
"environmental": {
    "depth": {
      "belowTransducer": {
        "values": [
          {
            "timestamp": "2015-08-25T08:12:17.824Z",
            "value": 6.5,
            "$source": "~/sources/port-dst800"
          },{
            "timestamp": "2015-08-25T08:12:17.824Z",
            "value": 9.5,
            "$source": "~/sources/starboard-dst800"
          }
        ]
      }
    }
  },
```

This also has the advantage that you can get the current value for any field in a single query, rather than having to identify the currentSource, then get the value for that source.

tkurki, 2015-09-01:
> 1. What happens when the currently consumed source for a datum goes offline when there are multiple sources for that datum? Is this handled server-side or client-side?

We are talking about a snapshot of latest values here. With just a single source the question is "how long do we keep serving it". With multiple sources the second question  is "how do we order the values considering also their freshness".

I don't think the spec should specify either, as this depends on the item in question and user requirements.

tkurki, 2015-09-01:
> By what mechanism does the publisher tell the consumer about the accuracy
of the readings available in the consolidated SignalK message, allowing
them to make a decision on prioritizing consumption?

That information should go to the `sources` section.

timmathews, 2015-09-01:
Oooh, I really like @faceless2's suggestion, the only issue I see with using an array like that is that it becomes difficult to map in the `$ref` field of a source:
```js
"$ref": "~/environment/speed/values/port-dst800"
```
no longer exists and
```js
"$ref": "~/environment/speed/values[0]"
```
doesn't really work because the order in the array may change.

We would need to use JSONPath style selectors:
```js
"$ref": "~/environment/speed/values[?(@.$source=='~/sources/port-dst800')]"
```

rob42, 2015-09-01:
Arrays are still evil! Using arrays means the data value no longer has an immutable url. eg if a new depth sensor is added and the first removed the array order is different, and a query will suddenly get port depth instead of stbd depth ;-(

Better that the original fails, and some action is required to switch, allowing for notifications etc. 

But otherwise its looking good. Using `value` and `values` just makes the simple case (the most common) easier. Saves a `values` structure for many misc keys that will never have duplicates

Does this work for incoming deltas? Can we still process them efficiently?

Also the fail-over and priority of alternatives is not a spec thing, Its a server implementation thing. 

+1 `provides`

I like the idea of `sources` broken down to `bus` or something. Just aware that we dont really want 10,000 in one level, by bus would auto-limit it to the bus limit - often ~256. But the sources structure desnt need to be formally defined (just the source object). Since its going to be $ref'd anyway, or just searched.

timmathews, 2015-09-01:
But in this case the array order _should_ be able to change. It specifies precedence.

pod909, 2015-09-02:
The array also means that it has to be scanned to find the value for a
particular sensor.

If information on the quality of the value is in the sources there's no
need to order the values.
For my 2c the values should be keyed on source.
The consumer can then do an infrequent assessment using the sources to work
out which one to use for each attribute and then use that to do the value
look-up on an ongoing basis.

Ric

tkurki, 2015-09-02:
Yep, arrays are still evil.

fabdrol, 2015-09-02:
I prefer maps over arrays too. The only real downside to that is that we can't specify the order in the map, but that is easily solved by a "sourceRank" array as proposed by Tim

pod909, 2015-09-02:
or under each source in sources there is a list of the values available
containing a rank or information such as accuracy and frequency to allow
the consumer to make the choice for them selves.

the source Rank suggestion adds infrequently changing data load to the part
of the JSON that will be consumed at high frequency

Ric

fabdrol, 2015-09-02:
@pod909 that is true, but putting that information in sources has a major downside too as a consumer would have to go through every source providing a certain value in order to determine the priorities

pod909, 2015-09-02:
True but they only have to do it once. I'm guesssing for many
implementaions manually as well.

Ric

fabdrol, 2015-09-02:
That's not always the case. Consider @rob42's catamaran; he might build a module for his server that switches the primary echo sounder between port and starboard depending on which hull is leeward at a given time (on sail).

pod909, 2015-09-02:
True but that's detail that can be discovered in the data at present under
any of the proposals (?)

tkurki, 2015-11-09:
https://github.com/SignalK/signalk-server-node/tree/multiple-values now has work in progress of handling multiple values. So far it is minimum effort, eg. minimum changes from the current master and I wanted to see how things actually work if you use just the delta information to construct the tree.

Works only for n2k data so far. At this stage the policy for distinguishing sources just compares `label` (currently empty), `src` and `pgn`. As there is no way to query the N2K bus for further information src is used. This version does not construct `sensors` subtree at all, it just creates the value tree.

Value/values logic is that the most recent data is duplicated in `value` and all the data is under `values`.
```
    "navigation": {
        "speedThroughWater": {
            "source": {
                "label": "",
                "type": "NMEA2000",
                "pgn": 128259,
                "src": "160"
            },
            "values": {
                "-115-128259": {
                    "value": 3.47,
                    "source": {
                        "label": "",
                        "type": "NMEA2000",
                        "pgn": 128259,
                        "src": "115"
                    }
                },
                "-160-128259": {
                    "source": {
                        "label": "",
                        "type": "NMEA2000",
                        "pgn": 128259,
                        "src": "160"
                    }
                }
            }
        }

fabdrol, 2015-11-12:
Looks good to me, two small points: 
- Shouldn't each value field have it's own timestamp, allowing consumers to discard a value if it's too stale from their perspective (putting that decision in the consumer, not the server)?
- Would we still have "singular" `value` if there is just one source, or always go with the "values" format?
- I would propose adding a `latest`/`newest` key (or something like that) somewhere, which has the name of the source/value updated last as value. That way consumers can retrieve the least stale value very easily without having to check the timestamp of each.

tkurki, 2015-11-12:
- I'll fix timestamps
- with just one value there is just `value`, no `values`
- `value` holds the latest value, as there is no other prioritisation. Not really a schema issue, more of a policy decision for the server implementation
- schema changes not started yet

rob42, 2015-11-12:
Yes, I think this works and fulfils all the use cases so far, with the server providing a real or virtual 'value' as required. eg A simple client just subscribes to `navigation.speedThroughWater` and the server provides the default `value`, even if it has `values`. A more sophisticated client would check for `values` and subscribe to a specific one. `navigation.speedThroughWater.values.-115-128259`

sumps, 2015-11-13:
If the consumer or the server for that matter (connected to a gateway like iKommunicate) is to make a reliable choice of the best source from multiple data source, then you will need to improve the metadata related to Sources. We currently have just four fields and in my opinion this is not enough.

I have studied this quite a lot over the last week or so and I think we should replace the "Sensor" object with a new "Source" object that would cover all sources of electronic data that comes in to a Signal K system. Each source will have a Unique Local Name i.e. "Depth Sounder 1" and all applicable data objects that need to be traced to a source will have a source field that stores the Name of its source e.g.  vessels/<id>/environment/depthBelowTransducer would have a field 
"Source" with value of the source's Unique Local Name.

An NMEA to Signal K gateway would be able to automatically create this Sources list from the NMEA data it is receiving and an application would be able to query and display all sources using the restAPI  "vessels/<id>/sources". These could then be presented to the user for manual priority selection or sorted by Interface type, N2K source address, instance or some other method.

I would strongly recommend that we set a field called "Priority" in each traceable data object so that even the simplest of consumer can always be displaying the automatically or manually set highest priority data source.

I think the following fields would be required for a new "Source" object and nearly all of these can be automatically populated by a suitable NMEA to SK gateway....
- uniqueLocalName
- productModel (i.e. N2NET NMEA 2000 AIS)
- productFunction (i.e. AIS)
- type (i.e. Combined GPS/GLONASS)
- protocol (i.e. NMEA2000)
- signalKSource (i.e. iKommunicate 1)
- priority (i.e. 0 highest, 99 lowest) 
- nmea2000SourceAddress (i.e. 23)
- sentenceID (i.e. VDO)
- location (i.e. ABCD dimensions like GPS position on AIS transponders)
- installationNotes (i.e. "Mounted behind chart table with Splitter and GPS antenna on Mizen")
- manufacturersInfo (i.e. "support@digitalyacht.co.uk   www.digitalyachtamerica.com")

fabdrol, 2015-11-13:
@sumps re the decision: for now I think most recent is enough, however...

I agree with your comments re sources, but that is a separate issue. Could you copy your comment to this one? https://github.com/SignalK/specification/issues/84

faceless2, 2015-11-13:
A few thoughts:

1. If you specify "value" as "holds the best value according to the server, which may be the latest or some other preferred value" then you'd cover this either way. It would be nice to allow the server to make the decision if the information is available, and not have the spec constrain it to just reporting the latest value.

2. Crazy not to use this opportunity to give each source a URN - e.g."self://org.signalk.n2k.115.128259", or (bearing in mind these will be used as keys in the "value" map) "self_n2k_115_128259". That you're only comparing one key to map from "source" to "values" is just the first benefit - it also allows for "self_n0183_dev_ttyACM0" or whatever without defining new keys (some of us are still on the old bus, thanks - no "pgm" here).The exact schema for URN can be defined later, but using a single, unique, vaguely hierarchical and vaguely human readable key for each source is, I think, a key point and a will make life easier everywhere: the UI (when choosing a sensor), logging (when identifying input from a sensor), broadcasting (should it ever come to the point where one boat has access to anothers sensor data), and new buses (the future Signal-K certified custom hardware I'm sure we're all working on :-)

3. Maybe I'm being thick but given your comments I'm expecting to see "navigation.speedThroughWater.value" in the above example, and I don't. Assuming it's supposed to be  there, I agree with Rob's observation

timmathews, 2015-11-14:
@faceless2,
 1. I believe the source for the top level `value` can change depending on logic in the server, the spec doesn't constrain it. It also doesn't require that the server have this capability. It's entirely possible for a server implementation to only populate `value` from the source the user selects.

 2. Agreed, better identifiers are required, and they need to support busses other than n2k. We've proposed using "maritime resource names" for vessel identification, these look like `urn:mrn:signalk:uuid:<uuid>`. We could (should) use a similar thing here. And to keep them shorter we can incorporate IPv6 shorthand notation: keep the colons, leave out redundant stuff like so: `urn:::n2k:<name>:<pgn>`.

 See below for more on what `<name>` means.

 3. Yeah, I think `value` got left out in the example above.

#### Identifiers for `values` members

NMEA 0183: `urn:::n0183:<device>:<instance>:<sentence>`
NMEA 2000: `urn:::n2k:<name>:<pgn>`

For n2k, `<name>` by default should be a 32-bit integer representing the unique identifier (ISO Identity Number) and manufacturer ID (these are fields 1 and 2 of PGN 60928 - ISO Address Claim). The server needs to maintain a mapping between these and the source address of the device. `<name>` can be overridden by the user, so something more friendly can be applied like `port-depth` or `aft-gps`.

For n0183, `<device>` is the serial port name (e.g. ttyS0 or COM1). We need `<instance>` to support NMEA 0183 multiplexers like the Actisense NDC-4. According to the docs for the NDC-4, it can be configured to tag sentences to identify the originating device, see [NDC FAQ #2](http://actisense.com/products/nmea-0183/ndc-4/faqs-multiplex4.html?faqitem=faqitem2). I don't have one of these, so I'm not clear on how it works, but it's a scenario we need to support. Whatever it uses to ID the device would go in the `<instance>` field.

That said, I think we may be able to leave the `<sentence>` and `<pgn>` parts off because even if a device transmits multiple PGNs or sentences for the same data (RMA/RMB/RMC sentences and the 1303xx PGNs for instance), the value will be the same since it's coming from the same piece of hardware. Is there another compelling reason to include `<sentence>` or `<pgn>` in the source identifier?

pod909, 2015-11-14:
SI this level of source information really necessary for a v1 spec?

tkurki, 2015-11-14:
Two separate issues to pgn/sentence: 
- is it included in the model?
- is it used to distinguish sources?

Imho it needs to go into the model, distinguishing is debatable.

tkurki, 2015-11-14:
I think the model should support also passive capture  playback eg situation where we have n2k query capability, just a dump of n2k traffic.

tkurki, 2015-11-14:
Id scheme must support also non-nmea sources.

tkurki, 2015-11-14:
Imho sentence/pgn should be there also for debugging purposes. When I see the value I want to be able to trace back where it came from.

[20 more replies]

closed 2016-02-03 by tkurki