SignalK/specification #56: Define specification for Perfomance data

zapfware, 2015-03-06:
To support Performance data like polar/target speed or gybe and tack angles, a new json specification has to be defined.

zapfware, 2015-03-08:
Here's my draft for the performance scheme:

```
{
    "type": "object",
    "$schema": "http://json-schema.org/draft-03/schema",
    "id": "https://signalk.github.io/specification/schemas/groups/performance.json#",
    "description": "Schema describing the performance child-object of a Vessel.",
    "title": "performance",
    "properties": {
        "targetSpeed": {
            "description": "The current target speed in m/s based on current polar, current trueWindSpeed and truewindAngle",
            "$ref": "../definitions.json#/definitions/numberValue"
        },
        "targetSpeedPercent": {
            "description": "The percentage of current target speed to the speed through water",
            "$ref": "../definitions.json#/definitions/numberValue"
        },
        "veloctictyMadeGood": {
            "description": "The current velocity made good in m/s derived from the speed through water and appearant wind angle",
            "$ref": "../definitions.json#/definitions/numberValue"
        },
        "veloctictyMadeGoodToWaypoint": {
            "description": "The current velocity made good in m/s to the next waypoint derived from the speedOverGround, courseOverGround",
            "$ref": "../definitions.json#/definitions/numberValue"
        },
        "beatAngle": {
            "description": "The beat angle for the best velocity made good based on current current polar and trueWindSpeed",
            "$ref": "../definitions.json#/definitions/numberValue"
        },
        "beatVelocityMadeGood": {
            "description": "The velocity made good in m/s for the beat angle",
            "$ref": "../definitions.json#/definitions/numberValue"
        },
        "gybeAngle": {
            "description": "The gybe angle for the best velocity made good downwind based on current polar and trueWindSpeed",
            "$ref": "../definitions.json#/definitions/numberValue"
        },
        "runVelocityMadeGood": {
            "description": "The velocity made good in m/s for the gybe angle",
            "$ref": "../definitions.json#/definitions/numberValue"
        },
        "leeway": {
            "description": "Current leeway in degrees Centigrade",
            "$ref": "../definitions.json#/definitions/numberValue"
        }
        "tack": {
            "description": "Magentic heading on opposite tack",
            "$ref": "../definitions.json#/definitions/numberValue"
        }
    }
}
```

tkurki, 2015-03-08:
Great! Would you mind forking the specification repo, adding this and creating a pull request?

Some typos there: magentic, velocticty.

Why is tack magnetic? Could we have tackMagnetic and tackTrue?

If there is velocityMadeGood and velocityMadeGoodToWaypoint should runVelocityMadeGood be velocityMadeGoodDownWind (and maybe velocityMadeGoodUpWind) or should velocityMadeGood just be aware of run/beat?

Btw did you have in mind a json representation for raw polar data?

zapfware, 2015-03-08:
Fixed he typos.

The velocityMadeGood is about your current VMG / what your actual sailing - and should be positive for Upwind and negative for Downwind (added it in the desrciption)

I updated the scheme maybe it's clearer now.

At least I will think about a json representation for polar data.

tkurki, 2015-03-11:
This looks good - could you create a Pull Request for this?

zapfware, 2015-03-11:
https://github.com/SignalK/specification/pull/57

closed 2015-03-31 by tkurki