SignalK/specification #66: Meta data for measurement units

MarkGlover, 2015-06-19:
I can convert from SI units to any popular unit. I can not implement and maintain a signalK database at each client just to know what the measurement unit is.

rob42, 2015-06-19:
Hi Mark, 
Could you elaborate more on the problem (your use case perhaps). When Signal K uses only SI units we mean a specific base SI unit. eg all velocity values are in meters per second, never km/hr. So its more a mapping of measurement type to SI unit you need, eg for velocity, temperature, distance, etc.

Also if you know the type of the destination (eg Nautical Miles) then the signal K source will be a 'distance', hence in meters

fabdrol, 2015-08-01:
Closing for now. This has been addressed by (1) Rob's comment and (2) the addition of meta properties to the spec.

closed 2015-08-01 by fabdrol