SignalK/specification #71: Vessel without MMSI

MariusVolkhart, 2015-08-17:
In the root document, a vessel can be identified by its MMSI or an UUID. The `vessel.json` specifies only a `MMSI` field. In the event that a vessel is identified by an UUID, what is the correct value of the MMSI field in `vessel.json`: `null`, empty string (`""`), or the UUID?

fabdrol, 2015-08-17:
@MariusVolkhart In that case there should be an `uuid` field on `vessel`. It should have been in there, thanks for spotting it. I'll update the document asap.

tkurki, 2015-08-17:
@fabdrol doc change is not quite enough, requires a schema change. How does json schema model "either of these properties"?

fabdrol, 2015-08-17:
@tkurki "document" == `vessel.json` ;). I'm not sure you can do "required unless", but I'll check it out.

fabdrol, 2015-08-19:
See #83

fabdrol, 2015-08-19:
@tkurki I don't think "required unless" is an option, let's just solve that in documentation?

tkurki, 2015-08-19:
Isn't this really `either MMSI or UUID`? 

See 
- http://json-schema.org/example2.html nfs example
- https://github.com/json-schema/json-schema/wiki/anyOf,-allOf,-oneOf,-not

MariusVolkhart, 2015-08-19:
`oneOf` is new in v4 of json-schema :(

fabdrol, 2015-08-19:
Yes but that all relates to the value, not the key. An option would be to use `oneOf` with a `RegExp` key

timmathews, 2015-08-19:
No need for "required unless". UUID is *always* required and valid as a hash key or the value of `self`. If the optional MMSI exists it is also valid as a value of `self`.

tkurki, 2015-08-19:
Sorry about my casual early comments that added just confusion.

So this is how it is in master:
- `root.self`is a required `string`
- `root.vessels.<id>.mmsi` is optional

@fabdrol's change in #83 
- same as in master plus
- `root.vessels.<id>.mmsi` is optional

So in effect the only thing that is required in addition to self is the `<id>` key under vessels. If the key is mmsi then it should appear also as  `root.vessels.<id>.mmsi`. In case it is uuid you can tell that it is uuid (eg. some generated key) by the fact that there is no mmsi matching the `<id>` key.

I would go with this and cancel the PR.

fabdrol, 2015-08-20:
Reason I like the `UUID` on vessel is for this use case:
- I subscribe to local boat
- I need to query or post to the REST api, and want to know the UUID. I could retrieve the entire tree first and read the value of "self", but that seems like overkill. 

It's easy for a server implementation to add the UUID to the vessel. Besides, since UUID has the same function as a MMSI, it should be implemented the same IMHO. Alternative is to remove MMSI as well, as it can also be inferred from the value of `/self`, but as I said I think that's overkill for the consumer.

tkurki, 2015-08-20:
Well, the node server provides `uuid` as per #83 and you can retrieve just the localboat subtree by the path `self`:

http://demo.signalk.org:3000/signalk/api/v1/self
[image]

http://demo.signalk.org:3000/signalk/api/v1/
[image]

fabdrol, 2015-08-20:
so it should either go into the spec or we should take it out of server.

sumps, 2015-08-20:
Forgive me if I have misunderstood, but it seems as if the UUID is the minimum required identification field for "Self" in order for the Signal K schema to work. I would suggest that the UUID is randomly generated by whichever device is responsible for "Self" in the system i.e. the Server or if no Server the Gateway or if no server or gateway the consumer (App). It is not necessary for the user to be aware of this UUID or to have any part in the generation and use of it.

There are four other identifiers for a boat i.e. Boatname, Call Sign. IMO number and MMSI, but in theory a user in their canoe with a smart phone will not have any of these identifiers, but the App on his phone could generate the UUID and still be able to send data to a Signal K compatible cloud server.

A server receiving AIS data will not have to worry about duplicate vessels as each Signal K vessel it stores will have a unique MMSI, must have to transmit over AIS system and we just have to specify that no Signal K device should transmit either Self or other vessel data without their being a valid MMSI or UUID field.

The only problem I see, is that the UUID is transient as there is no mechanism for the UUID to stay associated with a vessel and in fact it should not be used in this way only to allow Signal K data to flow around the world with unique identifiers. Worst case is that a Cloud server could receive data about a vessel from more than one Signal K device and each would have a different UUID even though it was the same vessel.

fabdrol, 2015-08-20:
@sumps the UUID is generated once, and should prob be based on time (ensuring - correct me if this is not true - a unique UUID). The idea is that vessels could share data over wifi for which they don't need an MMSI. However, since MMSI is unique too this can be used in place of the UUID.

Now, an UUID is generated by a server but only once and then stored. This way, vessels that have met before can recognise each other again (related to a yet-to-be security implementation). Of course, when a user wipes memory or gets a new server, they could end up with a new UUID but that's not too much of a big deal imho.

sumps, 2015-08-20:
Could we not use the Link Local IPv6 address as the UUID ? This is a well documented procedure and may be a useful piece of information in the future as IPv6 finally comes of age.

sumps, 2015-08-20:
MMSI and UUID must remain as two distinct and totally separate data fields. The original question raised by Marius spoke about putting the UUID in the MMSI field and this is absolutely not allowed and I would also suggest that putting the MMSI in the UUID field would not be good practice either. There have been many cases where MMSI numbers have been entered incorrectly, duplicated (by users) or just made up and having a UUID that is totally separate to the MMSI is a good feature, particularly if it is automatically setup and requires no user interaction.

timmathews, 2015-08-20:
Nevermind that a vessel could have **multiple** MMSIs and they can change over time. @sumps is right, UUID and MMSI are separate and must always be separate. The UUID is the canonical vessel identifier. MMSI (or IMO number) can be a useful way to look up a vessel, but cannot be the unique ID.

I retract my previous statement that MMSI can be a valid value of `self`. I was momentarily swayed towards this point of view, but upon reflection I don't think it's really viable.

The only remaining question is whether we want reproducible UUIDs or not (i.e. UUIDv4 or UUIDv5)?

MariusVolkhart, 2015-08-20:
IMO the UUID only needs to be unique to the current "session". What I mean by that, is that if I'm developing an application I will likely create a keying system of my own. I might key off a hull number, or the combo of skipper name and boat name, or something else that is relevant to my application. What I am hesitant to do is to rely on a keying system that I do not have control over and that might have unexpected side effects.

That said, if I repeatedly get the entire data model, during my current "session" it would be nice to have UUIDs be constant.

fabdrol, 2015-08-20:
@MariusVolkhart UUIDs need to be consistent across sessions for social boating, communicating with marinas etc; for local data consumers this is not such a requirement as they can simply query or subscribe (to) self .

fabdrol, 2015-08-20:
Btw, since we're going off-topic can I consider the original issue closed?

tkurki, 2015-08-20:
One more a bit off topic remark: so far we have considered alternative ways of keying the vessel object in the context of a single json doc and related json schema.

If we think about this more as a REST url modeling issue there is nothing keeping us from providing alternative paths to the same data:

/signalk/api/vX/vessels/byMMSI/ => index of all vessels with MMSI
/signalk/api/vX/vessels/byMMSI/123456789 => vessel 123456789 data
/signalk/api/vX/vessels/byUUID/ => index of all vessels with UUID (that is: all vessels?)
/signalk/api/vX/vessels/byUUID/A2345678 => index of all vessels with UUID (that is: all vessels?)

fabdrol, 2015-08-20:
In that case I would argue for an alternative endpoint, eg `/v1/query?byUUID=...` or `/v1/query/byUUID/..`.. The regular endpoint follows the same structure as the tree, which is a very nice convention IMHO

fabdrol, 2015-08-20:
I'm closing this issue, as the question was answered and it's fixed with #83. We can keep discussing this though

closed 2015-08-25 by fabdrol