SignalK/specification #74: How to handle Non-Vessel AIS messages ?

sumps, 2015-08-17:
How should SK cope with AIS targets that are not other Vessels i.e. Base Stations, AtoNs, SAR aircraft, AIS SARTs/MOBs, etc.

I am not comfortable handling these AIS targets as "Other Vessel" and think we should create a new "Mariners Object" (or similar) in the Root Schema for handling any object that is important to navigation but is not a vessel.

I am sure there are other objects that we would want to communicate in the future such as an Iceberg report, the position of Whales, etc. and so a more generic extensible object that could handle these AIS target types and other future objects is desirable. 

This should probably be considered outside the scope of 1st release but it needs to be considered at some point.

fabdrol, 2015-08-17:
Related to #35

tkurki, 2015-08-17:
Concrete examples in the form of NMEA0183 and N2K messages would help here. Either as raw data or web links.

sumps, 2015-08-17:
The best reference for all AIS message types is from the ITU document...
http://www.itu.int/dms_pubrec/itu-r/rec/m/R-REC-M.1371-5-201402-I!!PDF-E.pdf
It defines the different messages and the binary message format that is then encapsulated in the NMEA sentences and PGNs

sumps, 2015-10-15:
In the Root Signal K schema we should consider the following new category;
-  AISObjects
      - AtoNs
      - SARaircrafts
      - BaseStations
      - aisSARTs
This new category holds all of the non-vessel AIS targets. All will have an MMSI number to identify them and each will have specific data fields as defined in the ITU spec for AIS (reference given in previous comment).

tkurki, 2015-10-29:
I don't think we should extend the logical data model based on the fact that the information is from AIS. 

Take AtoN as an example: if you updated your Signal K data model with the information from the most recent Notice To Mariners you could have a UI that plots up to date AtoNs from manually entered and AIS data.

So AtoNs could go under `resources.aidsToNavigation`.

As far as I understand base stations are for synchronizing transmissions and the only useful information from it is location/position and the fact that base stations are in range. Reminds me of a waypoint, even if the name doesn't fit.

One cheapish way to include aircraft and sart is to create a top level category like  `other` and pick fields from `vessels` category and include enough metadata like `type`.

Do you have data sets that you could share with this data?

rob42, 2015-11-12:
I agree with Teppo on this, but with more categories maybe. 
* `vessels` takes normal ais and aisSARTs (treat a rescue beacon as a type of vessel)
* AtoNs could go under `resources.aidsToNavigation.`
* aircraft go under 'aircraft'
* base stations go under 'fixed'

but for v1 I think we  just put aircraft and fixed into 'other' and mark a experimental.