SignalK/specification #76: Remove derived values from environment.depth

MariusVolkhart, 2015-08-17:
`vessel.environment.depth` has 5 properties: `belowKeel`, `belowTransducer`, `belowSurface`, `surfaceToTransducer`, and `transducerToKeel`.

2 of these are calculated values.

I would expect that, as a skipper, I have to tell the system my `surfaceToTransducer` distance and my `transducerToKeel` distance. The transducer will produce the `belowTransducer` value, and clients will calculate `belowKeel` and `belowSurface` only if they want it. The argument could also be made that `surfaceToTransducer` and `transducerToKeel` are calculated.

The issue I see with having all 5 values in the spec is that there is a possibility for a conflict of truth.

Tangential thought: I imagine `surfaceToTransducer` and `transducerToKeel` will be set once and then forgotten. Does it make sense to have them in the `environment`, or are the better suited in the `design`? The latter seems more appropriate to me.

Thoughts?

fabdrol, 2015-08-17:
@MariusVolkhart I'd say they go in the group `sensors` (defined as a sensor position or something). But, that group isn't really defined yet/needs a lot of work. 

Sensors should be configured by the owner. I'd argue that a server shouldn't try to calculate anything unless the related sensor is defined in the sensors group. But, this needs more discussing. Any other thoughts?

sumps, 2015-08-17:
In general this info is configured on the Depth Instrument, Echo Sounder, Fish finder, etc. Then the NMEA0183 or NMEA2000 messages are populated as the settings allow i.e. DBK. DBS, DBT in the older NMEA0183 systems, DPT (which includes the transducer offset) in newer NMEA0183 systems and the Depth PGN in NMEA2000.

Many systems only store one Depth Offset value and if it is positive it is the transducer to surface offset and if negative it is transducer to keel offset (I might have got the polarity wrong ;-).

I would populate the Signal K values from the 0183/2000 data and not get too hung up on allowing customers to change the values on the gateway/server otherwise you could have different values which could be dangerous. 

Consumer developers can allow the depth value to be switched between surface/transducer/keel but my recommendation to theses developers is to make this depth value clearly indicated and also to warn the user if they have selected for instance keel depth and the offset value is 0.0, indicating that the value in the system is suspect.

tkurki, 2015-08-17:
I don't think that all derived values should be pruned from the spec. The same logic would apply for true wind, for example, but I don't think that every Signal K consuming application needs to know how to calculate it.

We have also envisioned that a Signal K server could have a mapping/enrichment layer to it, where one could place logic that derives new values from existing ones (for convenience) or adjusts values for more accurate reading. An example might be adjusting wind angle for heel. It doesn't produce a logically separate value, but nevertheless is a derived value.

Then there's the case of Depth Below Keel DBK NMEA 0183 sentence. What if that is all you get and nobody has configured in the server the proper value for conversion? http://www.catb.org/gpsd/NMEA.html#_dbk_depth_below_keel

You can think of the Signal K schema also as a REST api structure spec. A server could provide http://demo.signalk.org:3000/signalk/api/v1/vessels/10101010/environment/depth/belowTransducer, but if requested it could also provide http://demo.signalk.org:3000/signalk/api/v1/vessels/10101010/environment/depth/belowKeel as a convenience method, so that a consumer need not request/handle `transducerToKeel` separately.

Furthermore I don't really see a reason for strict pruning of the tree.

fabdrol, 2015-08-25:
There are a couple of ways server can know this information in order to calculate it. If this information is not present, the values could be left out of the tree (i.e. they're not `required`)

- The information is present from setting up the depth transducer in the `sensors` group
- A server implements a config page to set-up this information
- A server pre-fills this information from any of the N2K or N0183 messages, unless it's already been set by the user. 

This puts the responsibility on the server author to implement this correctly, but I do think it adds a layer of usefulness to SK servers. 

@MariusVolkhart would you agree, or do you find this an issue still (if so, let's discuss it some more - otherwise I'll close)?

MariusVolkhart, 2015-08-25:
Has there ever been any discussion around doing a server spec either as an addition to or instead of just the data spec? I ask because the server is The Brain of SK, and does much more than the spec defines, which is a good thing. However, I don't think json-schema lends itself particularly well to describing server behaviors, and so communicating expectations and what to expect to developers is challenging.

Take this issue as an example. As a developer of a consumer app, I 100% want the server to provide additional info. But I opened the issue because I didn't understand the server behavior to expect.

It has taken me a while of mulling this over to really figure out what it was that was weighing on my mind about this.

In terms of a server spec, I'm picturing something that describes the rest routes available and how the server will behave when it gets conflicts, how it prioritizes sources, and how a value from a `GET` call might return more info than a `POST` call.

I'm coming at this from the perspective of, "I want to build plug-and-play Android apps against a known server behavior". I don't know if the SK server is the one to provide this, or if another "application server" should sit between my app and the SK server, but I hope this helps in framing my thoughts.

Regardless, feel free to close this issue. We can take this conversation elsewhere if appropriate or nip it in the bud if it isn't befitting.

tkurki, 2015-08-25:
See https://github.com/SignalK/specification/issues/85 for further discussion.

closed 2015-08-25 by tkurki