SignalK/specification #86: Boat Brand and Model, plus standard "Self" Settings File

sumps, 2015-08-28:
There are a large number of available settings within SK and although I am sceptical that your average user would set all of these settings, I think we need to have a Boat Brand field, Model field and YearBuilt field so that standard boat setup files can be created.

For instance a Jeaneau Sun Odyssey 41DS will have a set of dimensions and polars that are the same for every boat and by storing the Boat Brand, Model and YearBuilt data we could encourage the collection and publishing on the SK site of the standard settings file for common boats. This would make it easier for customers as once one customer has gone to the trouble of entering all of the data, it will be available for others to use.

Thinking about this, it would be a good idea to define a standard "Self" settings file that all consumers/servers/gateways used to store this data. The file would have to have a Version Number that was linked to the SK specification version but it could be a useful and social way for SK users to share boat data.

tkurki, 2015-08-31:
It would be really cool to a have global shared & shareable (Creative Commons license or similar) repository of yacht technical data, addressable by url, say

http://yachts.registry.signalk.org/Jeanneau/SunOdyssey/41DS

fabdrol, 2015-09-01:
It would definitely make setting this up easier; after installing a sk device you simply connect to the internet, search for your boat, click, done. If it's not there, a user could choose to add it which means crowdsourced data..

rob42, 2015-09-01:
This is an idea I have had for Freeboard too. We could provide a service that has templates based on typical boat types, which does basic config for you, eg 11M sailing yacht. Then subclasses of increasing specialization. So eventaully Jeanneau/SunOdyssey/41DS. But more than just config you could collect and compare reliability, polars, comparisons, tracks, and social meets, swaps etc.

webmasterkai, 2016-07-08:
I'm interested in building out this resource. Perhaps something in partnership with sailboatdata.com? I posted a message on the forum asking about creating and API for the website http://sailboatdata.com/bbs/index.php?topic=2236.0

It would be trivial to extract the fields in the SignalK schema and save using the appropriate field ids. Initially read-only but eventually accepting info as @rob42 suggests.

sumps, 2016-07-09:
I think this is a great idea but it is important that the data is freely available to all Signal K users and developers. This is the first time that I have heard of sailboat data.com but it looks like it is a site setup by yachtsmen for yachtsmen so perhaps it would be a good partner.

sumps, 2016-07-09:
i am sure the core Signal K team would welcome any work you could do on this, have you joined the Signal K Slack channel ?

webmasterkai, 2016-07-11:
http://slack-invite.signalk.org Isn't working for me at the moment. I'll join when I get it to load.

I haven't heard back from Randy Browning of Norwalk, CT, creator of sailboatdata.com yet. Hopefully he's okay making the data accessible to all users. I was able to setup a parser to extract the data from the html. If I get the okay I'll create a file for each record and an index file to make searching by name easy.

sumps, 2016-07-12:
Just to clarify my previous comment, I don't think any Signal K user would mind registering with sailboatdata.com, so I am not suggesting that there is a completely open section of the site to download Signal K setting files.

Are we missing any boat data objects in Signal K that are collected and stored on sailboatdata.com that we should consider adding to the Signal K spec ?

joabakk, 2016-11-10:
It seems if we still want to go down that route, that YachtDatabase.com lends itself for scraping, as the search terms are open in the url. Reading from the discussion on sailboatdata, they don´t seem to be too keen on helping out.
but we might have to ask first: http://www.yachtdatabase.com/app/copyright.jsp