SignalK/specification #90: Storing app and server configs in the Signal K model

rob42, 2015-09-19:
Copied here from other discussion on slack, for wider feedback

The signalk model currently has allowance for `_xxxx` names, where the underscore denotes that it should never leave the server, eg stripped out of the messages. That makes them available only within the server.

So in the java server I will just store the config in `vessels.self.config` in a freeform json to suit myself. 
I have a jsonschema for the config data and get a nice screen by using http://jeremydorn.com/json-editor/

The config app reads the config schema, and shows the config data.

So I guess I need some security rules, so that config is only writeable by apps loaded from the same server, but I think that can be handled by the existing security model. Hence an app could bundle a config schema, which could then display using the system config screen. But maybe thats too restrictive.

Maybe `/signalk/v1/config/` would be useful though - very easy to restrict access. But I like the idea of storing in the main model, since we inherit all the existing functionality, even history (config auditability).
Plus we can distribute config updates across a big network of signalk devices the same as normal messages

fabdrol, 2015-10-10:
I like this idea. How free would server/gateway developers be in putting config data in here? Should we define the config data in a schema or allow whatever? I think we'll never be able to define all keys required for every configuration, but some standard ones would be nice as it would enable us to share config GUIs

sumps, 2015-10-10:
Can you give some examples of the sort of configuration settings you foresee being included in the schema ?

rob42, 2015-10-10:
Ive been working on this today, It seems to work quite well. I needed to have some sort of config gui to be able to set up users/roles/permissions to trial the security stuff.
So my scheme is poratble and flexible - basically Ive made a json schema for the config data, complete with labels and descriptions. 
Then I use http://jeremydorn.com/json-editor/ to recover the schema from my server, and then recover the /signalk/v1/api/config data from the REST interface. It displays a config screen, not ideally pretty, but pretty good.

Whenever the config is changed it sends it back to the REST interface as a POST, which updates the model. 

So in my model I have two heads config.* and vessels.*. For practicality the current model is saved to disk at shutdown, and I also save config.* to disk on every change. So at reboot I load config.*, then configure, and load self.conf - which restores the data to the data at last shutdown.

Hence all I would need to show your config would be a json-schema and the current state as json.

Now I can quickly change the jsonschema with options for users/roles/etc, and instant gui!

sumps, 2015-10-10:
@rob42 in a complex system, actually even just a system with a gateway and a server, will you not need to have a "master" and "slaves" type arrangement where one server takes the lead role and control of this configuration data ?

This goes back to an issue I made a few months back about a gateway having two modes (or a menu option) that makes it responsible for "Own" data and the way I see it, your concept of the config data is just an extension of this.

rob42, 2015-10-11:
I think the option to store config data in the sk model is purely optional. For me I need the ability to store config on a per server instance basis, and for it to be available to a web config screen, and potentially other clients or even servers. Since I already have a system (signalk) that can handle json data from all sorts of sources, and I'm inherently lazy, it was easy to reuse that. And it has the added flexibility of being easily available to any sk capable device.

Looking at two gateways, in my current setup they both share `config.*`, which would only work for common  config data. So thats not ideal, I'll have to think about that one.

So at present i think its got potential, but if you prefer to use a more traditional system thats fine too.

BTW the concept of master-slave is not the best analogy. Like Git they are all peers, but some are more important (or definitive) than others. Some are naturally subservient, like freezer temp transducer. So its more about capabilities like 'internet connection', or N2K gateway, etc. The full distributed design is still not clear yet, but peer-to-peer allows for much more powerful routing, availability, and management of messaging later.

eg The master-slave concept creates star-of-star networks. The problem is reliability - if the central gateway dies, everything is gone. In peer-to-peer there can easily be two duplicate gateways - one dies, just connect to the another. This is what the mDNS provides - each device advertises its capabilities.

tkurki, 2015-10-11:
Could we come up with use cases where config type data needs to be stored on the server and then see how this solution suits us? 

@rob42, what do you actually store there?

I see a need for storing consumer configuration data shared over consumers, for example InstrumentPanel layout & settings.

rob42, 2015-11-21:
Ive used this a fair bit now on the java server. It stores the server config, so its entirely implementation specific, and probably has no wider use, except if I have a collection of same brand servers.

The concept of using the signalk server functionality to manage and store config data works pretty well, as I can obtain it using REST, and POST updates etc. The same gui tools I use to display signalk data works for the config, so lots of reuse and good things.

But the downside is extra complexity in keeping the signalk vessel data separated from the config data. Its added a lot of `if(config..` to my code, which is not good. The more complex the more bugs, and less secure etc.

Probably the best solution is to run separate virtual servers in the real server - one for signalk data, one for config, on different ports. Then they are both simple cases, but can share functionality.

Anyway closing this for now to clear the list a bit

closed 2019-03-27 by rob42