SignalK/specification #92: Allow compact delta with default values omitted

tkurki, 2015-10-09:
https://groups.google.com/d/msg/signalk/XFnlGYweN9Q/ZwqJt9EsCAAJ

Minimal delta doesn't need
- timestamps
- source
- context

Change the schema to make these optional. 

It might be a good idea to allow optional validation with all fields mandatory, in case you want to check that you are outputting fully populated delta.

fabdrol, 2015-10-10:
If a producer sends a "minimal" delta without timestamp, source and context what information would a server use for these fields when constructing a full tree? Last values of these fields (which assumes a producer did send a "normal" delta at some point)? The timestamp received + the physical location of the source?

rob42, 2015-10-10:
I think its up to the server to add those (or not). The effect is that the reading is not as precise - eg it was 4.5M deep here, at some point in the past...
So if you are relying on the data you just need to take that into account. For most onboard use cases the data is basically assumed to be 'now' anyway, but for crowd-sourced depth its obviously useless.

So I think it works naturally - if you get depth readings from your own sensor, add current time, maybe source 'self' or something.  If you get depth from an external source with no timestamp, then use with caution.

sumps, 2015-10-10:
Delta's by there nature are output when values change, NMEA data is received, etc. and so a time stamp is not critical as the receiving system can take time of reception to be within a few milliseconds of time of occurrence. 

The chances of a system not having GPS data is unlikely so servers will always have GPS time to synchronise their internal Data/Time and so Delta data can be logged and time stamped by the server with a high level of confidence.

fabdrol, 2015-10-11:
Okay, makes sense. Who can do a PR?

tkurki, 2015-10-19:
Context was already optional as it was tagged `required` in a way that Json Schema doesn't support. No test to show this...

See b65ef0a71a5073c459d9ab5ecd3c574fa03bc7f0

tkurki, 2016-01-23:
I made `updates`, `values`, `path`and `value` required and added tests for this in 25af8d3f00f15344a0ec92ed6bbcc9c2fd1eeb1a. 

Other fields are thereby optional.

closed 2016-01-23 by tkurki