SignalK/specification #94: Data Time Outs

sumps, 2015-10-12:
We have to be careful that consumers (even simple ones) handle data time outs correctly, otherwise a user could be happily thinking that the readings are OK, when in fact a sensor sending the data had failed five minutes ago.

I would like to propose that every sensor data value has a time out value (in seconds) that is set by the gateway or server. Then the consumer can receive this time out value and know when they should time out the data and make some indication of the time out to the user.

rob42, 2015-10-12:
Good point - an additional key in `.meta` is probably best since thats the source of alarm and gui meta data

sumps, 2015-10-12:
As long as we can associate a time out with each data type i.e. Wind, Depth, Heading, GPS, AIS, etc.

tkurki, 2015-10-12:
Something like this:
- navigation
 - speedThroughWater: 2000 ms
 - courseOverGroundTrue: 2000 ms

sumps, 2015-10-12:
The timeout values may change from system to system depending upon the equipment installed but they can be detected and set by the gateway or server receiving the NMEA data.

I think it is fair to say that with its higher bandwidth, the timeouts for N2K data will be shorter than for 0183 where the sentence transmission rate is generally slower. For this reason different data sources need to  have different time outs.

rob42, 2015-11-21:
Adding it to the `.meta` is the best place. Even with multiple sources this still works. Also we dont have to send timeout with each message, as we have the timestamp, so a client can decide for their own case. 

The server can also monitor keys for timeouts and raise alarms as a background process. Those alarms can then drive more complex behaviour like not resending too old values, or resetting or diagnostics on the source. Plus a gui could allow users to adjust them

When the gateway obtains timeout data from N2K it can set it via a normal message by including `meta`, but I think this should only happen at setup or some specific case, not every message.

rob42, 2015-11-21:
Oops - just commited my timeout change to master - I thought it was in a branch, and I would do a pull :-(
Best have a look and reverse if needed - sorry

rob42, 2015-11-21:
Here: https://github.com/SignalK/specification/commit/e0563d4c7acba38922c2e0b82d1e053759c14dd9

tkurki, 2016-04-09:
Signal K schema provides the timeout property in the meta object that is included under all values. This can be used to specify and make available the desired timeout for individual items in the data model.

Data timing out is not to be indicated with a delta message, so that is it not confused with invalid data. If a timeout is defined in the meta object a server should observe that when serving data via the REST api.

sumps, 2016-04-09:
Subscription servers should also take time outs in to consideration and send null fields of data stored in the schema as timed out.

sumps, 2016-12-21:
So can we consider this Issue closed ? @rob42 added the Timeout field in the Metadata, so we have a mechanism for timeouts and I am happy to add a Timeout description to the documentation.

rob42, 2016-12-22:
ok by me

closed 2016-12-22 by rob42