SignalK/specification #96: How will Signal K handle Alarms/Alerts ?

sumps, 2015-10-15:
It would be very good if a Signal K gateway/server could pass through NMEA2000 Alerts, AIS MOBs, etc. or there was a mechanism for a gateway/server to trigger an alert/alarm. One way would be to create a new object in the Root called Alarms;
- Alarms
    - aismob#12
    - alert#367
    - alarm#490

Each Alarm would reference the Signal K object that the alarm was related to i.e. a sensor, engine, aisSART, etc. so that it would be easy to access the relevant properties of the alarm by calling the object....  signalk/api/v1/vessels/self/propulsion/transmissionOilPressure/value

Obviously there are a lot of things to consider such as resetting/timing out the alarm and it maybe out of scope for V1 but we need to start considering this issue.

sumps, 2015-10-17:
OK I have found Alarms in the Vessel object, not sure how I missed that before, and I will start thinking how this can be mapped to NMEA2000 alerts. 

The gateway will need to have a setting for alarms allowing some of the fields to be set like notification method, enabled/disable, etc.

I'll include my ideas in this issue.

rob42, 2015-11-21:
See https://github.com/SignalK/specification/wiki/Alarm-Handling for a description of alarm use. However the method described assumes there is a signalk key to set the alarm against. While most alarms are related to some sort of measurement and therefore have signalk keys, there are some that dont have obvious homes. I'm thinking about MOB, fire, etc. Maybe post a list of any that surface?

rob42, 2015-12-21:
@sumps Can you list a few NMEA alarms we need to deal with, other than those that will be handled by the current 'check against limits' scheme. MOB comes to mind - we dont want to compare 'crew' with limits in `.meta` and raise alarms if we loose any:-)

rob42, 2015-12-22:
Some more thoughts on this. The `vessels.uuid.alarms.` tree is assumed to be populated by the server or an alarms daemon from the rest of the signalk tree, by comparing with the values bounds in `.meta`

Then the alarms that exist there are notified to screens etc by an alarm monitoring process. So plugging into this would be a good thing.

There is no reason that a value cannot be simply posted directly in there, eg `vessels.uuid.alarms.mob`. All the better if they map back to a signalk key, eg engine fire would be `vessels.uuid.alarms.propulsion.engine1.fire` - which would be more useful than a simple fire alarm, as you immediately know where to look.
Will need to look at `alarmState`, probably needs another state `emergency`

rob42, 2016-01-10:
@sumps - any chance to have a look at this yet?

rob42, 2016-01-11:
See #152

rob42, 2016-01-14:
See https://github.com/SignalK/specification/wiki/Alarm-Handling

tkurki, 2016-04-06:
https://github.com/SignalK/specification/wiki/Alarm,-Alert,-and-Notification-Handling

rob42, 2016-06-10:
I think this is sorted now - closing

closed 2016-06-10 by rob42