SignalK/specification #99: Multiple SK Devices - Need for a Master/Slave Relationship

sumps, 2015-10-19:
In the various discussions that have taken place concerning UUIDs, MMSI, etc. it has highlighted again that in a more complex Signal K network, we could have multiple servers, gateways and sophisticated consumers but no method of synchronising important "Self" Configuration Settings, Meta Data and the UUID.

[image]

As it stands at the moment, all of this data would have to be entered in each device, with the possibility of wrong or different values being entered.

I propose that there is a need for a simple method or allocating one device to be the "master" network device and that this device once configured will announce its presence on the network via Bonjour/mDNS as being the master device and all other devices must (if they need it) request the "Self" Configuration Settings, Meta Data and UUID from the "master" and maintain a local record of this data and check it (re-sync) each time they power up and join the network. Slave devices should also maintain a record of who the "master" device is.

Should a device that is set to be a "master" power up and Discover that another "master" already exists on the network or a "slave" powers up and finds that a new "master" now exists on the network, then it should alarm/warn the user that an installation change has occurred and needs to be checked.

If a slave powers up and finds the "master" is not currently present, then it should be able to carry out its normal activity without the need of the "master" using its local record of the "masters" config, meta and UUID data.

An exact definition of what a "Master" device is responsible for needs to be agreed, a format/method for the "Master" to announce its presence via Bonjour/mDNS and further study of the use cases that could occur in a complex network (such as two masters, a new master, etc.) and how these are handled.

Comments please.

tkurki, 2015-11-01:
In the presence of multiple servers clients should get additional metadata in the discovery phase to be able to connect to the preferred server. This may or may not be the master server, eg. the one responsible for master config.

tkurki, 2015-11-01:
Proposal: add `roles` as
- key to mDNS metadata record
- Signal K data model eg. REST api
- part of the `hello` message sent when a client connects via WebSocket

Values:
- `master` / `slave` to guide config mastering
- `main` /  `aux` to guide consumers which server they should connect to

So in a simple shipboard system with just one server the server would have the roles `master` and `main`.

sumps, 2015-11-22:
Would it be possible for someone to generate a definitive example of what the mDNS text record should be ?

I guess the restAPI file could be identical in format, but is there a standard format for the Hello message in a WS ?

We have the basic mDNS done, but I want to get this tied down and finished off in iKommunicate

tkurki, 2015-11-23:
Hello message:
```
{
  "name": "signalk-node-server X",
  "version": "0.1.20",
  "timestamp": "2015-11-23T06:48:31.953Z",
  "self": "123456789",
  "roles": [
    "master",
    "main"
  ]
}
```

tkurki, 2015-11-23:
mDNS ad: keyword `roles`, comma separated list.

https://github.com/SignalK/signalk-server-node/blob/3cb87faaa339da1e4a8f6a293c4e8d537cf20fb9/lib/mdns.js#L35

sumps, 2015-11-23:
OK, checking online the mDNS txt_record is just a list of attribute keys and attribute values organised like this....

name=iKommunicate X
version=1.00
self=123456789
config=master
deltas=yes
subscriptions=yes
skversion=1.00

So the above txt record file would tell a consumer; the name of the device (set by user), teh firmware version of the device, the vessels self UUID (either MMSI or UUID), whether it is the master or a slave device i.e. has all of the configuration data for self, if it supports Delta Files and Subscriptions and the Signal K specification version.

Is there anything else that needs to be in this txt_record file ?

sumps, 2015-12-11:
Has anyone got anything else to add to this proposed TxtRecord ? Can we freeze this for V1 and add it to the Signal K website as part of the protocols ?

sumps, 2015-12-11:
I have edited the example above and this is what I am giving to John to incorporate in to the iKommunicate Bonjour TxtRecord.

tkurki, 2016-01-07:
* Master = definitive source for information about `self` (identity etc).
* Slave = opposite of master ;-) should get identity etc from master

* Main = clients should connect to this one
* Aux = opposite of Main 

as I thought that these two may be different, eg. the master may be less powerful but still the master and clients should connect to another slave server. This may be a bit far fetched and now that I think of this a "connection priority" might serve this better.

sumps, 2016-01-07:
I think we only need the one setting Master or Slave and this basically tells the consumer or another server if the device is the "Master" and responsible for Self settings and configurations. I am still not 100% clear on what these responsibilities are and I am sure we will work this out as we get more practical experience of more complicated SK systems but I feel putting this setting in the mDNS now and raising the issue of this master slave interaction will make it easier in the future.

sumps, 2016-01-07:
I think all devices should default to "Master" and consumers/servers should list all SK devices they discover and let the user decide which they want to connect to (with the option to remember this connection). If a User does not want to enter all of the Self data and settings on every consumer/server, then then can set the consumer/server to be a slave and take the data/settings from the Master.

sumps, 2016-01-07:
The only other consideration, is on the issue of saving the Consumer settings on a server. This has been discussed (https://github.com/SignalK/specification/issues/90) and it might be worth having a setting in the TXT_Record of whether the device is capable of "Storing Settings".

sumps, 2016-01-07:
@tkurki I note your comment about Text Records potentially creating confusion and we need to be very careful about restricting their use to only the most important of settings and making sure that each setting in the TXT_Record is well documented, but if we get this aspect of Signal K right, it will make for a much better User Experience.

jboynes, 2016-01-07:
I like the approach of allowing multiple masters @sumps described. I see a use-case in larger systems for a high-availability version using multiple masters and consensus algorithms for storing critical information. As hardware costs fall, this could become more commonplace i.e. RAID for SignalK servers.

This group (which could have only one member) would advertise itself over mDNS. When consumers connect, they would use endpoint discovery to to determine what endpoints were being offered by which servers. A server could return its own endpoints, endpoints of other members of the consensus, or endpoints of servers that were simply following changes.

This supports @tkurki's case for offloading UI. For example, I can see low-power, always-on headless devices being used as data masters, with higher-power devices like MFD panels being used for user interfaces.

timmathews, 2016-12-04:
Keep in mind that Signal K must be able to work without any information in the TXT record at all. From the RFC:

>The intention of DNS-SD TXT records is to convey a small amount of useful additional information about a service.  Ideally, it should not be necessary for a client to retrieve this additional information before it can usefully establish a connection to the service.  For a well-designed application protocol, even if there is no information at all in the TXT record, it should be possible, knowing only the host name, port number, and protocol being used, to communicate with that listening process and then perform version- or feature-negotiation to determine any further options or capabilities of the service instance.

Whatever information we include in the TXT records ought to be available by querying the SK server itself. DNS-SD TXT records are really just an _optimization_.

Based on this, while updating the documentation in #300, I've come to the conclusion that all we need in the TXT record is the following:

```
txtvers=1
roles=master,main
self=urn:mrn:signalk:uuid:c0d79334-4e25-4245-8892-54e8ccc8021d
server=signalk-server
version=0.1.23
```

pod909, 2016-12-07:
On see some more roles:

Input devices - sensors, control interfaces etc.
Output devices - displays, horns, actuators, valves etc
Gateway/broker - a device responsible for aggregating messaging but that doesn't store it

tkurki, 2016-12-18:
I think the current state of things is documented in http://signalk.org/specification/master/connection.html well enough. Extensions such as envisioned above should be discussed & specified  separately as use cases arise once we have v1 out of the door.

closed 2016-12-18 by tkurki