SignalK/specification #100: Specification is not yet complete: for v1 we need to support the following list of PGNs and sentences*

fabdrol, 2015-10-29:
As discussed on Slack and the mailing list, the specification as it stands right now doesn't support all PGNs and sentences we should support in v1. This GH issue is an attempt to collect and list all PGNs + sentences - both supported and unsupported. Supported messages will be ~~crossed off~~.

How this will work: below is an (initial) list of 0183 sentences and PGNs, based on the list of PGNs decoded by canboat and the 0183 sentences documentation from various online sources. This list is by no means complete. **Please post additions to the list in the comments, I'll add them to the list in the OP**.

*Note: I haven't yet crossed off any PGN and sentence, as I thought it to be a good idea to go through the list first. It could be that canboat supports PGNs that we don't want to support in v1. Please comment on that too.*

PNGs                                                                  |0183 Sentences
----------------------------------------------------------------------|--------------
**59392** ISO Acknowledgement                                         |APB           
**59904** ISO Request                                                 |DBT           
**60928** ISO Address Claim                                           |GGA           
**61184** Manufacturer Proprietary single-frame addressed             |GLL           
**61440** Unknown single-frame non-addressed                          |HDM           
**65001** Bus #1 Phase C Basic AC Quantities                          |HDT           
**65002** Bus #1 Phase B Basic AC Quantities                          |MWV           
**65003** Bus #1 Phase A Basic AC Quantities                          |RMC           
**65004** Bus #1 Average Basic AC Quantities                          |VDM           
**65005** Utility Total AC Energy                                     |VDO           
**65006** Utility Phase C AC Reactive Power                           |VTG           
**65007** Utility Phase C AC Power                                    |GSA           
**65008** Utility Phase C Basic AC Quantities                         |GBS           
**65009** Utility Phase B AC Reactive Power                           |GSV           
**65010** Utility Phase B AC Power                                    |BWC           
**65011** Utility Phase B Basic AC Quantities                         |ZDA           
**65012** Utility Phase A AC Reactive Power                           |AAM           
**65013** Utility Phase A AC Power                                    |BOD           
**65014** Utility Phase A Basic AC Quantities                         |RMB           
**65015** Utility Total AC Reactive Power                             |XTE           
**65016** Utility Total AC Power                                      |DPT           
**65017** Utility Average Basic AC Quantities                         |MTW           
**65018** Generator Total AC Energy                                   |VLW           
**65019** Generator Phase C AC Reactive Power                         |VHW           
**65020** Generator Phase C AC Power                                  |HDG           
**65021** Generator Phase C Basic AC Quantities                       |MWD           
**65022** Generator Phase B AC Reactive Power                         |XDR           
**65023** Generator Phase B AC Power                                  |              
**65024** Generator Phase B Basic AC Quantities                       |              
**65025** Generator Phase A AC Reactive Power                         |              
**65026** Generator Phase A AC Power                                  |              
**65027** Generator Phase A Basic AC Quantities                       |              
**65028** Generator Total AC Reactive Power                           |              
**65029** Generator Total AC Power                                    |              
**65030** Generator Average Basic AC Quantities                       |              
**65240** ISO Commanded Address                                       |              
**65280** Manufacturer Proprietary single-frame non-addressed         |              
**65285** Lowrance: Temperature                                       |              
**65286** Airmar: Boot State Request                                  |              
**65287** Simnet: Configure Temperature Sensor                        |              
**65289** Simnet: Trim Tab Sensor Calibration                         |              
**65290** Simnet: Paddle Wheel Speed Configuration                    |              
**65292** Simnet: Clear Fluid Level Warnings                          |              
**65293** Simnet: LGC-2000 Configuration                              |              
**65325** Simnet: Reprogram Status                                    |              
**65341** Simnet: Autopilot Mode                                      |              
**65371** Seatalk: Keypad Message                                     |              
**65374** SeaTalk: Keypad Heartbeat                                   |              
**65408** Airmar: Depth Quality Factor                                |              
**65410** Airmar: Device Information                                  |              
**65480** Simnet: Autopilot Mode                                      |              
**65536** Unknown fast-packet addressed                               |              
**126208** NMEA - Acknowledge group function                          |              
**126270** Maretron: Slave Response                                   |              
**126464** PGN List (Transmit and Receive)                            |              
**126720** Manufacturer Proprietary fast-packet addressed             |              
**126976** Unknown fast-packet non-addressed                          |              
**126983** Alert                                                      |              
**126984** Alert Response                                             |              
**126985** Alert Text                                                 |              
**126986** Alert Configuration                                        |              
**126987** Alert Threshold                                            |              
[... 117 lines cut ...]
**130818** Simnet: Reprogram Data                                     |              
**130819** Simnet: Request Reprogram                                  |              
**130820** Fusion: Transport                                          |              
**130821** Furuno: Unknown                                            |              
**130827** Lowrance: unknown                                          |              
**130828** Simnet: Set Serial Number                                  |              
**130831** Suzuki: Engine and Storage Device Config                   |              
**130832** Simnet: Fuel Used - High Resolution                        |              
**130834** Simnet: Engine and Tank Configuration                      |              
**130835** Simnet: Set Engine and Tank Configuration                  |              
**130836** Simnet: Fluid Level Sensor Configuration                   |              
**130837** Simnet: Fuel Flow Turbine Configuration                    |              
**130838** Simnet: Fluid Level Warning                                |              
**130839** Simnet: Pressure Sensor Configuration                      |              
**130840** Simnet: Data User Group Configuration                      |              
**130842** Simnet: AIS Class B static data (msg 24 Part B)            |              
**130843** Simnet: Sonar Status, Frequency and DSP Voltage            |              
**130845** Simnet: Parameter Handle                                   |              
**130847** SeaTalk: Node Statistics                                   |              
**130850** Simnet: Event Command: Unknown                             |              
**130851** Simnet: Event Reply: AP command                            |              
**130856** Simnet: Alarm Message                                      |              
**130880** Airmar: Additional Weather Data                            |              
**130881** Airmar: Heater Control                                     |              
**130944** Airmar: POST                                               |

fabdrol, 2015-10-29:
@canboat I used the `pgns.json` file as an input for the PGN list. Are there PGNs that can be taken of the list completely?

canboat, 2015-10-29:
All manufacturer specific PGNs can be removed. That is anything starting with regex `[A-z]+: `

canboat, 2015-10-29:
Also all CAN bus internal messaging: 59392, 59904, 60928, 61184, 61440, 65240, 65280, 126208, 126464, 126720, 126976.

fabdrol, 2015-10-29:
Thanks @canboat, list has been updated.

rob42, 2015-11-21:
Remeber we can do v1 without half of these as more pgns = additions, eg v1.1, v1.2 etc. Since key additions are generally non-breaking changes we can safely allow them. We do need to get the high level structure right though.

tkurki, 2016-01-23:
I don't see much value in this issue - it is not usable for tracking progress because there is no granularity and nobody has updated it for months. I'd like to close it.

tkurki, 2016-04-07:
The schema is in pretty decent shape in terms of supporting a reasonable NMEA coverage. This is issue has not been super useful in tracking sentence/pgn support, which is more driven by n2k-signalk and  signalk-parser-nmea0183 Development and should be tracked there.

closed 2016-04-07 by tkurki