SignalK/specification #104: Have just one VHF Call Sign but add an IMO number

sumps, 2015-11-14:
My understanding is that a vessel will only have one Call Sign and yet in the Communication Object there are two call sign fields...
- callsignDsc string 
- callsignVhf string 

Can we have just one call sign for VHF (DSC or non-DSC) but add an IMO number field, so that IMO vessels are supported in Signal K.

timmathews, 2015-11-14:
callsignDsc and callsignVhf do seem redundant, but they've been in since the very beginning, so I can't say what the impetus was for having both. I think it's safe to remove callsignDsc.

IMO number probably belongs in the Vessel object along with MMSI, name, etc.

rob42, 2015-11-21:
They are there from when I whipped up a first hack at communications :-)
We dont use DSC here, but my vhf has it - so I added the field as a guess!
Feel free to change to a better form, I dont know of anything that currently uses the keys, so nothing will break..

sumps, 2015-12-12:
PR created issue can be closed.

closed 2015-12-12 by sumps