SignalK/specification #111: Easy to use, localizable units & names extracted from schema

tkurki, 2015-11-21:
There is the manually created, easy to use and localizable file that contains short & long names and unit  for Signal K data items.

https://github.com/SignalK/specification/blob/master/i18n/en.json

Now that units are embedded in the schema we should produce this file from the schema automatically. English versions of short & long names should probably be embedded in the schema.

rob42, 2015-11-21:
Good by me - just for the record, the file naming should follow the common [locale]_[variant]_[sub-variant].json format, so I can mod this to en_nz.js. Then when looking for files, we trim and retry, `en_nz.js` > `en.js`

timmathews, 2016-09-30:
How about just wrapping the localization data in a locale object:

```
"en_US": {
  "alarms.anchorAlarmMethod": { },
  "alarms.anchorAlarmState": { }
}
```

This is the approach that RoR takes and it seems to work pretty well. That way it doesn't really matter what the files are called and you can even split up the localization among many files and just merge the objects.

tkurki, 2017-10-07:
No pull for localisation and I am not sure if this should be part of the spec or an external resource, maintained separately and just depending on the schema / keys list.

closed 2017-10-07 by tkurki