SignalK/specification #134: Define fuel rate for electric drives

jboynes, 2015-12-16:
[fuelRate](https://github.com/SignalK/specification/blob/master/schemas/groups/propulsion.json#L97) is expressed in `m3/s` which makes sense for diesel and gasoline engines but not for electric drives.

That could be expressed simply as input voltage and current. I've not thought about the implications of a 3-phase drive.

rob42, 2015-12-22:
The usual measurement is watt-hours ( watts*time), normally expressed as kWh (kilowatt-hours). But looking about the net there seems to be no mention of it in regard to electric drives. Maybe they are so pleased to be 'green' that they havent thought about it?
Or maybe we could measure it in kg (of batteries) :-)
It can wait for clarification or v2 I think. I dont see any pratcical  way to make the liquid fuel value relevant, so it will almost certainly be a separate key.

tkurki, 2015-12-29:
Isn't fuel rate for an electric drive its electric energy consumption, power (W, watts), not energy (kWh, kilowatt-hours)?

tkurki, 2016-02-03:
There is now https://github.com/SignalK/specification/blob/master/schemas/groups/propulsion.json#L104

Is this still needed?

timmathews, 2016-09-30:
@tkurki, was your link above to https://github.com/SignalK/specification/blob/ffb098dd490374230a10d694e17a3a30f71a4b0a/schemas/groups/propulsion.json#L104? `engineLoad`?

Let's revisit it later when one of us gets a boat with an electric drive.

closed 2016-09-30 by timmathews