SignalK/specification #135: Using enum constraints

jboynes, 2015-12-19:
In [a comment](https://github.com/SignalK/specification/pull/120#issuecomment-166017110) on #120, @tkurki wrote:
> My suggestion is that
> 
> * we continue with the enum typed values for stuff that is naturally enumerable
> * make sure the enum values cover most use cases
> * write software defensively, eg. even if a field is an enum in the spec the software should not break with the appearence of "illegal" values (remember, somebody might add values to the spec behind your back)

My concern is that a validator like tv4 will report this an error and the application then has to determine which ones it can ignore. If we add an `anyOf` clause:

```json
"properties": {
  "value": {
    "anyOf": [
      { "enum": ["Forward", "Neutral", "Reverse", "Fault"] },
      { "type": "string" }
    ]
  }
}
```

the json will validate with other values. The schema calls out the well-known values but specifically says other strings are allowed thereby telling the application developer to be defensive.

sumps, 2015-12-19:
This seems like a good solution, if I have understood it correctly. A way to combine Enum and Strings which is compatible with test and validation tools.

sumps, 2015-12-19:
NMEA data uses Enum a lot and so having this in SK makes the job of a gateway very straight forward and efficient but if we can allow an SK server or consumer to enhance or modify the data it receives from the gateway by using strings, then this is a big advantage.

tkurki, 2015-12-20:
>In json itself there is no distinction between string and enum values. There is no schema validation in node server and I doubt there is any in the java server or other servers that provide a Signal K interface.

>Node server actually is not really aware of the schema at all, it just acts on incoming data and from what I gather so does java server.

>With enums we have the option to validate the structure, value types and values in Signal K data. This way we can check that widely used data such as NMEA0183 and NMEA 2000 data is mapped in a predictable way.

>If somebody chooses to use enum values that are not in the set specified in the schema it may break some consumers. With a string typed field the same thing would happen. Same thing if the software is not updated in sync with the spec.

Thanks for pointing out the `anyOf`possibility. It provides a way to explicitly list a set of special values instead of just listing them in a comment.

However it is not any different from `string` field in validation - any value will validate.

Your concern is that ad hoc values not in the enum list will cause validation errors and there is more value in making the fields just strings so that extending the value set freely is valid.

My concern is that with no way to validate well known values we will end up with more bugs and interpretations. In practice this doesn't forbid ad hoc extensions if SK handling software is able to deal with them, which I think should anyway be the case.

One way to have the cake and eat it would be to do the `anyOf` trick and provide custom validation implementation that has a *strict mode* for this kind of structures.

sumps, 2015-12-20:
I had not appreciated the important point that you noted Teppo and unless we have a large number of volunteers to produce detailed implementation and real like example documentation then Enum has to be the way to go for V1 as it is effectively self documenting and ensures we retain maximum compatibility with NMEA data formats.
Unless custom validation is quick and easy, my vote stays with Enum.

rob42, 2015-12-20:
The anyOf enum/string is a neat solution but as @tkurki says it means extra work at present in the custom validation. I think its easy to forget that unlike other protocols the signalk format is extensible. So we can have an enum with all the common values we know know for v1, and simply add values for v1.1 as required.

A v1 device will still be able to use the V1.1 message, especially if it doesnt validate (as is the case with java and node servers). The only effect the additional v1.1 items will have on the v1 device is if custom processing or action is required for the new value in which case defensive programming of the action should take over.

eg the case above v1 might be:
```"enum": ["Forward", "Neutral", "Reverse", "Fault"] },``` 
V1.1 might be
```enum": ["Forward", "Neutral", "Reverse", "Fault", "Service"] },```

A V1 gauge or alarm would either ignore the Service param, or show some sort of 'Unknown' indicator, or even just a user message [Unknown state: "Service"]

sumps, 2015-12-20:
Good example @rob42 and confirms that Enum is the best solution for SK V1, but thanks @jboynes for making such a positive and informed suggestion on this issue.

rob42, 2015-12-21:
Just occurred to me we can simply use 
```"enum": ["Forward", "Neutral", "Reverse", "Fault", "Other"] },```. 
I did a  a pull request with changes on #120

jboynes, 2015-12-21:
Taking @rob42's example, "simply adding values" in v1.1 will cause errors in a v1.0 application regardless of whether it is using the JSON schema rules or has custom validation; at the v1.0 level the "Service" value is _strictly invalid_ and will result in an error as `enum` does not allow undefined values. The consequence is that a system cannot reliably[1] exchange v1.1 messages until all components have been upgraded from v1.0. I don't think that's the intent.

Contrasting with `object` `properties`, we do have an open contract as the [`additionalProperties`](http://json-schema.org/latest/json-schema-validation.html#anchor64) schema facet is generally omitted which _explicitly allows for_ properties not defined by the schema. There's no need to version the schema and everything using it in order to add a new property; the only constraints are that the property must not change the meaning of any existing ones (so that applications that ignore it still work correctly) and must not conflict with values added later by the specification (which can be achieved with namespaces).

If JSON-Schema had a similar contract for an open `enum` we would not have an issue. The `anyOf` workaround is mimicking that to some extent but it does not allow for a "strict" [validation mode](https://github.com/json-schema/json-schema/wiki/ban-unknown-properties-mode-(v5-proposal)) that flags additional content; it is always allowed.

My concern is that if we are going to allow applications to use JSON-Schema validation then it should work correctly. We should not tell application developers "you can try validating against the schema but you might get errors from undefined values in enums which you have to determine if you can ignore." That requires the application to have custom validation for enum values so we might as well write the schema so that the basic validation check passes in the first place.

There are certain applications where validation may not be needed at all. For example, an aggregating server can just build the tree based on messages received and serve the tree it has to clients; it does not need to verify that the tree is valid in order to function. Our Node and Java server implementations currently operate in that mode. Similarly, a NMEA to SK bridge deals with a known set of PGNs where each has a defined mapping into SK. It can safely ignore any open `properties` in the SK model because it has no way to present them in NMEA.

[1] I say "reliably" because although message exchange can occur if the applications ignore errors that tends to lead to other failures later.

jboynes, 2015-12-21:
That was a long post, so TL;DR: There are three modes for dealing with a SK JSON:

No validation:
1. Parse the JSON (syntax)
2. Do **not** validate against the schema
3. Validate specific values as they are needed for processing

Validating in the presence of closed enums:
1. Parse the JSON
2. Validate against the schema; expect errors from undefined values in enum; choose which to ignore
3. Validate specific values as needed for processing

Validating with open values:
1. Parse the JSON
2. Validate against the schema; errors indicate a basic problem
3. Validate specific values as needed for processing

Our current servers are operating the first way. We are proposing the second model which I think is problematic and am advocating for the third.

sumps, 2015-12-21:
@jboynes a version 1 consumer will request data from a Signal K gateway or server using the V1 calls either through the restAPI or streaming a Delta file (Websocket)...

restAPI           -       /signalk/v1/api/......
websocket     -      /signalk/v1/stream/....

It is likely that servers/gateways will support multiple versions of Signal K so for instance simultaneously providing a V1 consumer with V1 data and another V1.2 consumer with V1.2 data.

jboynes, 2015-12-21:
@sumps Which translates the problem into version negotiation and
translation. A client has to discover which versions a server supports and
talk at the appropriate level; the server has to be able to translate
between all versions except it can't translate to newer versions than it
implements. This causes a problem with down-level intermediaries - for
example, a V1.2 client cannot exchange V1.2 data with another V1.2 client
via a V1.0 server if that server validates and rejects non-V1.0 enum values.

The current solution, where the server does not validate, is a good one
from the openness perspective as it does not have that problem. It means
there will be data in the model that some consumers (clients or servers) do
not understand but that is OK so long as they can agree on the core (well
known keys, values and message formats).

This issue is a variation of the "Open-Closed Principle" about being
designs being open for extension and closed for modification; enums by
their nature require modification and so violate this.

rob42, 2015-12-21:
Yes I see your point. Basically we are trapped by a weakness in the enum schema definition, either we lock down and cause version problems later, or we open and cause validation complexity now.

It looks like its a question of choosing the lesser evil here. The compatibility of clients across new versions is a prime goal, so we should make that a priority. Which means we take a hit on validation and go with the `anyOf` solution

```
"properties": {
  "value": {
    "anyOf": [
      { "enum": ["Forward", "Neutral", "Reverse", "Fault"] },
      { "type": "string" }
    ]
  }
}
```
Any ideas on a validator or mitigation that will reduce the effects of the above on our schema validation? I think it is still important to be able to flag a value as 'not in enum' for a given version. Is there a way to adjust validators globally for tv4?

rob42, 2015-12-21:
Looks like Line 23 in https://github.com/geraintluff/tv4/blob/bd19f5aa481afc1bf9434b9f91ffa30221cc3d09/source/combinations.js

```
ValidatorContext.prototype.validateAnyOf = function validateAnyOf(data, schema, dataPointerPath) {...}
```

Maybe we could override it for our enum needs?

sumps, 2015-12-21:
I think this is a big mistake. We need to get new developers using Signal K and opening up a wild and flexible world of strings might end in so much confusion and broken apps that Signal K does not have a future, so making it fully extensible and flexible will have been a waste of time.
Who will define the common strings ? Who will document them ? We struggle enough with volunteers to get the code written and if it needs loads of documentation then I think this is a risk.
I would prefer to see a step by step approach. We make V1 fairly controlled but validated and reliable with extensive use of Enums and then once everyone is comfortable and we have a stable and growing  community, we can have a more flexible and extensible V2.
Signal K is going to grow and evolve in ways we will never be able to predict, so we do not have to get everything supported and right first time, but we should grow it from a firm and stable base.

jboynes, 2015-12-22:
@sumps I'm with you on building a community of new developers. I think once basic hardware is there (I'm looking forward to iKommunicate for example), I think we will see new, interesting and very varied applications especially given the number of simple sensors that can be built and added to the SignalK network. An open model facilitates that as developers won't be blocked waiting for specification enhancements.

Answering specific questions:
* The specification is defining the common keys and values.
* The specification is documenting them

jboynes, 2015-12-22:
@rob42, @sumps on ideas for mitigating random expansion, HTTP allows this for message headers by specifically defining any beginning with "X-" as being extensions; MIME (content-types) does something similar. Taking that idea, how about we specify:

* All keys and enum values are reserved for use by the specification, except keys and enum values beginning with "x-"
* Applications may use additional keys or enum values but they **must** begin with "x-"

We can define a schema constraint for that as follows:
```json
"gear": {
  "type": "object",
  "properties": {
    "value": {
      "oneOf": [
        { "enum": ["Forward", "Neutral", "Reverse", "Fault"] },
        { "type": "string", "pattern": "^x-.*$" }
      ]
    }
  },
  "patternProperties": {
    "^x-.*$": {}
  },
  "additionalProperties": false
}
```
In plain English:
* "gear" is an object
* it has one well know property `value` which must be one of the 4 enum values or a string beginning "x-"
* it allows additional properties but only if their name begins with "x-"

@rob42 for automated validation, rather than tweak `tv4` we could write a JSON-Schema transformer that removed all the extension points to produce a "strict" schema. An application could then use a normal validator to check if a document had any open content. As an example, the fragment above would get transformed to
```json
"gear": {
  "type": "object",
  "properties": {
    "value": {
      "oneOf": [
        { "enum": ["Forward", "Neutral", "Reverse", "Fault"] }
      ]
    }
  },
  "additionalProperties": false
}
```

tkurki, 2015-12-22:
What would be the reason to use private extensions and not add them to the spec?

jboynes, 2015-12-22:
Primarily to allow experimentation before the spec gets updated. Once there's consensus then it would officially get added to the spec. These are not really _private_ extensions as anyone can see/use them, they are just not-yet-formally-codifed functionality. It helps avoids mistakes in the spec that can be very hard to undo.

By analogy, the X-Forwarded-For header is widely used and it didn't need an official update to the HTTP spec to gain acceptance (which is a good thing considering how disruptive a new version of HTTP is).

tkurki, 2016-01-09:
As there are no known validating consumers and we will benefit from being able to validate against enums without special validation code I'm closing this for now. The issue may be revisited once we have v1 out.

closed 2016-01-09 by tkurki