SignalK/specification #146: Add definition of mdns/bonjour service

rob42, 2016-01-05:
We currently advertise the signalk services via mdns in both the java and node servers, and reference the idea in http://signalk.org/developers/apis_and_conventions.html, but we have no formal service name and message content written yet.

So as a starting point the java server advertises:
```
Service Type: _signalk-http._tcp
Service Name: signalk-http 
Domain Name: local
Interface: wlan0 IPv4
Address: 10-1-1-64.local/10.1.1.64:8080
TXT vessel_uuid = urn:mrn:signalk:uuid:c0d79334-4e25-4245-8892-54e8ccc8021d
TXT server = signalk-server
TXT version = 0.1
TXT vessel_mmsi = 123456789
TXT path = /signalk/v1/api
TXT vessel_name = motu
```
```
Service Type: _signalk-ws._tcp
Service Name: signalk-ws
Domain Name: local
Interface: wlan0 IPv4
Address: 10-1-1-64.local/10.1.1.64:3000
TXT vessel_uuid = urn:mrn:signalk:uuid:c0d79334-4e25-4245-8892-54e8ccc8021d
TXT server = signalk-server
TXT version = 0.1
TXT vessel_mmsi = 123456789
TXT path = /signalk/v1/stream
TXT vessel_name = motu
```

```
Service Type: _signalk-tco._tcp
Service Name: signalk-tcp
Domain Name: local
Interface: wlan0 IPv4
Address: 10-1-1-64.local/10.1.1.64:55555
TXT vessel_uuid = urn:mrn:signalk:uuid:c0d79334-4e25-4245-8892-54e8ccc8021d
TXT server = signalk-server
TXT version = 0.1
TXT vessel_mmsi = 123456789
TXT vessel_name = motu
```
If this is correct then I will write up some docs etc

rob42, 2016-01-05:
This has the additional service ```signalk-tcp._tcp``` added. Thats for a straight tcp connection transferring json text. Ive added a separate issue for the tcp protocol, mainly message delimiter definition.

sumps, 2016-01-06:
@rob42 we need to also consider the points raised in this issue...
https://github.com/SignalK/specification/issues/99

tkurki, 2016-01-06:
@rob42 just to be clear: in RFC 6763 terms what we need to specify is *Service Name*, which are
* `_signalk-http._tcp`
* `_signalk-ws._tcp`
* `_signalk-tcp._tcp`
and I believe that is it! 

In addition to this the service advertising the Signal K services will have an Instance Name and the full domain name will be something like  `MySignalk._signalk-ws._tcp.local`.

https://tools.ietf.org/html/rfc6763#section-6.3 says
>For a well-designed application protocol, even if there is no information at all in the TXT record, it should be possible, knowing only the host name, port number, and protocol being used, to communicate with that listening process and then perform version- or feature- negotiation to determine any further options or capabilities of the service instance.

Thus all more detailed information, such as master/slave/main/aux in #99 should be available over websocket and http at a minimum. To discourage using TXT records for application logic I think we should not specify anything there in the Signal K specification. Remember that not all clients will use mdns/bonjour.

You stated that the _we have no formal service name and message content written_ - what did you mean by that? I thought that apis and conventions would be to place where stuff like this should be found? In effect "the specification" would grow out of the apis and conventions and json schemas.

rob42, 2016-01-07:
Re 'no formal service name and message content written', was just an observation. I looked through the schemas, and the API page but the only ref to mdns was a brief paragraph at the bottom of the API page saying we would use `_signalk-ws._tcp, _signalk-http._tcp`
When I implemented this in the java server I had no idea of what to add and what was required.  What I will write is a more complete reference that states what MUST, SHOULD and MAY be in the mdns record.
Personally adding a  `TXT vessel_name = motu` would help find a vessel your on the marina wifi :-), but not mandatory.

So currently for the three protocols:
MUST:
```
Service Type: _signalk-http._tcp
Service Name: signalk-http 
Domain Name: local
Interface: wlan0 IPv4
Address: 10-1-1-64.local/10.1.1.64:8080
```
May:
```
TXT vessel_uuid = urn:mrn:signalk:uuid:c0d79334-4e25-4245-8892-54e8ccc8021d
TXT vessel_name = motu
```

rob42, 2016-01-07:
Oh and a note on how to get the other info via a signalk request

rob42, 2016-01-07:
What about the roles https://github.com/SignalK/specification/issues/99#issuecomment-158863742 ?

tkurki, 2016-01-07:
The proper mechanism for distinguishing servers available via mdns discovery is Service Instance Name. Think "Upstairs Printer" and "Lobby Printer" - this is not TXT but Instance Name. See the example MySignalk above.

To me the short paragraph at the bottom of apis and conventions is enough, as it actually has the information needed for advertising the service. The TXT stuff adds confusion imho, witness this and other discussions. If you feel that it should be expanded I think a reference to the rfc would be great and a short explanation is probably in order.

I guess TXT is ok for making the role available already in the discovery phase. A client could connect to each server as well and extract this info from the hello message.  I just realised that the sensible default here is that in the absence of role information the default is `master, main`.

sumps, 2016-01-07:
I can accept not including this "Roles" issue for V1.
Are we definitely going to include the Signal K version supported by the device in the Bonjour/mDNS record ? I note in Robs initial post (above) that he included a TXT entry for Path that included the SK version number, but does it not make sense to just include the SK version number and let the consumer/server work out the URL as it is consistent for each service type and some services do not have a path i.e. TCP.
Also having the SK version number and linking this to the Bonjour/mDNS format version means that we can expand the Bonjour/mDNS record format in the future.

tkurki, 2016-01-24:
@sumps at http://signalk.org/developers/apis_and_conventions.html we specify that the rest endpoint at `/signalk` must provide information about the versions that the server supports. Thus a client should discover the version information there. This way the version information is available to both clients that do and do not support mDNS. SSDP discovery has also been mentioned.

Combine Murphy's law with several ways to do one thing and you get discrepancies.

If you want to include version information in the TXT record you can certainly do so, but I think this should _not_ be included in the Signal K spec and we should stick to the bare minimum.

sumps, 2016-01-25:
OK - I understand and will make sure we follow this procedure, which I will document for our own purposes and also post here so that we can use it for the signalk.org site.

sumps, 2016-01-25:
Here is the procedure....
[image]

tkurki, 2016-04-20:
See https://github.com/SignalK/signalk-server-node/issues/80

rob42, 2016-06-29:
We have some better notes on http://signalk.org/developers/apis_and_conventions.html (at least after https://github.com/SignalK/signalk.github.io/pull/32 is merged). If thats enough we can close now?

timmathews, 2016-12-08:
PR #300 addressed all of this in the documentation (I think). Closing.

closed 2016-12-08 by timmathews