SignalK/specification #147: Tcp service - raw json over a tcp connection

rob42, 2016-01-05:
In #146 a new signalk service for raw jsom messaging over tcp is added ```signalk-tcp._tcp```. This is pretty simple and easy to define, we just make a tcp connection, and tx/rx signalk messages as json text. There are a couple of issues that need definition:

1) the json text MUST be UTF-8.
2) session handling. Server should assign a session id per connection, implementation specific.
3) security + authentication - should do https, could do certs, what about user/login - how is that managed
2) message delimiter - do we need a message delimiter or is it implicit in the json format. Ive used a state engine in arduino - it tracks where in a streaming message it currently is, so it knows where start and end is. Quite simple and fast. But for a currupted message or poorly formatted json we may need a RESET message?

pod909, 2016-10-24:
On 4) to enable SignalK delta to coexist on a stream with NMEA0183 how about using \r\n as a record delimiter? Slightly limiting in terms of JSON but anything else and it would break it completely.

rob42, 2016-10-24:
Simple \r\n is what we use currently, since signalk json doesnt use \r\n in values (so far anyway) it works quite well. It also assumes that the json is not pretty printed, eg all on one line.  In theory we could also have \r\n in descriptions or some other fields, which would break the current scheme.
Tracking the json format (first { to last }) would fix that, but seems quite complex.
I am more inclined to make the start/end of a message quite distinct, for easy parsing. eg
`!{{....}}` or similar. Also easy to wrap/unwrap, could add a checksum in too !{{....}0aBf}

rob42, 2016-10-24:
...only for serial BTW...no need for `{{..}}` on packet or message based transports.

tkurki, 2016-10-24:
>Simple \r\n is what we use currently
Do we have that documented or where is it stated?

Control characters are not allowed in JSON in unescaped form so they are not a problem. Pretty printing is not something to be considered here.

I think this issue should result in an addition to the Signal K documentation: a retitled http://signalk.org/specification/master/streaming_api.html with tcp transport added, as well as additions to http://signalk.org/specification/master/connection.html.

tkurki, 2016-10-24:
The doc sources are at https://github.com/SignalK/specification/tree/master/gitbook-docs

pod909, 2016-10-25:
> The doc sources are at https://github.com/SignalK/
> specification/tree/master/gitbook-docs

TCP already handles checksum and rebroadcast so no need to add.

... Serial is really the territory of a low bandwidth binary protocol (?).
Tend to feel that the first step there is to not get into the transport
argument as there is no ubiquitous standard emerging for iot yet. A lot can
be done by standardising keys, units and type/size. Rob, I think you have a
lot to offer there with the work you've done...

Double brackets aren't very common. The { is the initial identifier of a
json string, how about something that possessively identifies what follows
such as {signalk-delta:{ ... }}

tkurki, 2016-10-25:
Isn't just defining a message delimiter as  \r\n enough? It can not occur in a valid json message's strings and if we assume no pretty printing, single line json what more do we need?

rob42, 2016-10-25:
Just had a good look at json.org, and I think you are right that the control chars \r\n cant occur in the json. The _string_ "\r\n" can, but not the chars [CR][LF], aka decimal [13][10]. So if we specify no pretty-print, then [CR][LF] makes a good delimiter, and works for NMEA0183 too. 
That means no `{{...}}` needed. Maybe add optional checksum keys, `crc` for serial, `md5` for message validity for all transports?

tkurki, 2016-10-25:
TCP already handles integrity so I don't think checksum is needed here as @pod909 said above.

If we want to specify checksum for serial transport I don't think it is a good idea to include it in the json payload (as I understand you suggested with the word _key_ above), as it is a bit cumbersome and error prone to add the checksum in the middle of the data you are checksumming. For that I would go for `{....}crc\r\n` or `crc{....}\r\n`.

rob42, 2016-10-25:
I only meant the crc checksum for serial transports. They often do need some sort of integrity check. 
The md5 checksum was a thought that adding a checksum might prove the message was untampered since its originator sent it. I know ssl does this, but only point-to-point. An md5 would prove the message was verbatim from me, even though it passed through severral servers?
Anyway thats of topic here. 

`{...}crc\r\n` gets my vote for serial, and the `crc` is optional. CRC-16 or CRC-32?. 16 is two extra chars, 32 is 4. 

We should name a specific std ...there are quite a few https://en.wikipedia.org/wiki/Cyclic_redundancy_check#Standards_and_common_use :-(

pod909, 2016-10-26:
> Standards_and_common_use :-(

Straight CRC-32 looks like the most ubiquitous. I'd plump for that.

Compression? How about LZW?

pod909, 2016-10-26:
Work for me in terms of SK over TCP/Websocket.

Serial: we have to be careful here that the delimited sentence is still
valid json. Perhaps something more explicit like a "checksum":1234}
attribute as part of the json it's self?

tkurki, 2016-10-31:
I think it would be a good idea to provide the same self message over tcp as we do over ws. If nothing else then parity, and it provides the self id.

tkurki, 2016-10-31:
@pod909 what are we checksumming here? the JSON message string representation? If we are checksumming the JSON message string then we can not alter it by adding the checksum, because then we are altering the string by inserting data in the middle of it and that will change the checksum of the string.

Naturally we can specify very specifically how the checksum is inserted into the string, but keeping the checksum totally external to the data over which it is calculated is way more foolproof. 

Checksum verification should happen before JSON parsing, because the checksum is about the validity of the string, not the parsed object's data.

pod909, 2016-10-31:
Agree, bad idea from me

afischerdev, 2016-11-01:
One more issue for the tcp connection:

* a hello message for the first contact like found in the ws connection
  see http://signalk.org/specification/master/streaming_api.html

In my current test scene I have the problem all update messages have a context like 'vessels.urn:mrn:imo:mmsi:xxx'. And one of the boats is the own ship. So it would be quite nice to find the self in start connection.

tkurki, 2016-11-01:
I think we are mostly set here in terms of design. Now this should be documented as part of the docs.

I would be happy to help, but it would be really really great if somebody else took the lead on this.

tkurki, 2016-11-01:
I don't understand the need for session in the original issue. Unless a clear need surfaces I think we should proceed without it.

pod909, 2016-11-01:
Happy to take the lead on updating the doc. Just take me though it step by
step first time round

tkurki, 2016-11-01:
- fork (create your own copy of the Signal K specification on Github)
- make the changes to your own copy to a branch
- create a Pull Request asking to review and pull your changes as addition to the official SK spec

See http://stackoverflow.com/a/14680805 

TODO (suggestions, I think at least these should be altered):
- change the index entry from "Streaming WebSocket API" to "Streaming API" [SUMMARY.md](https://github.com/SignalK/specification/blob/master/gitbook-docs/SUMMARY.md)
- modify [streaming.md](https://github.com/SignalK/specification/blob/master/gitbook-docs/streaming_api.md) to cover also TCP as discussed here 
- add TCP mdns to [connection.md](https://github.com/SignalK/specification/blob/master/gitbook-docs/connection.md), see also https://github.com/SignalK/specification/issues/146#issuecomment-169156907

and then it would not be bad idea to read the whole docs over to see what else should be updated.

Let me know how things go. Slack is also a great channel to ask for more specific advice.

rob42, 2019-03-27:
Closing due to age

closed 2019-03-27 by rob42