SignalK/specification #163: Satellites in view

tkurki, 2016-01-31:
Extend schema to support satellites in view information as available in GSV and PGN 129540

tkurki, 2016-01-31:
https://github.com/SignalK/n2k-signalk/issues/12

faceless2, 2016-02-12:
I'll start the ball rolling then:

```
"gnss": {
    "satellites": {
        "constellation": {
            "gps": {
                "1": {
                    "elevation": 85,
                    "azimuth": 24,
                    "distance": /*not sure this is useful */,
                    "snr": -76,
                    "active": true      /* whether the satellite is being used for a fix */
                },
                /* keys continue, key being the PRN of each satellite */
            },
            /* keys continue, being other constellations of satellite eg GLONASS */
        }
    },
}
```

Justification:

1. Keying the satellites by PRN makes updating the model easier, and no arrays are required.
2. An "active" field means you can combine the GSV and GSA NMEA0183 messages into one - these are all the satellites in view, and some of them are also in use (active).
3. I'm not familiar with the N2K PGN referenced so would need clarification on the meaning of "Range Residuals" is and what values of "Status" are other than "Used". If it's just "Used or "Not Used' then the "active" field would cover it.

tkurki, 2016-02-14:
Is there something that should appear as a sibling of constellation? Why should it be there?

Other than that I'm good with this - can you take it further?

I have no in depth knowledge of the 129540 details either.

faceless2, 2016-02-14:
You're correct - either "constellation" or "satellites" is redundant and could be removed. "satellites" is more immediately intuitive, all else being equal I'd suggest we keep that one.

By "take it further" you're going to make me learn to use Git and submit a pull request aren't you? Argh. I knew this day would come, but I have been putting it off :-) Will try to get up to speed on this and on the spec structure.

sumps, 2016-02-15:
Its easier than you think, I had never used it and when I met the guys at METs I promised I would have a go. Once I sat down and started, I found the GitHub part was easy and the only thing holding me back was my knowledge (zero) of JSON. 
You do not need to clone a repository locally on your PC, just start by going to one of the files in the Master branch that you want to change and once you are looking at the contents of the file just click on the Edit (pencil) icon. This allows you to immediately start editing the file and automatically creates an auto-named branch of Master with just the file you are editing. Once you have done the change, you click to create a PR, add your notes and justification for the change and post it. Then others will review and merge it for you. Go on give it a go !
[image]

faceless2, 2016-03-07:
An update based on comments above and also some data derived from @timmathews original at http://signalk.org/api/gnss-field-guide.html which I've just seen.

"gnss": {
    "satellites": {
        "gps": {
            "1": {
                "elevation": 85, /*elevation in degrees */
                "azimuth": 24,  /* azimuth in degrees */
                "range": 123123123, /* distance in meters from centre of ellipsoid, optional */
                "snr": -76,
                "active": true      /* whether the satellite is being used for a fix */
                "differential": false   /* whether fix on this satellite is differential. default is false */
                },
                /* keys continue, key being the PRN of each satellite */
            },
            /* keys continue, being other constellations of satellite eg GLONASS */
        }
    },
}

Changes were adding range in meters - definitely useful as with elevation/azimuth it allows us to convert between other representations of position, e.g. ECEF. And adding "differential", for a differential fix. This field plus "active" would allow us to represent all the options from PGN 129540:9..+7 - well, except for "Not tracked" which I don't really understand. There is no analogy for this status in NMEA0183.

canboat, 2016-03-07:
elevation and azimuth need to be in radians?

SNR is a ratio, but then we lose the logarithmic nature so should we maybe add a dB unit?

tkurki, 2021-01-02:
Need to account for
- non-GPS GNSSs https://en.wikipedia.org/wiki/Satellite_navigation#Global_navigation_satellite_systems
- a single device / data source using multiple systems
- multiple devices
The examples in this issue predate multiple sources implementation.

Today I would rather start from deltas than full, as that maps the real world incoming data better to message granularity.

Also need to consider linking a position update to the underlying fix data. Is this just by coincidental timestamps or is some other mechanism possible?

tkurki, 2021-01-03:
A log snippet with multiple GSVs and GSAs
```
$GNGGA,111103.500,5054.3949,N,00107.7952,W,1,12,0.5,50.8,M,46.7,M,,0000*64
$GPGSV,3,1,09,07,16,321,37,08,29,281,33,10,29,143,35,16,75,216,35,0*6E
$GPGSV,3,2,09,18,38,057,35,20,44,105,40,21,81,117,33,26,43,164,25,0*63
$GPGSV,3,3,09,27,62,289,41,0*5B
$GLGSV,3,1,10,65,14,112,18,71,15,018,11,72,25,069,31,77,10,181,30,0*79
$GLGSV,3,2,10,78,52,221,38,79,44,310,28,80,00,342,,81,35,261,40,0*7E
$GLGSV,3,3,10,87,41,052,31,88,75,350,33,0*73
$GAGSV,2,1,07,01,37,308,33,03,09,074,35,05,07,025,,13,85,237,31,0*7F
$GAGSV,2,2,07,15,39,060,33,21,63,228,39,26,30,239,40,0*44
$GNGLL,5054.3949,N,00107.7952,W,111103.500,A,A*5F
$GNZDA,111103.500,08,07,2020,,*41
$GNRMC,111103.600,A,5054.3949,N,00107.7952,W,0.02,0.00,080720,,,A,V*1E
$GNVTG,0.00,T,,M,0.02,N,0.0,K,A*11
$GNGGA,111103.600,5054.3949,N,00107.7952,W,1,12,0.5,50.8,M,46.7,M,,0000*67
$GNGSA,A,3,18,27,16,20,21,08,26,10,07,,,,0.9,0.5,0.8,1*35
$GNGSA,A,3,78,81,87,79,65,88,71,72,77,,,,0.9,0.5,0.8,2*31
$GNGSA,A,3,03,21,26,01,15,13,,,,,,,0.9,0.5,0.8,3*34
```

tkurki, 2021-01-03:
Note the talker ID to distinguish between the systems:

https://www.u-blox.com/en/docs/UBX-13003221
[image]

Hakansv, 2021-01-08:
@tkurki 
Although a JacaScript novice I've tried to add the satellite constellations. My first attempt gives the result as of the pict below.
See also (mostly up) http://signalk.stupan.se:3000/signalk/v1/api/vessels/self/navigation/gnss
The code is here: https://github.com/Hakansv/nmea0183-signalk/blob/addGSV/hooks/GSV.js
I divided the constellation path. One for each GSV message. I can't say I'm fully satisfied with the outlook. Any advise to code in another way is appreciated. Like the "satelliteConst_msg1"
But so far all data needed for OCPN are there and would be possible to catch but as mentioned above there is more in life than O. (?)
Thanks
Håkan

[image]

Hakansv, 2021-01-09:
Another way to name the constellation path:
Views?
Håkan
```
navigation:
	gnss:
		satellites:
			value:			        10
		satConstellation:part:1
			value:
				satNr:                  "2, 26, 29, 2"
				elevation:              "0.7165, 0.2618, 0.2259, 0.8356"
				azimut:                 "4.756, 0.5061, 0.2567, 2.735"
				SNR:	                "29, 31, 35, 24"
		satConstellation:part:2
		satConstellation:part:3
		satConstellation:part:x(<=9)
```

Hakansv, 2021-01-09:
The gnss type and satellites used for fix calculation is present in xxGSA.
So I intend to add also that sentence.
Questions:
- Is the path "type" intended to be used as of below? (Or would I use: "gnss/methodQuality" )
- I didn't found any path for used satellites in the spec. Please advice what to use ( "usedIDforFix"). 
- Pls advice where i would add the meta description for a new path.
See below extract from my tests.  (and:  [My SK ](http://signalk.stupan.se:3000/signalk/v1/api/vessels/self/navigation/gnss) )

```
type:   
   meta:    
      description:       "Fix type"
         value:          "3D"
         $source:        "blueconv.GP"
         timestamp:      "2021-01-09T15:18:30.153Z"
         sentence:       "GSA"
usedIDforFix:   
   meta:                 {}
         value:          "24, 17, 19, 12, 15, 25, 32"
         $source:        "blueconv.GP"
         timestamp:      "2021-01-09T15:18:30.153Z"
         sentence:       "GSA"
```

Hakansv, 2021-01-09:
Many Qs now...
I found a meta description file here:
/usr/lib/node_modules/signalk-server/node_modules/@signalk/signalk-schema/dist/keyswithmetadata.json
Is that the correct file? I saw that description was also included in files for "vessels" and not "self" as the above.
If so it's not included in the NMEA0183>SK project but has to PR to SK itself. Or?

For a test I used this meta descr. Would that fit? (The path itself would possible be changed I suppose.)
"usedIDforFix":{"meta":{"description":"Satellite IDs used for Fix calculation"},

Hakansv, 2021-01-09:
re: "Note the talker ID to distinguish between the systems:!" some posts above
Neither the NMEA0183 4.11 seems to message the gnss system. See [this](https://www.nmea.org/Assets/20190515%20nmea%200183%20gsv%20sentence%20corrections.pdf)
But, GPS satellites has satellite ID from 1 - 32. Higher IDs are from the other system so that will be the way to distinguish for now. See the above linked doc for details.

tkurki, 2021-01-10:
I don't think separate `constellation` elements 1-3 are a good idea. The division to several sentences/lines is just in 0183 data. In NMEA2000 it is a  [one](https://gist.github.com/tkurki/ae4836b44aa54e3311cc7cbae7d0773c) or [more](https://gist.github.com/tkurki/5066e090bae633ce1bdc7ae548b8e630) pgn 129540 messages. We just need to parse several GSV lines into a single delta, just like AIS messages are parsed.

Furthermore 129540 seems to have  also data about the use status of each satellite. I need to dig a little deeper and check also other data sets. One possibility is that in nmea0183-to-signalk parser we combine GSV and GSA sentences to a single delta, that is pretty close to PGN129540.

Thanks for the link to NMEA0138 Errata. I did not understand 
>Neither the NMEA0183 4.11 seems to message the gnss system
because the document repeats what I wrote above about talker id being significant.

That document also says _To reliably identify when the satellites in view were observed a corresponding transmission of the ZDA sentence is recommended._. So in effect the GNSS is saying "at time xxx I saw satellites a, b and c from system A and x, y and z from system Y and I used a and b for the fix.", but that is split to several sentences in 0183.

Mind if I take over from here? We need to make the parser stateful, so that it can combine several sentences.

Do you happen to have any data with tag blocks that work in the way described in the errata? Sounds like tag blocks are used to tie together the sentences into a single unit.

tkurki, 2021-01-10:
keyswithmetadata are generated from the specification schema files - imho we should first get this working in 0183 and n2k and only then add this to the schema.

tkurki, 2021-01-10:
Oh right, the one versus many PGNs is just an artefact of how the inputs that I am looking at are generated.

Hakansv, 2021-01-10:
- I'm pleased if you take over. Very good.
- Talker ID for gnss type: Yes that can for sure be used. What I discussed was a new field added for "GNSS system ID" but was inhibited in the document I refereed to.
- re: "Do you happen to have any data with tag blocks..." I'm not sure what you mean but the GSV's first message indicates how many messages there are.: 
Field Number:
    1) total number of GSV sentences to be transmitted in this group
    2) Sentence number, 1-9 of this GSV message within current group
And GPGSV is streaming on my SK server
But pls describe again if I misunderstood.

Hakansv, 2021-01-10:
re: " One possibility is that in nmea0183-to-signalk parser we combine GSV and GSA sentences to a single delta"
But don't rely on GSA is sent. I'd say it's seldom used. It depends on the GNSS device. It's not present from my onboard GPS.

tkurki, 2021-01-10:
tag block is a mechanism to put additional fields in the beginning of NMEA0183 data line, see https://github.com/SignalK/nmea0183-signalk/blob/master/test/tagblock.js#L24

You need to reread the bulletin, there are three parts
- _introduction_
- _Incorrect Sentence_  that seems to list how it used to be, with GNSS System ID in strikethrough
- Corrections, that says how it should be with `Signal ID`

tkurki, 2021-01-10:
Yep, may need to add an option to combine the sentences.

Hakansv, 2021-01-10:
I'll read and learn and look forward to study your solutions.
About PGN I've no particular experience from N2k yet. Not more than some test with Openplotter and SK. 
If needed I'll test when you need and also consider a OCPN parsing of the gnss data from SK-GSV. That was one of my goals with this.
Thanks
Håkan

Hakansv, 2021-01-10:
Just for info I think a single delta for all (<=12?) satellites will be manageable for OCPN Dashboard. I just have to group them up to four each, as the GSV does. The same instrument coding will be used both for SK and GSV data. But no problem.

larsth, 2021-06-26:
Regarding help wanted to create a schema for satellites in view,
A program called gpsd has JSON output that can easily be transpiled into something else, i.e. a SignalK message

You can look at my repositories and find them using a browser search using "gpsd" (without the "s of course)
My repositories : https://github.com/larsth?tab=repositories

FIY,
I had jumping into this conversation without reading it, becuase SignalK looks interesting.

s2tenacity, 2024-12-09:
I am not finding much information on the NMEA sentence $GNGSA other than this thread.

Despite listing GSA in the Data Connection - Ignored Sentences (GSA,VTG,RMC,GGA,GSV), it still shows in the data stream coming from my maiana AIS unit.  All I want is the  GLL and AIVDM sentences.  

[image]

1733756419119;N;$GNGLL,2713.1863,N,07819.3902,W,150018.000,A,D*59
1733756419220;N;!AIVDM,1,1,,B,15O:fRRP00rISqp?TWPv4?vV2<03,0*37
1733756419457;N;
1733756419481;N;
1733756419570;N;
1733756419633;N;$GNGSA,A,3,10,32,31,25,26,03,16,02,28,21,,,1.02,0.70,0.73,1*0F
1733756419691;N;$GNGSA,A,3,73,71,80,69,79,,,,,,,,1.02,0.70,0.73,2*09

Any thoughts on why the GNGSA isn't filtered out? It's not a ton of data, but it bothers me I can't shut it off.....

On a related note, the data doesn't show in the data browser. It's coming in, but doesn't appear to be processed or displayed.

jrversteegh, 2024-12-12:
Silly question maybe, but why would one care about the satellite positions other than for quality of fix (hdop and so on..) ? I personally never look at the satellites except maybe out of general interest, which is not a good reason to bloat the SignalK schema?