SignalK/specification #164: Vessel/Ship Type

tkurki, 2016-01-31:
Support generically ship/vessel types and specifically AIS static data ship types.

tkurki, 2016-01-31:
```
{
  "timestamp": "2014-08-15-19:48:15.573",
  "prio": 6,
  "src": 43,
  "dst": 255,
  "pgn": 129794,
  "description": "AIS Class A Static and Voyage Related Data",
  "fields": {
    "Message ID": 5,
    "Repeat indicator": "Initial",
    "User ID": 258858000,
    "IMO number": 9391000,
    "Callsign": "LAKN6",
    "Name": "RYSTRAUM",
    "Type of ship": "Tanker hazard cat B",
    "Length": 129.0,
    "Beam": 20.0,
    "Position reference from Starboard": 5.0,
    "Position reference from Bow": 103.0,
    "ETA Date": "2014.08.17",
    "ETA Time": "10:00:00",
    "Draft": 7.50,
    "Destination": "STETTIN",
    "AIS version indicator": "ITU-R M.1371-3",
    "GNSS type": "GPS",
    "DTE": "available",
    "AIS Transceiver information": "Channel B VDL reception"
  }
}
```

faceless2, 2016-02-12:
#171

tkurki, 2017-10-07:
Solved by /vessels/<RegExp>/design/aisShipType

closed 2017-10-07 by tkurki