SignalK/specification #166: Great Circle and Rhumbline Support

sumps, 2016-02-01:
Currently we do not differentiate between Rhumbline and Great Circle calculations when we store Bearing and Distance to Next Waypoint data or Bearing from Origin to Next Waypoint data.

Checking the NMEA specs, there is some ambiguity related to Rhumbline and Great Circle calculations. In theory every bearing and distance between or to waypoints should be identified as Rhumbline or Great Circle. In reality, the only sentences in NMEA0183 that explicitly state their type are the BWC and BWR sentences. None of the other sentences BOD, APB and RMB state what calculation they are using and the assumption is Rhumbline.

In NMEA2000 the Navigation Data PGN 129284 has a specific flag that says the Calculation Type Rhumbline or Great Circle and another flag that says if the bearings are true or magnetic. The other PGN 129285 Navigation - Route/WP Information is not used as much and I would not try to support this for now.

All things considered, I would have two fields for Bearing to Next WP, distance to next WP and Bearing from Origin to Next WP one for Rhumbline and one for Great Circle and then we can populate whichever field we receive data for.

That is probably all we need to do to support Great Circle and Rhumbline calcs in SK, pretty much the same way as we handle True and Magnetic bearings

Consumers will need to be aware of these differences and display suitable labels so that it is clear that a waypoint bearing is True/Mag and RL/GC and a waypoint Distance is RL/GC.

rob42, 2016-02-02:
#167 will provide support for rhumbline (default) and great circle values for heading and distance. Ive put the great circle data into a `greatCircle` object to keep things cleaner.

rob42, 2016-02-03:
#167 is superceeded by #168 
Pls review

sumps, 2016-12-21:
Wow this was a long and complex issue, but looking through all of the issues that were referenced and superceeded, it appears to me that this issue is actually closed, but no one has closed this original issue.

We have the two navigation sub-groups of Rhumbline and GreatCircle in the latest schema and I think @rob42 also created the two arrival alarms, so I think we can close this issue now.

closed 2017-01-04 by tkurki