SignalK/specification #188: Sails group

joabakk, 2016-03-05:
great work so far!

I have a use case:
I want to make the most out of my polar diagrams, to improve racing speed. Say I have sailed in similar conditions before, I want to know what sails I had set, and if they were reduced (furled or reefed)

In propulsion, we have each engine witth a state. This could also apply for each sail ( available, set and damaged/unavailable). Also, the sail area, the reduction by reefing or furling, and perhaps ven the shape, in a few years time, we might all be doing this on the cheap http://www.sailingworld.com/gear/new-science-sail-trim

tkurki, 2016-03-08:
See https://github.com/SignalK/specification/blob/master/schemas/groups/sails.json#L43

No way to set reefing status though.

joabakk, 2016-03-08:
Thanks. If this is something we want, this can either be done as number of reefs set or a ratio set (includes furling also), or simply boolean (reduced=true or false)

- number of reefs set is simple and user friendly, but does nothing for furling
- ratio is more correct, but requires a system for measuring luffs or the like, and would need a lookup table for reefable sails
- reduced boolean is simple and user friendly, and indicates that the resulting polar diagrams will be less trustworthy

closed 2018-06-18 by joabakk