SignalK/specification #203: Signal K to control stereo etc?

joabakk, 2016-03-23:
I vaguely remember having seen n2k stereos that can be controlled from mfd's. Do we have a strategy for where signal k will stop, or should we add provisions for such functions? 
On one side, it is difficult to imagine all the possible use cases (stereo, video, lights, thrusters etc) but on the other hand, it is relatively simple to include some and pave the way for a future standard. 
In a way, we are telling app coders and equipment manufacturers what could be possible using only signal k. Just a thought

canboat, 2016-03-23:
That's right. Fusion started this, but there are now multiple consumers and producers for this data over N2K, and the sentences (PGNs) have been standardised as part of N2K.

So I guess, yes, this will eventually become part of Signal K.

joabakk, 2017-01-19:
Closing this here as #293 is more extensive

closed 2017-01-19 by joabakk