SignalK/specification #205: apt-get install mdns fails to build

CaptainRon47, 2016-03-28:
I got down to the point in your Wiki tutorial for install the node server on the Raspberry Pi 3 where you install the Bonjour service - mdns
The command failed on a build err
../src/mdns.hpp:32:20 fatal error: dns_sd.h: no such file or directory

I placed this issue on the mdns page as well, but not sure if the issue is there or here?
there is a debug log file but I am new and have not figured out how to get a file from the pi to the pc.

Thanks for any help you can provide.

Ron

tkurki, 2016-03-28:
Thanks for the error report.

The command that failed was probably `npm install mdns` and not the one you mention in this issue title?

At least with Jessie Lite you need to do  `sudo apt-get install libavahi-compat-libdnssd-dev` as mentioned in https://github.com/agnat/node_mdns#installation.

@sumps update the RPi tutorial to include this nugget?

CaptainRon47, 2016-03-28:
Teppo,

It was my first post on Github of an issue and I was not sure where to put it (or start a new one, since then I have).  You are right it was the “npm install mdns” command that failed.  In the error code it said it failed because it could not find the file dns_sd.h.    I looked in the code for mdns.hpp and saw that down at the bottom it does reference that file and that file does not exist with that extension in your directory.  So I assumed that was the issue.  

 

Previously I loaded these libraries     sudo apt-get install libnss-mdns avahi-utils    after doing so it indicated

The following packages were automatically installed and are no longer required:

  libc-ares2 libv8-3.14.5

Use 'apt-get autoremove' to remove them.

 

So I removed them, I thought this might be the problem, so I reinstalled them, but that did not help.

Question, I assume I don’t really need them and  I can just run   ‘apt-get autoremove’ to remove them again ?

 

I eventually (it was late or early in the morning depending on where you are), figured I could just use a USB memory stick to copy a file from the Pi to the PC.   So I am attaching the debug file from the ‘npm install mdns’ command.

 

Thanks, I will try your instructions and see if that loads the required packages.

 

Thanks for the quick response,

Ron

tkurki, 2016-03-29:
Added libavahi-compat-libdnssd-dev to the libraries to be installed at https://github.com/SignalK/specification/wiki/Raspberry-Pi-Installation-(Node-Server)

closed 2016-03-29 by tkurki