SignalK/specification #207: Range as sensor value

joabakk, 2016-03-29:
Some sensors such as the gobius tank sensors (gobius.se) will indicate only if a tank is almost empty, half full and/or almost full. How do we represent this in signal k?
With the current schema, we will have to round up, down or pick a level in between. Perhaps not a big issue, but would it make sense to allow some values to represent a range? From the top of my head, I cannot think of other examples where this makes sense or is necessary

timmathews, 2016-05-17:
Does #222 solve this? It adds tank level as 0-100% which should cover this use case, no?

joabakk, 2016-08-05:
I'm sorry for the long wait, but no. #222 only allows for a specific level. If I have 1 or 4 binary sensors, I will not know the exact level, only that it is not full (in the case of 1 sensor) or that the level is eg. between 25 and 50% (in the case of 4 sensors)

sumps, 2016-08-05:
It does cover this, the controller that is monitoring the binary sensors and producing the Signal K data just has to output 100%, 75%, 50%, 25%, 0% it is the only way to have a consistent and logical data value.

joabakk, 2016-08-06:
Ok thanks. I agree that it needs to be consistent and logical. My objective was to make sure we cover realistic use cases.

closed 2016-08-06 by joabakk