SignalK/specification #210: Is `self` really required?

timmathews, 2016-04-10:
Currently, we specify that the `self` property for the top level Signal K object is required. However, there are instances where this may not make sense. In fact, most of our tests are currently written without an explicit `self` and this causes them to fail when the required-ness of `self` is specified via the `required` array on the top level object instead of as a boolean attribute of itself (tv4 assumes draft 4, so ignores the current `required` attribute).

Check out the [draft04](https://github.com/SignalK/specification/tree/draft04) and run the tests to see.

tkurki, 2016-04-11:
I've always thought of `self` more like an alias path in the REST api and not present in the full document.

That said the current Node server implementation only has the notion of `/vessels/self`, not `/self`.

So I would 
- remove self from schema
- add a more precise of how it works to Signal K specification

jboynes, 2016-04-11:
Also touched on in #150 for SK servers not associated with a vessel.

Using `/self` to mean "the physical location of the server" would allow it to redirect to `/vessels/{id}` for a vessel, or to some other URI for a server not on a vessel (e.g. at a ground station or company HQ). `/vessels/self` would be a more explicit alias for a server actually installed on a vessel and would avoid the need to configure the vessel id into devices installed on a vessel-local network.

For both options, as @tkurki says, this is really defining well-known URIs in the REST api and would not be in the data model. Presumably these URIs would only support GET operations and all updates would need to be made using the `/vessels/{id}' path.

timmathews, 2016-04-11:
OK, then I'll remove it from the list of required props for the top level SK object.

sumps, 2016-04-11:
So how does a simple consumer know what URL to call to get for instance the depth value ?

sumps, 2016-04-11:
At the moment anyone can work out what the URL is, does this mean additional programmatic effort?

tkurki, 2016-04-12:
This change has no effect on the urls.

tkurki, 2016-04-19:
@timmathews do you have this somewhere as work in progress? Are you removing it as a required property or alltogether from the schema?

timmathews, 2016-04-19:
It'll stay in the schema, just not required. I'll open a PR for the draft04 branch with this change tonight

tkurki, 2016-05-01:
How about cutting or tying this loose end.

timmathews, 2016-05-17:
PR #223 will close this

closed 2016-05-18 by timmathews