SignalK/specification #218: Add resources section

tkurki, 2016-05-01:
#211 specifies references to `resources` section containing waypoints & routes.

Usage outline in https://gist.github.com/tkurki/afd45cc48551320615f3c172c2231f40 and in https://gist.github.com/tkurki/afd45cc48551320615f3c172c2231f40.

timmathews, 2016-05-17:
Resources under a specific vessel? We have a resources section at the top level (equal to vessels).

rob42, 2016-06-29:
This needs some thought. Ive been implementing current tracks in freeboard-sk and storing them in `vessels.self.resources.routes...`. These contain geojson LineStrings up to 5000 elements. My original plan was to slice the geojson  out of incoming messages and store separately, with a uri to retrieve them, but it became complex, so I have put them back in the main model for now.

But when you subscribe to `vessels.self` by default, you get a whole lot of BIG messages :-(
I can avoid that by subscribing more selectively, but it shows that having the resources inside the vessels.self is not a great idea, although I think it was mine!

So I favour resources at the top level now.  Also if we have a default subscribe for `vessels.self` we need to ensure the default data volume doesnt get unreasonable.

tkurki, 2016-09-30:
Is this still relevant?

timmathews, 2016-09-30:
I don't think so. Vote to close.

closed 2016-09-30 by tkurki