SignalK/specification #220: SignalK install on RASPI B+ - nodejs segmentation fault

SailorMax59, 2016-05-15:
Hello Teppo,
Thanks for the very helpful tutorial. If not already described in another place, I would like to help other signalK users. Nodejs seems to be very picky regarding its version used on ARM, there is some discussion on that on the web. When following the tutorial, I was getting “segmentation fault” after installing nodejs and probing it with node –v. The cure was to install node-v4.4.4-linux-armv6l.tar.gz from www.nodejs.org/downloads. The node-v4.4.4-linux-armv7l.tar.gz did also produce the “segmentation fault” error.

Regards Matthias

Raspi B+ V1.2 (2014)
Raspian Jessie

canboat, 2016-05-15:
Matthias, an original RPi is ARM v6, not v7, so that makes full sense that an ARM v7 compiled version doesn't work for you.

tkurki, 2016-05-15:
@SailorMax59  which tutorial did you use? The wiki tutorials you can update yourself by adding instructions for ARM v6 node installation.

FYI there are Ansible tasks for installing current Node on RPi, see https://github.com/tkurki/marinepi-provisioning/tree/master/roles/node/tasks

I rewrote the issue title to better reflect what needs to be done.

SailorMax59, 2016-05-16:
Hello Teppo,

I used the github tutorial at https://github.com/SignalK/specification/wiki/Raspberry-Pi-Installation-(Node-Server).

 

Thanks for your quick response !!!

 

Matthias

closed 2017-10-07 by tkurki