SignalK/specification #236: Pushing signalk data from the server

rob42, 2016-07-28:
Currently all signalk data is transferred by pull methods. The source device advertises itself (mDns) and the receiver finds, connects and receives the data. 

Another use case is push based. If I am on my vessel, and want to share a data with a cloud service, how do I initiate that?

Assuming the cloud service is passive, and I want to push my position to it every 10 minutes, how do I execute what amounts to a subcription in reverse, eg make the cloud service subscribe to my vessel? 

Currently I can send via a REST PUT, but I cannot make the remote server connect to my vessel by websocket, and I must explicitly subscribe to the remote server for incoming data separately.

We need a well-defined process for connecting based on the push scenario

rob42, 2016-07-28:
Hmm, maybe this is actually a server implementation issue. I can open a websocket connection to the remote server, and get self.* by default subscription. 

I am then free to send what I want, whether its requested or not. Its up to the receiver to ignore or reject it. No explicit subscription is required, so I can send my position every 10 min, and the remote server can choose to record it, which the cloud service would do.

So to do this my server implementation must be capable of a client connection and subscription to a remote service (which it is) and defining data to _send_ to the remote service (which it isnt).

Maybe all thats needed is to also define _push_ filters when configuring a client connection?

tkurki, 2016-07-30:
Words like client, server and push and pull become somewhat overloaded or ambiguous with duplex connections and servers talking to each other.

If we just think about the ws connection ws client and ws server are well defined.

The ws client can open the connection with some well defined subscriptions specified in the url. In addition it can send subscription commands to the ws server.

If we allow the subscription protocol to be symmetric so that it works both ways the ws server could also issue subscription commands. 

For symmetry's sake we might want to add a `hello` message from the ws client to the server as well so that the ws client can tell what version of Signal K and what features it supports.

The ws client could also just start pushing all the data to the server or with a predefined set of subscriptions. Modeling these like Signal K subscriptions sounds pretty natural to me. In effect you could define a connection to say `vessels.signalk.org`, add credentials for authentication and associate a set of subscriptions with the connection. The "local" Signal K server would act as a ws client, connecting to the "cloud" Signal K server, handling the subscriptions internally and sending the resulting delta stream to the "cloud" Signal K server.

I think it would be a good idea to figure out a handshake here, where the ws client waits for the `hello` message from the server which may or may not include a set of subscription commands. This would allow the "cloud" server to specify what data it is prepared to accept and at what rate or leave that decision to the "local" server.

rob42, 2016-07-30:
We have SUBSCRIBE, do we need OFFER?

tkurki, 2016-07-30:
Maybe? Let's build vesseldata.signalk.org and find out?

itemir, 2016-10-05:
AISHub for Signal K? Can be an interesting starting point.

tkurki, 2016-10-05:
MarineTraffic input is 0183 position data over UDP. This summer I used https://gist.github.com/tkurki/b80610a9dcb9541e795dcc1f098f4231

(it is missing some stuff that I will include/rewrite with https://github.com/SignalK/signalk-server-node/pull/105)

rob42, 2019-03-27:
Closing due to age

closed 2019-03-27 by rob42