SignalK/specification #247: Passthrough (binary) data

tkurki, 2016-09-06:
There are valid cases where we would want to be able to transmit data from a source connected to a Signal K gateway/server to the clients in uninterpreted format.

These include
* XDR NMEA0183 sentences not understood by the server
* proprietary N2K sentences
* binary data, such as image data

tkurki, 2016-09-06:
Something like this for n2k?

```
{
    "context": "...",
    "updates": [{
        "source": {
            "type": "NMEA2000",
            "pgn": "130919",
            "src": 
            "label": "n2k-1",
        },
        "timestamp": "2016-09-06T19:52:44.000Z",
        "data": {
          "encoding" : "hex",
          "value" :   "3b,9f,03,02,04,5a,a1,12,19,09,09"
        }
    }]
}
```

jboynes, 2016-11-07:
Base64 is a very common encoding used for binary data in JSON and more efficient than the simple hex format above.

Do we need to allow options for how binary is encoded?

tkurki, 2016-11-07:
+1 on Base64. I thought that the `encoding` would be simple way to stay future proof, even with one encoding.