SignalK/specification #248: Extend Environment Schema to include user-defined temperatures

mark-tilden, 2016-09-12:
I suggest that the temperature elements (under the "internal" and "external" elements) be extended to include "user defined" temperatures, not just the fixed "engine room", "main cabin", "refrigerator", etc. There are two many temperatures (e.g. "Alternator temperature", "Hydraulic oil temperature, etc..) for the specification to anticipate all the possibilities.

User defined temperatures are defined in NMEA 2000 PGNs.

tkurki, 2016-09-12:
Your concern is really valid, but at this stage I think we should [define generic way to extend the SK schema](https://github.com/SignalK/specification/issues/206) rather than think about just one area.

I'll close this issue for now, let's continue there.

closed 2016-09-12 by tkurki

mark-tilden, 2016-09-12:
Thanks, Teppo. I'm still pretty new to signal-K and would actually be happy
to get more involved, but my work currently doesn't leave me a lot of free
time.

I love the concept of signal-K (and am enthusiastic about the iKommunicate
gateway). My *primary* objective is to expose the large amount of data on
my N2K network, primarily gathered from Maretron sensors. Thus, the
interest in supporting the Maretron sensors for switch closures,
temperatures, etc..

I'm happy to even make proposals for the extensions, but I want to respect
the process you guys are using to develop this important spec.

Mark

tkurki, 2016-09-12:
You can help by documenting the real world cases that you have and by including real world data samples, in N2K world with actisense_serial output for example.

mark-tilden, 2016-09-12:
Hi Teppo:

I'm happy to describe the real-world N2K network I have and the data I want
to capture. Unfortunately, I don't have an actisense serial gateway on the
network.....

I'm curious how they handle new/custom PGNs. Do they just attempt to
convert ALL PGNs to 0183-style sentences? Do they have a defined set of
PGNs/sentences they understand and translate?

Mark

tkurki, 2016-09-12:
NGT-1 gives you access to the raw n2k bus data so it does not need to be aware of individual pgn structure afaik.

mark-tilden, 2016-09-12:
Interesting....I will have to dig into that some more...