SignalK/specification #258: Missing Units in Doc

tkurki, 2016-09-20:
For some keys schema has units but it is missing in docs.

https://github.com/SignalK/specification/blob/master/schemas/groups/navigation.json#L373-L382

http://signalk.org/specification/master/keys/#vesselsregexpnavigationheadingmagnetic

joabakk, 2016-09-20:
I also noticed electrical.batteries.0.voltage showed up without description or unit

fabdrol, 2016-09-20:
I'll have a look when I get the chance

tkurki, 2016-09-20:
When you have a reference it replaces all the siblings at the same level, so if you have 

```
    "headingMagnetic": {
      "$ref": "../definitions.json#/definitions/numberValue",
      "description": "Current magnetic heading of the vessel",
      "units": "rad"
    },
```
`description` and `units` are never handled: https://github.com/SignalK/specification/blob/ccbf79301cf01c61bf0b68cc601f782b8797d0b0/scripts/processSchemaFiles.js#L348-L352

tkurki, 2016-09-20:
@joabakk there is no `electrical.batteries.0.voltage` in the doc at http://signalk.org/specification/master/keys/ ???

joabakk, 2016-09-20:
Ok it may be an implementation fault on my side. What would battery voltages' addresses be?

This is strictly rs11 -> canboat -> signal k

timmathews, 2016-09-20:
`electrical.batteries.0.voltage` exists, it's just another victim of the `$ref` overwriting siblings.

commit f6cae531 (2016-09-30): Fix  overwrites, ignore allOfs

Fixes #258
Do not overwrite properties with values from resolved . Treat explicit
values in the schema as overriding any coming from .
Values from allOf were overwriting schema properties as well. Judging
from the diff against the previous keys/index.md this version produces
better results.
closed 2016-09-30 by tkurki