SignalK/specification #263: Electrical:  acBus definition - inclusing of acQualities

thomasonw, 2016-09-28:
There is currently a entity called 'ac' in the definition - with the description that it is the 'AC Bus'.  It primarily includes acQualities (voltage, current, etc).   I offer that in its current form it is a bit problematic.

The 1st problem I see is that while the concept of Bus is good, associating directly a property like measured voltage is not.  Voltage is measured at a point, and in the case of a bus, the resulting voltage measurement can very greatly depending on IR drops.  So, if SignalK reports out the 'AC Bus Voltage is 117v', where is that in reference to?

It is more correct that voltages associated with a bus are reported out by devices.  An inverter can report the AC Bus voltage as measured at its point of attachment.  An isolation transform the same, perhaps a switch too.  Or, one could add a 'Voltage Meter' device/object.

Another problem with AC is that it creates a circular reference:  It has declared its self as the 'AC Bus', and  calls for acQualities, which includes 'associatedBus' - presumably pointing back to its self...

I propose that the current 'AC' definition should be removed and replaces with a 'electricalBus' definition.  In that we can add basic defining quantities such as:
  -nonomialVoltage
  -maxCurrent
  -numPhases
  -hz  (DC, 50, 60, 400, etc)

It would be expected that dcQualities and acQualities would reference one of these bus entries via their assocatedBus string.  And that for real-time measurements, those would be associated with the actual device that took the measurement.

canboat, 2016-09-28:
The names ‘AC bus’ etc are named like that because they are what J1939 PGNs use. 

I think you should read this as ‘bus voltage at the device that reports it’.

thomasonw, 2016-09-28:
OK, I can see that.  Then this should be renamed to 'acMeter', not 'Bus'.  or perhaps acBusMeter

And brings up question:   Is there a need to add proper 'bus' definitions?  One that can be referenced by acQuanities and dcQuanities?  (Could be a post v1 topic - but I do think the 'ac' definition needs to be clarified)

thomasonw, 2016-09-28:
"... I think you should read this as ‘bus voltage at the device that reports it’."

THAT concept is already covered in each device via their inclusion of acQualities / dcQualities.

tkurki, 2016-09-28:
>Voltage is measured at a point, and in the case of a bus, the resulting voltage measurement can very greatly depending on IR drops.

If you compare voltage to practically any other measurement is there any real difference? DC voltage depends very much on connections & instantaneous current and you get different readings depending on where in the circuit you are measuring. `navigation.position` is different depending on where your gps receiver is, or if there are several you have multiple values (I have 2, about 5 meters apart).

Furthermore all the values originating from sensors (unlike for example `navigation.state` for self) are just that, values from sensors, and the Signal K source concept tells which sensor/source produced the value. If SignalK reports out that _AC Bus Voltage is 117v_ it includes the identity of the source for that data, both in delta and in full format. Why would we add `Voltage Meter` and not `Wind Meter`?

In essence the current `ac` element is a way to
- list all ac buses
- have a place to represent acQuantities independent of the only Well Known AC Device, inverter

You have a point in the self reference in ac bus, but that is very minor I think. None of the properties are required, so you can just leave that out.

The model already has frequency. If needed we can add `nominalVoltage`,`maximumCurrent` and `numberOfPhases` to the properties under ac bus.

tkurki, 2016-09-28:
Have you looked at https://github.com/SignalK/specification/blob/master/test/data/electrical.json#L160?

Sorry about the sources, they are just placeholders and the test example is not very complete or as real world like as I would like. Feel free to improve it.

The raison d'etre for the files in this directory is that there is a test for each one that asserts that the example is valid against the current schema. You can run the tests with `npm test (given that you have Node installed).

thomasonw, 2016-09-30:
Hey, am not done here!   Just other priorities for time, and spotty internet access.  Need to think on this some (esp the Navigation example) - and also need to consider DC as well as AC....      But for now, too many other distractions!

thomasonw, 2016-10-03:
@tkurki - good prompting of things to think about.   Looking at navigation.json vs., electrical.json it occurred to me there is a fundamental different approach to how information is structured.  navigation.json has what I will call a ‘Data’ centric approach, XTE, COG, Position, # Satellites;  while electrical.json used primary a ‘device’ centric approach:  Inverter, battery, charger.   Navigation.json makes it easy to see what the  current SOG is, I just look at the SOG value  However, gaining  details on the device that produced it needs some digging.   If one wants to gain some confidence on the quality of the SOG value - they need to watch other data items, ala gnss.methodQuality, satellites, maybe some other things – and make the linkage via the source ID.    It is difficult.

On the other hand, getting some insight into a devices state of health is simple with the device centric structure, just monitor electrical.inverter and all the relevant details are there.  But simply wanting to know the AC voltages on the boat?  That takes digging – requiring one to look at ALL potential sources:  inverters, generator (tbd), etc..  Hence the data-centric entry ‘AC’ - that makes it simple again.   But this brings up a few NEW questions:

1. Primarily:  how should information get propagated into the two views?  If an inverter has a new AC voltage measurement, is it ITs responsibility to push out two SignalK sentences, one for electrical.inverter.ac.voltage, and one for electrical.ac.voltage?   Is there some way the SignalK server can handle these replications?  Perhaps RFC0001 comes into play - linking the two views of the same data? 

2. Should everything be data centric?  Should electrical.json be restructured into data views?  Instead of:  electrical.inverter.temperature, would have electrical.temperatures.inverter  Would get back to a single architectural approach for how information is represented.

3. If both views are kept,  maybe some guidelines for deciding when to place an information value only in the Device view vs. also the Data view…   Example, maybe GNS details (# sats, etc) could be placed under a (not yet defined) GPS device.   Just thinking might be good to have some thought into overall guidelines for schema’s.

4. Is there any though at doing value-add processing of information inside signalK?  Perhpas qualify data-centric values, prioritizing and selecting the ‘Best Representative Value’?   Specifically:  I have two NMEA connected GPS’s, one a proper GPS, and the other imbedded in the AIS transponder.   Both are connected to the chart plotter.  Somewhere there is a decision tree to select which one the chart plotter will use.  – extending this, if I had  a 2nd device monitoring progress, it too would need to decide which one to use.  AND if their approaches to qualify the data is not the same, those two devices will report different positions.   In the real world perhaps not an issue, but as a generalized example, there always is a need to qualify, prioritize, and select when multiple data values are presented – with anything.  But perhaps SignalK could do a qualification and present 'the best' source  via the data centeric view, of course ALL the information would still be available under the device views..  

Ok, enough for now.  I can now see the benefit of adding ‘AC’ as it is to the electrical schema (DC as well?).  But do want to get some conversations around my questions here. 

Couple of other quick notes:  Might be a good project to go over the navigation schema, assuring it has all the traditional navigation data values included.  And I see that the electrical.ID has redundancy with sourced.json, might be able to remove them (e.g., manufactures name, part number, location of device, etc.  are in both…)  AND seems an opportunity to add some device centric things in navigation (electronics.json??)

thomasonw, 2016-10-25:
Based on what I have seen recently, I have a question.  Assuming I have this correct one is able to access 'AC' bus voltages using the data centric approach along the lines of:

_electrical.ac.Bus1.1.lineNeutralVoltage_

And receive back a series of AC voltages.   Am I correct that one could get a like list by doing a wild-card query of the devices, ala:

_electrical.*.ac.lineNeutralVoltage_

====================================
And perhaps related, there seems to be an issue with how phases are handled.  acQualities has kind of a mixed approach, including lineLineVoltage, but not offering a way to define which line (or phase) is being referenced.  Let alone having _any_ phase value in it, as a result the inverter (one example) does not seem to have a way to indicate which phases it is associated with.