SignalK/specification #267: RCF 0002: Cleanup of Source.json scheama

thomasonw, 2016-09-30:
# Summary
[summary]: #summary

This RFC proposes edits to the specification.json schema, and associated test file.  The edits are to remove redundant entries in the current definition, as well as reorganization to place NMEA-2000 specific items in the N2K section,  and assure more general items are in the general section.

# Motivation
[motivation]: #motivation

Having a common set of source entries independent of the underlying technology is a benefit as additional technologies for sourcing data are added to SingalK.   From an overall structural standpoint it is appropriate to restrict technology specific entries to the applicable section.

it is expected this proposal will provide an extendable model as additional data source technologies are added.

# Detailed design
[design]: #detailed-design

Edits were made with two prime objectives::
- Remove redundant entries from the schema
- Place technology specific entries under the appropriate section

In addition, the scrubbing of the data types for NMEA2000 CAN items: src (J1939 assigned node ID number), and productID (21 bit manufacture assigned number for each product) were reviewed to change from String type to Number type.

Additional details / conversation may be viewed here:   Issue https://github.com/SignalK/specification/issues/226

Two files are edited:
  - /specification/schemas/groups/sources.json
  - /specification/test/data/sources.json

Specifics of proposed change may be seen in this pull-request: https://github.com/SignalK/specification/pull/266

# Drawbacks
[drawbacks]: #drawbacks

There is already 3rd party development underway using the existing schema for Source.  Changing would require updating of any existing applications.  

# Alternatives
[alternatives]: #alternatives

An alternative would be to leave the existing schema in place, eliminating any existing application impact.  However doing so would leave in place the  fundamental disconnect of duplication entries, as well as unclarity on how new data source technologies should address N2K specific entries at the general level.

# Unresolved questions
[unresolved]: #unresolved-questions

Attempting to modify the .src from a string to a number caused validation errors across many portions of the SignalK specification.  Submitter is not clear the reason for this.  If it is possible to correct these errors it would be  appropriate to change the data type for .src to a number type - to align with the  underlying source of the data.

thomasonw, 2016-09-30:
@timmathews Please also add the RFC flag to this, I am unable to.

timmathews, 2016-10-06:
Al, I appreciate your enthusiasm for the RFC process, but I think the PR is probably sufficient for this. We can restrict the RFCs to larger structural changes and features/expansions.

closed 2016-10-06 by timmathews

thomasonw, 2016-10-07:
@timmathews  Hey, no problems - happy to follow what ever process is mapped out (though will note I submitted this this RFC at the prompting of tkurki  [https://github.com/SignalK/specification/issues/226#issuecomment-250465200](url)

So from here - is there anything else you want me to do?
And what about the outstanding pull request #266