SignalK/specification #271: Source handling in subscriptions

tkurki, 2016-10-13:
The current subscription protocol predates multiple sources handling.

There is no way to subscribe to the data from a particular source.

The current spec does not specify how multiple sources when the subscription is throttled (`minPeriod`) or periodic (`period`). Should all values be sent or just the latest one?

rob42, 2017-03-11:
Lets just define it to clarify. A subscription to a key is for the simple value of the key, eg for a subscription to `navigation.speedThroughWater` we expect to get `navigation.speedThroughWater.value`

If we want the `values` we need to recover the `values` object by REST or `get` request, and subscribe to `navigation.speedThroughWater.values.n2kFromFile.43.value`

I'll update docs (soon as I find them) :-)

rob42, 2017-03-11:
Commited a change to gitbooks https://github.com/SignalK/specification/commit/9f101b6dbbc9d3507943bb660267d56f61d584c9 but it needs publishing I guess, as I cant see it yet

tkurki, 2017-03-11:
Good idea - I didn't think of this solution.

However I find your documentation addition a bit confusing, partly because the property name under `values` is dot-delimited. 

Let's continue in #347