SignalK/specification #275: positionFixTimestamp is extremely important for racing apps

sailracerdotnet, 2016-10-19:
Problem: to calculate COG and SOC (other than provided by GPS) using different averaging methods in for flat or for choppy conditions it is a MUST to have proper time stamp of a coordinate fix. It is also a MUST calculating the time of crossing start/finish line or taking a mark or gates on backend. The updates/timestamp is not an option because it is local time of SignalK server and might contain errors in system tyme setup.

Possible solution - to add UTC time stamp with miliseconds  (provided by gps receiver) to:
option 1) updates/values/*[path="navigation.position"]/value {latitude, longitude, **timestamp**}
option 2) in updates/values add another node  *[path="**navigation.positionFixTimestamp**"]/value="2016-01-01T00:00:00**.000**Z"

tkurki, 2016-10-25:
>The updates/timestamp is not an option because it is local time of SignalK server 
The delta schema doesn't specify this to be the local time of Signal K server. Furthermore it is specified as UTC, so it is definitely not _local_. Is this mentioned someplace else in the documentation?

When you brought the issue up in Slack I mentioned that the current timestamp is probably not well enough defined for your purpose. That we can fix I think, by stating that

```The timestamp is the best available information about the time of origin of the related value.````

For example in your case of position the data may be coming from NMEA 0183 GGA message that carries the UTC time of the fix. This should be used as the timestamp when converting to Signal K. For example [nmea0183-signalk](https://github.com/SignalK/signalk-parser-nmea0183/blob/master/codecs/GGA.js) already does this, using system time to fill in date part and using GGA data for time.

I am not sure about N2K data - is there a source for accurate time of a position fix?

Anything downstream in the SK processing chain should not meddle with the timestamp if there is one in the input SK it is processing.

That still leaves the millisecond part of your requirement. I suggest we explicitly allow it, by specifying a regular expression for timestamp with millisecond precision at https://github.com/SignalK/specification/blob/90f7d9b08c7add281dab9b74055780e4a3ef3bb2/schemas/definitions.json#L8-L13 and add that to the spec documentation.

What do you think?

bkp7, 2017-01-24:
This seems like a good solution to me. The delta currently only specifies (optionally) timestamp at the high level, ie. there is not a timestamp for each 'path', 'value' pair. I can't see a reason why this should not be added.

When we define the regEx we also need to consider the other extreme where we have no time at all. ie. do we want to make timestamp required but have a value in it which means 'now', or should it be an optional key, but if present complies with the regEx.

How can the expected accuracy of a timestamp be passed? Given there can be many data sources on a Signal K network you may want to do calculations using time on data from various sources. GPS time from receiver within a device should be easily capable of sub microsecond accuracy. Conceivably a device with GPS could act as an NTP server which could give all devices < 100ms accuracy which might not be good enough for all applications. Similarly a RTC might degrade it's accuracy over time. Since the timestamp is a string one option might be to let the number of decimals indicate accuracy? But what happens if accuracy is less than 1 second?

What formats of time are to be permitted? The definitions.json schemea currently just says ISO-8601 (UTC) date and time. Subject to the accuracy question I would suggest allowing YYYY-MM-DDThh:mm:ss.sssZ OR YYYY-MM-DDThh:mm:ssZ.

Finally, need to consider time zones. See #322

tkurki, 2017-01-24:
>The delta currently only specifies (optionally) timestamp at the high level, ie. there is not a timestamp for each 'path', 'value' pair.

You can achieve timestamp per path value pair by creating deltas with just one path-value pair per update:
 ```
{
  "context": "vessels.urn:mrn:signalk:uuid:c0d79334-4e25-4245-8892-54e8ccc8021d",
  "updates": [{
    "source": {
      "label": "n2kFromFile",
      "type": "NMEA2000",
      "pgn": 128259,
      "src": "160"
    },
    "timestamp": "2014-08-15T19:04:41.842Z",
    "values": [{
      "path": "navigation.speedOverGround",
      "value": 1.1112
    }]
  }, {
    "source": {
      "label": "n2kFromFile",
      "type": "NMEA2000",
      "pgn": 129291,
      "src": "170"
    },
    "timestamp": "2014-08-15T19:04:41.846Z",
    "values": [{
      "path": "environment.current",
      "values": [{
        "path": "navigation.speedOverGround",
        "value": 1.1123
      }]
    }]
  }]
}

Slightly more redundant, but totally doable.

tkurki, 2017-01-24:
Rereading the original issue:

>local time of SignalK server and might contain errors in system tyme setup

This is not true, see my comment above. 

As far as I can see the current setup allows milllisecond accuracy if it is available from the original source. Feel free to reopen with a concrete example if you feel that is not true.

closed 2017-01-24 by tkurki

bkp7, 2017-01-24:
@tkurki Sorry to open this up again...

Your point well made about a timestamp for each one if necessary. Agreed.

What about a RegEx for timestamp as you suggested Oct 25. Seems like a good idea to me to define it in the Schema. If you agree I will do that and also write up the example you have given with accurate timings into the documentation.

tkurki, 2017-01-24:
We already have `"format": "date-time"`, which is defined in http://json-schema.org/latest/json-schema-validation.html#rfc.section.7.3.1 as https://tools.ietf.org/html/rfc3339#section-5.6 and I believe we have working validation against that.

My comment above predates `date-time` use.

I don't think RexEx would provide much if any added value.

bkp7, 2017-01-24:
The timestamp definition in definitions.json is currently simply a string. Adding "format": "date-time" produces a whole raft of failures in the tests.

You obviously expected this to already be in place so I will add the format (rather than a regex) and update all the test files to have correct datetimes.

bkp7, 2017-01-25:
Issued pull request #324 for this.

tkurki, 2017-01-25:
Thanks for clearing this up - I did mix up datetime validation for values and timestamp.