SignalK/specification #277: Schema files versioning

tkurki, 2016-10-19:
We have now reasonable versioning scheme for Signal K specification documentation (a major part of the Specification, in addition to the schema files).

However we are currently not versioning the schema in any effective way: the [published schema files](https://github.com/SignalK/specification/blob/1fdb0e1761f1f4b13c5ab2d3042c3047e5cb5873/schemas/signalk.json#L4) state
`"id": "https://signalk.github.io/specification/schemas/signalk.json#"``
which provides no way to specify a version.

We should be able to publish explicit versions and keep also development versions available.

See also http://snowplowanalytics.com/blog/2014/05/13/introducing-schemaver-for-semantic-versioning-of-schemas/

tkurki, 2016-10-19:
This came up with https://github.com/SignalK/specification/pull/276 when the published schema files were (a) different from local files (b) not the ones in `master`.

bkp7, 2017-01-22:
Getting the versioning right is vital, and it needs to be documented exactly what it means so each implementation can correctly deal with compatibility even against versions coming into existence after the product was released.

The example at snowplow you point to is well thought out. However, as things currently stand the SignalK schemas are very open. ie there is not much restricting of properties. As such if it is left like that it will be most unusual to see 'ADDITION' increments on this project as things change going forwards.

It is really vital that the changes immediately following release of v1 can be managed well as this will be the make or break of the project.

Finally, a decision need to be made as to the format of the version number. Should it be v1.0.1 or just 1.0.1 for example. Currently the Version numbers in the examples are 'varied'.

tkurki, 2017-04-16:
Todo:
- [ ] publish the schema under signalk.org/version/schema (currently http://signalk.org/specification/schemas/signalk.json)
- [ ] publish the docs signalk.org/version/docs/ (currently http://signalk.org/specification/master/ and versions in line with npm published artifacts like http://signalk.org/specification/0.0.1-1/)
- [ ] publish the npm package with versioning in line with the doc

rob42, 2017-04-16:
FWIW the npm `0.0.1-1-alpha` style is also valid maven structure. The suffix is allowed to have a numeric start,  which is a build iteration increment, and normally maven will just choose the latest (largest number) when you ask for a `0.0.1`.  It will sort these numerically, but if you use `0.0.1-alpha-1` the sorting is not numeric anymore ( it becomes alpha-numeric on the suffix) and it becomes unpredictable.

tkurki, 2017-06-13:
@SignalK/core 
- I just used `schema:publish`  publish schemas to http://signalk.org/specification/0.9.0-SNAPSHOT/schemas/, see for example http://signalk.org/specification/0.9.0-SNAPSHOT/schemas/signalk.json
- `docs:publish` created docs at http://signalk.org/specification/0.9.0-SNAPSHOT/doc/

This contradicts what’s in the schemas themselves, they have https://signalk.org/specification/schemas/0.9.0-SNAPSHOT/signalk.json#

I think `version/schemas|doc` is more natural & futureproof (we may switch to something better than json schemas in a future version). Or did you @rob42 have something in mind that I've missed?

So what remains to be done is 
- [ ] convert the ids in the json files and js code to reflect this
- [ ] create a shorthand version of the "master" doc  that redirects to the version from master at  http://signalk.org/specification/master/doc/
- [ ] write versioning instructions
- [ ] add git tagging the version when publishing?

rob42, 2017-06-13:
'schemas/0.9.0-SNAPSHOT' is just brain fade on my part. I agree it should be consistently 'version/schemas'
I will do a pull request to fix it.

rob42, 2017-06-18:
See #364

tkurki, 2017-10-30:
The schemas have been available under https://signalk.org/specification/VERSION prefix for quite a while. 

Run `npm run schema:publish` to publish the version specific schema.

There is one glitch: https hosting the schema files at Github Pages throws the SSL warning in browser, because the cert is Github's, not ours. I think this is a separate issue and we can tackle it if somebody gripes about it. Possible solution for posterity: https://hackernoon.com/set-up-ssl-on-github-pages-with-custom-domains-for-free-a576bdf51bc

closed 2017-10-30 by tkurki