SignalK/specification #279: SK Delta: value age

pod909, 2016-10-27:
In some instances a signal k delta may be generated by a low power node that is not aware of the universal time. The timestamp should be applied to the value by the first node that it passes through that is aware of UTC. Due to the nature of the measurement of real world values in the marine environment and transmission delays a value may have some age by the time it is time stamped. A failure to recognize this decreases the accuracy with which calculations can be performed.

To allow accurate time stamping of delta values it is proposed to add an optional ISO standard date/time 'age' attribute to the delta format for each value. This is applied by source nodes that do not have an RTC to reflect the age of the value at transmission. It may be added to by intermediate node that transfer the value to reflect any filter or transmission delay.

To timestamp a value a node with RTC awareness must apply a timestamp of UTC minus "age".

Example:
{
	"updates": [{
		"source": {
			"$source": "airmar.st300.184399022"
		},
		"values": [{
			"path": "navigation.speedThroughWater",
			"value": 6.1234,
			"age":"PT1.25S"
		}]
	}]
}

toqduj, 2016-10-27:
Since the "timestamp" field itself was stated to be ISO8601 compliant, it already (theoretically) supports both an absolute timestamp as well as a "duration" (age in this case). 
I would suggest that when the timestamp field contains a duration-type time (added by non-time-aware sensors), that this is replaced by the first available node by (current absolute timestamp - duration). That way we don't need to add fields, and stay within the standard specification. 

If you do want to add a field, I wouldn't call it "age", as it can be interpreted as a running value since recording. Instead, perhaps "delay" might be somewhat more accurate?

pod909, 2016-11-05:
Based on having the timestamp in the stream hello there is no need for this. The client can work out a time delta from the timestamp in the hello message and the local time it received it and apply that to any subsequent delta messages.

Suggest this issue is closed

toqduj, 2016-11-06:
@pod909 The problem isn't the discrepancy in time between SK sender / sensor and the interpreting SK receiver. It's when the SK sender is sending a time-averaged value which was (most) true 2.5 seconds before it was sent (i.e. from a paddle-wheel sensor averaging pulses over the preceding 5 seconds). If the SK sender does not keep track of an absolute timestamp (for example, when it doesn't have an RTC itself, and sleeps in between sending), the delay information between hello and receipt is no longer true.

pod909, 2016-11-06:
I've suggested that eh issue is closed as it's addressed by having a
timestamp in the hello message

closed 2016-11-07 by pod909

timmathews, 2016-11-07:
Low power (i.e. sleeping between measurements) and realtime seem fundamentally at odds with one another. I think that @pod909 is correct that as long as the sensor gets the system time at startup (and perhaps periodically thereafter), it should be able to send out accurate time stamps.

-Tim

toqduj, 2016-11-08:
Seems complicated to do timekeeping on the sensor when it can be avoided by allowing duration in the "timestamp"-field. And sleeping of microcontrollers can be done quite efficiently between processing steps to save energy. It's not sleeping for half an hour, it's sleeping for a second. Anyway, if it's not needed now, it can always be considered later.