SignalK/specification #282: /self should be required

tkurki, 2016-10-31:
/self should provide a well known path to retrieve the server's self identity. 

Under /vessels/self the id is in a variable path mmsi, uuid or url and not in a standard format, as mmsi is the actual number, not the id.

sumps, 2016-10-31:
Should we provide some guidance to equipment developers as to which ID should be given highest priority in the situation where more than one ID type is available i.e. MMSI, URL and then UUID ?

jboynes, 2016-12-15:
The typical way with HTTP like the REST API uses would be to have the server issue a 301 redirect to the resource for the vessel

You can see the `curl` command doing that in this trace:
```
$ curl -vL http://localhost:8080/signalk/v1/api/vessels/self
*   Trying ::1...
* Connected to localhost (::1) port 8080 (#0)
> GET /signalk/v1/api/vessels/self HTTP/1.1
> Host: localhost:8080
> User-Agent: curl/7.43.0
> Accept: */*
> 
< HTTP/1.1 301 Moved Permanently
< Date: Thu, 15 Dec 2016 04:48:37 GMT
< Content-length: 0
< Location: /signalk/v1/api/vessels/234567890
< 
* Connection #0 to host localhost left intact
* Issue another request to this URL: 'http://localhost:8080/signalk/v1/api/vessels/234567890'
* Found bundle for host localhost: 0x7fa640c10040
* Re-using existing connection! (#0) with host localhost
* Connected to localhost (::1) port 8080 (#0)
> GET /signalk/v1/api/vessels/234567890 HTTP/1.1
> Host: localhost:8080
> User-Agent: curl/7.43.0
> Accept: */*
> 
< HTTP/1.1 200 OK
< Date: Thu, 15 Dec 2016 04:48:37 GMT
< Transfer-encoding: chunked
< 
{...}
```

tkurki, 2016-12-15:
`/self` is meant as a well defined path to retrieve the self identity, not for accessing the self's data.

jboynes, 2016-12-15:
Except that a vessel has multiple identities in SK (mmsi, url, uuid) as well other ways to uniquely identify it such as IMO number, flag & documentation number, even port & berth or just position.

It seems to me the main reason for providing this resource in the API is to allow clients to use the API easily - it makes a simple entry point that allows a client to easily determine "the API resource for the vessel associated with this server" which is a very common use-case.

sumps, 2016-12-15:
We had a few developers ask for this so that a client that only used RestAPI could easily get the Self SK Identity of the vessel. Until we added this, they had to call http://<ip_addr>:port/signalk/v1/api/vessels/self and then extract this info from the large amount of data that was returned.

tkurki, 2017-10-14:
`/self` should provide the identity that the server is using in outgoing self deltas. This must be the same `self` that is provided in websocket `hello` message by the server.

commit d024228b (2017-10-19): feature: make /self required

Fixes #282.
PR #385 (2017-10-19): Feature: make /self required
Fixes #282.
commit 7673e754 (2017-10-19): feature: make /self required

Fixes #282.
closed 2017-10-19 by tkurki