SignalK/specification #285: Trouble understanding the delta update schema

ph1l, 2016-11-05:
So `updates` is an array of objects and it is required to contain the property 'values'. but this next thing is structured weird:

```
        "not": {
          "allOf": [{
            "required": ["source"]
          }, {
            "required": ["$source"]
          }]
        },
```
Does that mean that you can't have both, but either or none is fine?

in context: https://github.com/SignalK/specification/blob/master/schemas/delta.json#L23-L29

tkurki, 2016-11-07:
>Does that mean that you can't have both, but either or none is fine?

Yes. The idea is that 
- source is optional - for example a SK producing sensor talking to a SK server can send deltas without source, since it is sort of irrelevant to the sender - it only knows about itself and the server should fill in relevant information about the sensor to the deltas, especially if the server is multiplexing data from multiple sources
- source is either the `source` object with relevant properties or the `$source` property, which is a pointer to possibly more data under /sources and a unique id at the same time, encoding 

See also https://github.com/SignalK/specification/pull/238

especially 

https://github.com/SignalK/specification/pull/238#issuecomment-237996702

Hmm, I wonder where in the documentation this should be included?

Did that answer your question?

ph1l, 2016-11-07:
That answers the main question.

Should I open another issue for the ALSO part? Basically, I'm confused as to why delta.json spec contains all the other stuff (subscribe, unsubscribe, get, list, and put). Aren't those messages that I, as a client, would send to the server?

tkurki, 2016-11-08:
_subscribe, unsubscribe, get, list, and put_ are for replicating HTTP api functionality over ws. 

They should be separate as they are not directly related to deltas, see https://github.com/SignalK/specification/pull/239.

Ping @rob42

ph1l, 2016-11-08:
that's what I assumed, thanks!
-- 
Sent from my Blackphone with K-9 Mail. Please excuse my brevity.

closed 2016-11-08 by tkurki

ph1l, 2016-11-08:
should the issue be closed if the specification is not clarified?

other attempting to develop clients against the spec may have problems as well.

best,

@ph1l
-- 
Sent from my Blackphone with K-9 Mail. Please excuse my brevity.

On November 7, 2016 10:31:09 PM PST, Teppo Kurki <notifications@github.com> wrote:
>Closed #285.

tkurki, 2016-11-08:
Do you think there is something not covered in #239?

ph1l, 2016-11-08:
sorry not able to check, is #239 still open pending updates to the
docs?  I'm just trying to say maybe they could be improved. Your call.
just trying to help out...