SignalK/specification #287: Sending sk-delta over a serial or UDP connection

pod909, 2016-11-07:
What rules to we need to put in place for sending sk-delta over a serial connection?

Explicit message terminator of \r\n has been defined for TCP: carry over to serial?
Check sum required: CRC-32 proposed
How is a hello requested?

jboynes, 2016-11-07:
I suggest moving UDP support to a different issue as it's quite a different beast (serial is layer 2 in the OSI model vs. layer 4 for UDP). You can run IP (and hence TCP or UDP) over serial e.g. using PPP  which would mean we could reuse all the other parts of the spec (not that I'm advocating for that).

What type of serial connection are you envisioning? The [subscription protocol](http://signalk.org/specification/master/subscription_protocol.html) requires a duplex connection for the receiver to indicate which deltas it wants to receive. That would be fine on a point-to-point bi-directional connection (RS232) but not for a multi-drop model like '0183 uses.

I can see a few use-cases which have different issues:
- connection between a device and a SK server e.g. using a micro-controller's UART
- broadcast from a SK server to multiple listeners using EIA-422 (i.e. like '183)
- create a network of SK nodes using CAN (i.e. like NMEA2000)

sumps, 2016-11-08:
I think keep this simple.

The goal is to allow the development and use of simple sensors, with no Ethernet capability, to send Signal K format Delta messages to a server.

There is no subscription, the sensor simply powers up and starts sending out the Delta messages for the data that it can measure. I would recommend that manufacturers use a differential output and an opto-isolated input but at the end of the day all we are doing is replicating NMEA0183 but in an open data format, where there is the flexibility to support and add more data types.

pod909, 2016-11-08:
Agree with that.

Point to point communication over any serial connection (UART, RS232, RS422). Like UDP, CAN is a different beast.

2 use cases: TX only, asynchronous Tx/Rx.

Proposal 1: Each object MUST be followed by a CRC-32 checksum followed by a \r\n
Proposal 2: Transmission SHOULD be 115k BAUD, 8-bit, 1 stop bit, no parity
Proposal 3: hello message MUST be sent when ever the Tx has been unable to maintain a consistent time base e,g, on power up or wake up
Proposal 4: for Rx/Tx devices when ever a device receives a hello it MUST send a hello at the next opportunity (once it ass finished sending the currently buffered objects)
Proposal 5: white space MUST be eliminated

jboynes, 2016-11-08:
Agree on scenario. Comments on proposals:
- How is the CRC encoded? 8 hex characters?
- 4) sounds like you could end up with ping-pong hello messages; is this necessary given 3)?

Also specify the character encoding to use - suggest UTF-8. The CRC should be of the bytes sent after encoding.

sumps, 2016-11-08:
There is even less need for a time sync on serial comms, as data cannot be held up and is instantly available.

I would suggest that the checksum used is the same as that for NMEA0183 as many marine developers have this code and we are trying to make the transistion from NMEA0183 to SK as painless as possible and even allow them to coexist. 

I have never heard of the NMEA0183 checksum not being good enough and it is well documented around the internet.

pod909, 2016-11-08:
Proposal 6: The characters of the signal-delta message MUST be UFT-8 encoded
Proposal 7: The CRC-32 checksum MUST be calculated lowest significant bit first
Proposal 8: The CRC-32 check sum MUST be sent as a little endian 4 byte binary value (this will -- I think?! -- allow the checksum to be evaluated with out the need to recalculate the checksum on the incoming message)

description of CRC32 https://en.wikipedia.org/wiki/Computation_of_cyclic_redundancy_checks
lib for CRC32 http://www.stillhq.com/gpg/source-modified-1.0.3/zlib/crc32.html

pod909, 2016-11-08:
Proposal 1 option 2: Each signal-delta object MUST be followed by a singel byte sequential XOR checksum of the object followed by a \r\n terminator

@sumps: still the issue of value "age" though, timestamp isn't a MUST though so can be ignored if wished

sumps, 2016-11-09:
OK this sounds good. So in summary, Signal K Deltas over Serial should be...

- Two wire Differential (RS422) talker to listener type connection (or talker to multiple listeners)
- 115K (Default) baud rate, 8 data, 1 stop bit, no parity
- Standard UTF-8 encoded
- Same Delta Format as WS but TimeStamp and Source are optional
- Each Signal K Delta message to be followed by a single byte sequential checksum after the }
- terminated with \r\n

Is that everything ? Can we agree on that ?

pod909, 2016-11-09:
Hello and hello reply? Happy to make this 'if a timestamp is sent MUST send a hello when ever...'

Time stamp and source aren't mandatory over TCP or WS either.

jboynes, 2016-11-09:
I think @sumps is a proposing uni-directional talker to listener connection, but the 'hello' reply @pod909 is proposing requires a bidirectional connection. The former mirrors a datagram-based model rather than the stream based model of TCP or WS.

I don't think we need to define the message contents here but can leave that to the stream and datagram topics. We should simply define the layer-2 wrapping for the messages.

@sumps with a UTF-8 encoding, I don't think we can use a single binary byte for the checksum as it may not be a valid UTF-8 value. For example, a checksum with bit-7 set could be confused with a continuation byte. Alternative:
- encode the checksum as JSON _int_ calculated over the bytes from the first '{' on the line to the '}' immediately preceding the checksum
```json
{}2␍␊
{"hello":"world"}123␍␊
```
where 123 is the XOR over 17 bytes. It would be an error if the checksum was not immediately preceded by a '}'

A line-based parser can then easily separate the JSON text from the checksum, validate it and hand the buffered text off to the JSON parser.

A purely event based parser would be more complicated, but it would have to be anyway to handle undoing events if the checksum indicated the data was invalid.

jboynes, 2016-11-09:
Proposal: no more than 512 bytes may be sent between \r\n line breaks

Limits the amount of buffer space needed by the receiver.

pod909, 2016-11-09:
No reason not to support both. For serial connections that need time stamp
synchronisation there needs to be a way for an Rx to request a new hello. 2
examples being where the original hello was not delivered successfully and
where a stream is joined after the tx starts transmitting.

It's the same as MQTT. If you know the real time or you don't send a time
stamp then mono directional connection is all good and needn't be written
out of the spec.

sumps, 2016-11-09:
Sorry the single byte was a mistake, not use to being down in the bits and bytes !!

I wanted us to use the same two byte ASCII hex type checksum used in NMEA0183 that everyone is use to and many people have code to generate and parse.

jboynes, 2016-11-10:
For a line-based parser using hex or decimal for the checksum wouldn't matter and hex is probably more familiar. My thought on using decimal was that it would be valid JSON which might make an event-based parser a bit easier as the token stream would just be "object int object int ..."

pod909, 2016-11-21:
The code for either is virtually identical. 

How would an event driven parser recover from noise @jboynes

pod909, 2016-11-25:
The output of this issue has been included in RFC 0005

closed 2016-11-25 by pod909