SignalK/specification #288: Device in sk-delta hello

pod909, 2016-11-07:
The signalk-delta message contains a section that allows a node to publish the source of a value. This can lead to a verbose message with the same source detail broadcast as part of each value update.

To reduce the overhead it's proposed that:

1/ A "device" property should be added to the 'hello' message.

2/ A list should be maintained in the SignalK repository to allow manufacturers to tell the community how their devices will identify them selves.

3/ Manufacturers should be encouraged to have their devices identify them selves using a sting in the format manufacturername-productname-uuid e.g. airmar-st300-12345678

tkurki, 2016-11-07:
@pod909 are you talking about the connection between a SK producing sensor and SK server or between SK server and SK client?

sumps, 2016-11-07:
Is this really necessary? Surely in order to get the Hello message the server would need to know the IP address of the sensor to establish a websocket. 

If the sensor is intelligent enough to have a wired or wireless ethernet connection, then it needs to support mDNS/Bonjour, otherwise how else is it to be configured to connect to the server or are you expecting the server to initiate discovery and connection to the sensor ?

jboynes, 2016-11-08:
I would expect a sensor to connect to a SK server rather than the other way around. That would make the sensor simpler requiring less hardware (and power) to support. It would discover the SK server in the same way a UI client would (e.g. mDNS), connect, identify itself and start sending data.

The identity of the sensor could simply be a manufacturer provided id (e.g. serial number), or could be more robust (e.g. pre-installed PKI client certificate, possibly tamper proof). The SK server can map that source to a node in the model. For example, the "acme-voltmeter1-12345" could send out basic DC voltage values, the configuration in the SK server (with its support for client UIs) would map that to the house or engine battery voltage node at the same time it granted write access.

pod909, 2016-11-08:
@jb sensor connecting to a server is a much harder use case.

Sensor would have to:
connect to look for servers on a regular basis
ask each server if it wants what it has
remember which servers want what it has
connect in turn to each and send data

That's a lot more complicated than:
Announce its existence
Accept a connection
Send data until the connection closes

toqduj, 2016-11-08:
Not sure it's more complicated or not. At the moment, my GPS sensor connects to the OpenPlotter SK server in the way @jboynes suggests, and sends its position using SK deltas over MQTT (though it uses a hardcoded IP address for now until I've got mDNS running on it, and the deltas are not interpreted by OpenPlotter yet). 

It's not asking the server if the server wants what it has (nor should it, that's not the business of the sensor), it just sends the information. 

As a bonus, now it only connects over wifi when there is a GPS fix, saving energy otherwise. 
Code here: https://bitbucket.org/toQDuj/signalk-gps-nodule

tkurki, 2016-11-08:
Why would we want to mandate who initiates the connection? Why would it matter?

tkurki, 2016-11-08:
@toqduj Would you care to open a new issue about MQTT sensor connections? And add the way you use it as a starting point.

 I think it is high time to have a documented way to use MQTT in this context.

toqduj, 2016-11-08:
@tkurki will open a new issue this evening.

pod909, 2016-11-08:
Any device sending (TCP server) and receiving (TCP client) signalk-delta.
Let's say a sensor (server) and a Signal K Node Server (client).

So far the only method defined for transmitting Signal K is for a client to
pull it from a server. The same thing would work for a connection where the
sensor pushed a value at a server but it needs to be though about a bit
more in terms of discovery of the server, the client contacting the server
to find out if it actually wants the data it provides etc. A look at the
MQTT protocol is a good start to get thinking about these questions. I
think it's the subject for another ticket though.

If the server sleeps when it's not connected then the hello message will
let the client synchronize the time stamps of the 2 systems with out any
message synchronisation issues or intelligence in the sensor.

pod909, 2016-11-08:
OK but you have implemented MQTT which handles all the "do you want
this/subscription" messaging

pod909, 2016-11-08:
@sumps So far the only thing published over mDNS is the existence of a Signal K interface, no info on what properties may be available. Can mDNS support more detail?

tkurki, 2016-11-08:
To be exact this is DNS-SD. Yes it can: https://www.ietf.org/rfc/rfc6763.txt

pod909, 2016-11-08:
So...

Option 1: sensor tells an aggregator what it is in a consistent enough manner that the aggregator cam lock up its capabilities in a catalogue

Option 2: sensor lists the properties it provides in the hello message.

Option 3: sensor only responds to mDNS requests that name a property it provides

3 wouldn't be consistent across TCP and serial
2 is verbose
1 has maintenance issues

jboynes, 2016-11-13:
Option 3) and 1) seem problematic. I took a swag at option 2) and came up with a 'hello' message like this:

```json
{
  "version": "x.x",
  "self": "sources[\"acme.THM-1.123456\"]",
  "sources": {
    "acme.THM-1.123456": {
      "serialNumber": "123456",
      "manufacturerName": "acme",
      "productName": "THM-1 Thermometer",
      "keys": {
        "sources[\"acme.THM-1.123456\"].data.temperature": {
          "units": "K"
        },
        "sources[\"acme.THM-1.123456\"].data.humidity": {
          "units": "ratio"
        }
      }
    }
  }
}
```

I reused the `sources` property as I think there would be substantial overlap in the metadata describing the device. We might be better off have a different group for actual source devices and keep `sources` focused on the connection it was received on.

An update message would look like:
```json
{
  "context": "sources[\"acme.THM-1.123456\"].data",
  "updates": [
    {
      "values":[
        {
          "path":"temperature",
          "value":292.15
        },
        {
          "path":"humidity",
          "value":0.39
        }
      ]
    }
  ]
}

```

This allows a device to specify the keys it is emitting. In the example here, the keys are defined by the device itself and don't need to be configured for its application; for example an ACME thermometer would always emit its temperature under the same key and would not need to be configured based on where it was installed. The aggregator would use the id in `self` to distinguish between devices and map them into its vessel-wide model.

A more complex device could still support configuration of the keys it emitted. For example, a multimeter that had a UI to select between measures could include the setting of the 'house' vs. 'engine' label on a voltage and emit the `vessel.self.electrical.batteries.house1.voltage` key.

It is verbose though :(

pod909, 2016-11-13:
Yep, been going in that direction too but...

If the source is now a device or piece of software (agree with that) then
context is something different.

The context for the value and the sensor is the vessel (or some other
physical or logical grouping.

I'm also beginning to think that the value JSON needs to be the same for
streaming values as communicating them + That there's no difference between
an updates array and just sending the updates individually as an update to
a single attribute.

Agree on the hello but possibly the keys need to be defined in "sends" and
"receives" arrays.

I have the signal-delta JSON looking something like....

{
"path":"environment.temperature",
"value":1.2345,
"timestamp":"00/00/00T00:10:15",
"$source":"acme.THM-1.123456"
}

... with the path excluded for MQTT

or the more complex...

{
"path":"environment.temperature",
"value":1.2345,
"timestamp":"00/00/00T00:10:15",
"$source":"acme.THM-1.123456",
"values":
{
"1":
{
"value":1.2890,
"timestamp":"00/00/00T00:10:10",
"$source":"acme.THM-1.123456"
},
"2":
{
"value":1.3890,
"timestamp":"00/00/00T00:10:05",
"$source":"acme.THM-1.123456"
},
"3":
{
"value":1.9890,
"timestamp":"00/00/00T00:10:00",
"$source":"acme.THM-1.123456"
}
}
}

... note the path would only reside at the top level and the values are
index by importance

The full signal k would end up looking something like:

{
"vessels":
{
"mmsi:boaty:mcboatface":
{
"environment":
{
"temperature":
{
"value":1.2345,
"timestamp":"00/00/00T00:10:15",
"$source":"acme.THM-1.123456",
"values":
{
"1":
{
"value":1.2890,
"timestamp":"00/00/00T00:10:10",
"$source":"acme.THM-1.123456"
},
"2":
{
"value":1.3890,
"timestamp":"00/00/00T00:10:05",
"$source":"acme.THM-1.123456"
},
"3":
{
"value":1.9890,
"timestamp":"00/00/00T00:10:00",
"$source":"acme.THM-1.123456"
}
}
}

}
}
},
"sensors":
{
... your sensor JOSN for the hello here with an additional property
"context":"vessels.mmsi:boaty:mcboatface"
}

pod909, 2016-11-13:
The generic schema for all that looks something like...
`

{
	"definitions":
	{
		"description":"This section describes the words that have a defined meaning with in this document.",

		"attribute":
		{
			"description":"Something that may be given a value in the Signal K model and that may contain other attributes."
		},

		"context":
		{
			"description":"A physical or logical grouping of attributes, nodes or other contexts."
		},

		"node":
		{
			"description":"Any device or piece of software that sends and/or receives Signal K objects."
		},

		"streamed":
		{
			"description":"The transmission of the between NODES of a value for an individual ATTIBUTE."
		}
	},

	"sk-full":
	{
		"description":"This section describes the JSON objects that are used to store and comunicate the full Signal K object.",

		"sk-object":
		{
			"description":"The full Signal K object in which is used to communicate the ATTRIBUTES in one or more CONTEXT.",
			"type":"object",
			"properties":
			{
				"vessels":{"$ref":"#sk-full/context"}
			}
		},

		"context":
		{
			"description":"",
			"type":"object",
			"patternProperties":
			{
				"[a-zA-Z]*":
				{
					"oneOff":
					[
						{"$ref":"#sk-full/context"},
						{"$ref":"#sk-full/attribute"}
					]
				}
			}
		},

		"attribute":
		{
			"description":"",
			"type":"object",
			"patternProperties":
			{
				"[a-zA-Z]*":
				{
					"oneOff":
					[
						{"$ref":"#sk-full/attribute"},
						{"$ref":"#sk-full/value"}
					]
				}
			}
		},

		"value":
		{
			"type":"object",
			"properties":
			{
				"$ref":"#sk-core-definition/value",
				"$source":{"$ref":"#common/path"},
				"not":{"allOff":["source","$source"]}
			}
		}
	},

	"sk-streamed":
	{
		"description":"This section defines the JSON objects that are used when Signal K values are STREAMED.",
		"sk-delta":
		{
			"description":"sk-streamed MUST 1: Updates MUST be STREADMED as sk-delta objects. Each update MUST contain 1 object.",
			"type":"object",
			"properties":
			{
				"path":{"$ref":"#sk-core-definition/path"},
				"$ref":"#sk-core-definition/value"
			}
		}
	},

	"sk-core-definition":
	{
		"description":"This section defines JSON objects that are used else where in this document. That are not used in their own right as part of Signal K.",
		"value":
		{
			"type":"object",
			"properties":
			{
				"anyOff":
				[
					{"value": {}},
					{"values":
					{
						"type":"object",
						"patternProperties":
						{
							"[0-9]*":{"$ref":"#sk-core-definition/value"}
						}
					}}
				],
				"timestamp":{"$ref":"#sk-core-definition/timestamp"},
				"source":{"$ref":"#sk-core-definition/source"},
			}
		},

	}
}`

pod909, 2016-11-25:
Now covered in RFC 0005

closed 2016-11-25 by pod909