SignalK/specification #298: Replace N2K src with CAN NAME

jboynes, 2016-11-29:
Something has been bugging me with the way we handle N2K sources based on the 'src' address field. I'm not privy to the spec so may be off base here, but digging into the N2K data stream from my boat I see devices, primarily my Raymarine autopilot, claim different addresses over time.

I see references to PGN 060928 which [is publicly documented](http://www.nmea.org/Assets/20140710%20nmea-2000-060928%20iso%20address%20claim%20pgn%20corrigendum.pdf) as an address claim per ISO 11783-5 which associates the address with the actual NAME of the device (i.e. its 64 bit unique id).

From what I'm seeing on my network this address appears ephemeral and can vary each time the N2K CAN network boots. Now it seems to settle into a repeatable state but I think that's just the result of the devices I have connected and not any address reservation scheme so may change if I add/remove a device. NMEA hints at that with a [TC](http://www.nmea.org/Assets/20160209%20nmea%202000%20te%20200020160208%20address%20to%20name%20association%20errata.pdf) that states each device is required to build a table that "provides the means to associate a parameter group with a specific NMEA 2000 certified device even though the device’s address may change at any time."

It seems to me that our use of `src` to contain the CAN source address should be replaced with use of the NAME that has claimed that address at the time the message was sent. If someone more familiar with the requirements can verify this behaviour I will put together a PR.

tkurki, 2016-11-29:
I believe this is something that should be done implementation first, so that we don't end up with a spec that is unimplementable or with no open source implementation.

Some provision should be made for playback of n2k capture files, where no bus interrogation is possible.

pod909, 2016-11-29:
An investigation into the NAME(s) of a device connected to 2 busses might be wise.

sumps, 2016-11-29:
I'm happy with the implementation as it is now.

jboynes, 2016-11-29:
I have [a capture](https://github.com/SignalK/specification/files/619556/actisense.txt) of my boat's startup taken by actisense-serial from an Actisense NGT-1. Running it through canboat analyzer, these are the 60928 PGNs I see:
```
2016-10-08T21:04:45.078Z 6   5 255  60928 ISO Address Claim:  Unique Number = 0x9000e; Manufacturer Code = Raymarine; Device Instance Lower = 0; Device Instance Upper = 0; Device Function = 190; Device Class = Instrumentation/general systems; System Instance = 0; Industry Group = Marine
2016-10-08T21:04:45.334Z 6   1 255  60928 ISO Address Claim:  Unique Number = 0x0; Manufacturer Code = Raymarine; Device Instance Lower = 5; Device Instance Upper = 0; Device Function = 150; Device Class = Steering and Control surfaces; System Instance = 0; Industry Group = Marine
2016-10-08T21:04:46.884Z 6  35 255  60928 ISO Address Claim:  Unique Number = 0x37f33; Manufacturer Code = Airmar; Device Instance Lower = 0; Device Instance Upper = 0; Device Function = 130; Device Class = Navigation; System Instance = 0; Industry Group = Marine
2016-10-08T21:04:57.025Z 6   0 255  60928 ISO Address Claim:  Unique Number = 0x10a348; Manufacturer Code = Raymarine; Device Instance Lower = 1; Device Instance Upper = 0; Device Function = 150; Device Class = Instrumentation/general systems; System Instance = 0; Industry Group = Marine
2016-10-08T21:04:57.046Z 6   1 255  60928 ISO Address Claim:  Unique Number = 0x10a348; Manufacturer Code = Raymarine; Device Instance Lower = 1; Device Instance Upper = 0; Device Function = 150; Device Class = Instrumentation/general systems; System Instance = 0; Industry Group = Marine
2016-10-08T21:04:57.046Z 6   1 255  60928 ISO Address Claim:  Unique Number = 0x0; Manufacturer Code = Raymarine; Device Instance Lower = 5; Device Instance Upper = 0; Device Function = 150; Device Class = Steering and Control surfaces; System Instance = 0; Industry Group = Marine
2016-10-08T21:04:57.046Z 6   2 255  60928 ISO Address Claim:  Unique Number = 0x10a348; Manufacturer Code = Raymarine; Device Instance Lower = 1; Device Instance Upper = 0; Device Function = 150; Device Class = Instrumentation/general systems; System Instance = 0; Industry Group = Marine
2016-10-08T21:04:57.879Z 6   3 255  60928 ISO Address Claim:  Unique Number = 0xd467c; Manufacturer Code = Raymarine; Device Instance Lower = 1; Device Instance Upper = 0; Device Function = 150; Device Class = Instrumentation/general systems; System Instance = 0; Industry Group = Marine
2016-10-08T21:04:57.879Z 6   2 255  60928 ISO Address Claim:  Unique Number = 0x0; Manufacturer Code = Raymarine; Device Instance Lower = 1; Device Instance Upper = 0; Device Function = 150; Device Class = Instrumentation/general systems; System Instance = 0; Industry Group = Marine
2016-10-08T21:04:57.879Z 6   3 255  60928 ISO Address Claim:  Unique Number = 0x10a348; Manufacturer Code = Raymarine; Device Instance Lower = 1; Device Instance Upper = 0; Device Function = 150; Device Class = Instrumentation/general systems; System Instance = 0; Industry Group = Marine
2016-10-08T21:04:57.879Z 6   3 255  60928 ISO Address Claim:  Unique Number = 0xd467c; Manufacturer Code = Raymarine; Device Instance Lower = 1; Device Instance Upper = 0; Device Function = 150; Device Class = Instrumentation/general systems; System Instance = 0; Industry Group = Marine
2016-10-08T21:04:57.879Z 6   4 255  60928 ISO Address Claim:  Unique Number = 0x10a348; Manufacturer Code = Raymarine; Device Instance Lower = 1; Device Instance Upper = 0; Device Function = 150; Device Class = Instrumentation/general systems; System Instance = 0; Industry Group = Marine
2016-10-08T21:05:32.682Z 6 160 255  60928 ISO Address Claim:  Unique Number = 0x122873; Manufacturer Code = Raymarine; Device Instance Lower = 0; Device Instance Upper = 2; Device Function = 150; Device Class = Instrumentation/general systems; System Instance = 0; Industry Group = Marine
```
You can see the negotiation between 0x10a348/1/0, 0xd467c/1/0, 0x0/1/0 and 0x0/5/0 over addresses 1, 2, 3 and 4 with 0x10a348 having to claim new addresses as other devices join the bus.

pod909, 2016-11-29:
May need sorting out when we look to devices proper but how much of an issue is this now?

sumps, 2016-11-30:
This is normal NMEA2000 behaviour, the equivalent of DHCP on an Ethernet network.

jboynes, 2016-11-30:
So we have the same issues DHCP has where, because a node's IP address can vary with each lease, it can't be relied on to identify the device and instead a persistent identifier is needed e.g. something derived from its MAC. For CAN, this is the NAME in the address claim. 

This causes a problem for SK because the address appears in the path for references e.g. `/sources/N2K-1/{src}/...` making existing references to the source invalid when the address changes.

tkurki, 2016-11-30:
So far this has not been a huge problem, as there are few clients that rely on source information, even less on /sources, but operate on paths. 

However this needs to be handled before long.

As I've pointed out this above there is little use in just speccing this, but we need an implementation as well.

sumps, 2016-11-30:
I think more is being made of this issue than is necessary. A well designed server or client should not rely on a source always being present and should be timing out and reviewing sources as necessary.

timmathews, 2016-12-01:
It's not about assuming a device is present or not, it's about data continuity. And knowing that the device with a `src` of 3 is the **same** device which just a moment ago had a `src` of 10.

Instrument Panel (IP) [currently uses](https://github.com/SignalK/specification/blob/master/index.js#L223) the `src` value in the deltas that it consumes to identify a stream of data. So, if a device's `src` changes, the next delta received by IP will be interpreted as coming from a new device. A new gauge display will be created on the screen and the old one will stop updating.

I suppose you could argue that this is a deficiency in IP, but it's also a limitation of the delta format as it stands today. If we used the `NAME` value to identify the source of the data, this wouldn't be a problem.

canboat, 2016-12-02:
I agree, our N2K implementations should use the supposedly unique N2K device ID, which on N2K is the best we can do.

pod909, 2016-12-03:
If we make this change can we also only allow sources to be identified by transport where there is no device id made available?

sumps, 2016-12-03:
This "name" is a combination of various fields and may or may not be unique depending upon the manufacturer's implementation and programming each device with a unique serial number during production. Also if the installer changes the instance the name changes.

@jboynes have you actually managed to see a device change its CAN address on your network, from one power cycle to the next. Devices should only change addresses if a new device is added to the network that has a higher priority.

jboynes, 2016-12-05:
@sumps By "NAME" I specifically meant the term as defined by ISO 11783-5:2001 section 4.3.1, which is a 64 bit entity, and which "network integrators and ECU manufacturers shall ensure that each ECU on a particular network has a unique NAME not possessed by another ECU on that network.

Per the (public) [TE above](http://www.nmea.org/Assets/20160209%20nmea%202000%20te%20200020160208%20address%20to%20name%20association%20errata.pdf), address changes are allowed by NMEA:
> In practice, a table may be used to store associations between specific functions and specific addresses. Once a global request for ISO Address Claim parameter group (NMEA 2000 PGN 060928) is detected, a NMEA 2000 certified device starts to build its own table during the address claim process. The table provides the means to associate a parameter group with a specific NMEA 2000 certified device even though **_the device’s address may change at any time_**. 

As the dump above shows, the `0x10a348` device changes its address even when power is still connected. There are devices on my network that seem to remember their address each time they are powered on (the MFD and AIS for example), but there are some like that one that don't.

@canboat I don't think the 21 bit unique id is unique enough, it would need to be the full 64 bits the bus protocol uses. Even then, NAME need only be unique on a single bus so it is possible to have devices on different buses with the same NAME. That may be more likely for ones like I have that both have the unique id 0x0 but differ in the device instance part of the name which I guess will be set by an installer to a common value like 1, 2, etc.

sumps, 2016-12-05:
@jboynes, I know what you mean by "NAME", plus I know how it is built up and why it is used, as I have the full NMEA2000 specification, not just an errata.

What you are talking about is a significant change to Signal K and should not be taken on the basis of one installation and an errata. You have already recognised in your Raymarine installation that some devices have the same "unique" id and different device instances that can be changed at any time by an installer.

The reality is that on most installations after the system is powered up and this address claim process is complete that all of the devices end up with the same address as they had on the previous power cycle.  I have not heard of any iKommunicate/Signal K installations reporting an issue with the current method.

Using a 64bit binary name in the URL will not make for very human readable instrument panel labels and it will be meaningless to the end user and not as easy as it is now to quickly match a SK source to a real or virtual NMEA2000 network device.

I am not saying that the solution we have now is perfect, and developers should not "assume" that all N2K sources will always have the same CAN address, but we have no reported problems and perhaps there is a smarter way to do this than just so from CAN address to CAN Name.

One last point, using CAN address currently provides an automatic way to assign priroity to multiple sources of the same data. Lower the CAN address = Higher the Priority.

tkurki, 2016-12-07:
>sources to be identified by transport where there is no device id made available?

The current schema allows `$source` in delta that is a `sourceRef`: in practice it is just a string, albeit with a bit of guidance on how it should be constructed

https://github.com/SignalK/specification/blob/master/schemas/definitions.json#L58-L63

SourceRef is a reference to a source under /sources, with `.` as delimiter. The pattern is `[A-Za-z0-9-_.]*`.

This came up with discussion on non-nmea sources: https://github.com/SignalK/signalk-server-node/issues/101

See also https://github.com/SignalK/specification/blob/90f7d9b08c7add281dab9b74055780e4a3ef3bb2/test/sources.js#L37-L51 for non-nmea.

Note that you can put anything you want in sourceRef in delta and once https://github.com/SignalK/specification/pull/301 is merged the Node server will produce that hierarchy under `/sources`. There is no functionality to the values there though, just empty hierarchy with no real values. I think at least a timestamp would serve debugging purposes, other metadata to be added when need arises.

tkurki, 2016-12-07:
I don't think the changes implied in this thread need break backwards compatibility, but this can be sorted out once we come up with the solution that allows persistent source identities.

pod909, 2016-12-07:
I've done a lot of work on incorporating the current source usage into
streaming. Looks like it may be pretty straight forward

sbender9, 2016-12-07:
@sumps I have definitely seen devices end up with a different address between power cycles.

sumps, 2016-12-07:
@sbender9 was the network in exactly the same state with the same devices all powered up ?

Using the CAN address has some key benefits, using a 64bit binary value (that may or may not be unique and may or may not change) is less appealing and I am going to need a lot more convincing to back any change to the current method. The reality is that a client has all of the information it needs in the Source object to create its own human friendly persistent name.

I would push this back to the Consumer (Client) to come up with a better solution using the available data, than Instrument Panel currently uses.

sbender9, 2016-12-07:
@sumps all the devices may not have been on. That would explain it. 

This is not an issue for me as far as displaying data, i don't rely on the $source at all.

tkurki, 2017-03-04:
https://github.com/canboat/canboat/blob/master/util/list-product-information

sbender9, 2017-03-04:
I definitely have a real world case now that shows that the N2K device id's can change any time a system is powered on. And can probably change any time  any device is powered on in the network.

sbender9, 2017-03-04:
I'm not sure if this is a spec issues, but we definitely need a way to provide source information to client that does not change.

tkurki, 2017-03-04:
I consider this a spec level issue as it concerns all N2K-SK gateways and how sources are represented under `/sources` and in `$source`.

sbender9, 2017-03-04:
To add some more detail, I have a user that has three Airmar depth sounders in his system. We are seeing that every time his system is powered on they can end up with different N2K device id's. I assume they all fight over the same initial id when they power on instead of using the last now good id.

tkurki, 2017-03-09:
I decided to bite the bullet and figure out how can name could be used with real life data. 

https://github.com/SignalK/signalk-server-node/pull/162 has a version of server that

- keeps an in memory map from src's to data received for this source in PGN
60928
- uses this data to add `manufacturerCode` and `uniqueNumber` fields to
each delta update's source object
- if there is no 60928 data a 59904 PGN is sent to request the data
- includes a capture file that contains relevant 60928 data

Deltas for which there is no can name data can be optionally ignored. That will make most capture log file playback fail miserably - it might be useful to create some pseudo can name for that purpose.

Furthermore the delta source object => $source path reference mapping is updated to use `manufacturerCode` and `uniqueNumber`. This allows attaching metadata to the permanent paths under `/sources`. For a client to be able to retrieve the metadata for an incoming delta's source the mapping from can name to source path must be standardised as part of the specification.

See [the example delta and the resulting full json](https://gist.github.com/tkurki/d82e34275e17093704ad8dddc5d5da92).

All this now operates on canboat-generated values, meaning that uniqueNumber is a decimal value (currently a string in JSON) and manufacturerCode is the value converted to a shortish string according to [mapping in canboat](https://github.com/canboat/canboat/blob/e154d2b1d8260dec69abae50ca324b93f58d4177/analyzer/pgn.h#L4619-L4727).

@tim presented the idea that the [manufactureCode and uniqueNumber] could be combined to create a single 32 bit unique identifier. Since there are two distinct fields I think it would be a good idea to bring that structure to the JSON and not treat it as a single field. We could use it to our advantage in creating per manufacturer subtrees under source (unlike the current implementation here, which just creates a flat mapping).

All in all I think this is perfectly doable and a really important piece of the puzzle. 

Furthermore I think the logic related to 60928/59904 should be a part of the N2K-SignalK mapping function (gateway/server). Nothing further down the processing chain will have the capability to query the bus for data about unknown sources.

Comments, please!

tkurki, 2022-03-20:
SK server now supports optionally using can name in the source data.

closed 2022-03-20 by tkurki