SignalK/specification #304: Key hierarchy causes confusion

jboynes, 2016-12-09:
Observing the discussions on electrical, propulsion and now environment, trying to determine an appropriate hierarchy for key names seems fraught and arbitrary. In the recent discussion on temperatures, there are names derived from the N2K spec source type, but also a generic catch-all `environment.temperature.<sourceID>` (which also needs to include an `<instanceID>`)

What value do we get from trying to figure out this hierarchy? Especially when combined with the idea of a manifest as proposed in [RFC0001](https://github.com/SignalK/specification/issues/264).

We could dispense with the hierarchy entirely, and instead:
- include the devices, with source and instance ids in the manifest
- label the device/source/instance with a human-readable name (e.g. "Sea Temp" or "Aft Cabin Temp")
- derive a unique but opaque id for the value e.g. from the device/source/instance components
- use that as the key name in the vessel's context

A display client would define its panel in terms of the opaque ids, can get the display label from the manifest and the value from the vessel context by id.

For example, on a Mareton TMP100 device, channel 2 may be measuring temperature in the fore cabin and channel 3 in the aft cabin. These two sensors may be given ids `tmp100-1234-2` and `tmp100-1234-3` respectively. In the manifest they would be defined as instances 1 and 2 of the `Inside Air` source type and be given the 'Fore' and 'Aft' labels. When a PGN was received, the server would map (tmp100, Inside Air 1) to `tmp100-1234-2` using the manifest and update the `tmp100-1234-2` in the `vessel.{vesselId}` context. A display client using the REST interface would see a reference to `tmp100-1234-2` in the display panel configuration, retrieve the label from the manifest entry for that key in order to display the "Fore Cabin" label and subscribe to the `vessel.{vesselId}.tmp100-1234-2` path to receive the updates for the temperature value.

By flattening the data we avoid all the problems of trying to figure out the right place in the hierarchy for values and make the model extensible for additional data. We need all the plumbing anyway to deal with undefined entries like `environment.temperature.<sourceID>` and to assign human-readable labels to instance ids.

tkurki, 2016-12-09:
I agree that the hierarchy causes trouble. It is essentially arbitrary grouping for humans. It plays a role in REST api, as you can retrieve a subtree's values by path.

However I think your reasoning is missing one important aspect, that I mentioned in #206: the schema establishes shared semantic meanings to the values. It is not just about that a value is a temperature, but it is a certain temperature that has a meaning.

Let's use engine exhaust gas temperature as an example and imagine that we could use Maretron TMP100 to measure it. Now let's assume that I made a really intelligent engine diagnostics app, able to say something definitive about your engine conditions just by monitoring your exhaust gas temperature.  I just happen to like my labels in Finnish, so on my vessel that temp is labeled _"Pakokaasun lämpötila"_. The app will not be able to automatically get the data if all the information available is opaque ids and human-oriented arbitrary labels.

One way to think about this aspect of the schema as **Well Defined Names**.

I believe this is an extremely important aspect of the Signal K schema. Cabin temperatures are an example to the contrary, but if we apply your reasoning to for example speeds how would things look? We would have just speeds with some labels: how do you tell that speed A is wind speed, speed B STW and speed C lateral speed through water? And how can my fancy performance algorithm figure out that it should use speed C? Or that the livewell temperature logger should use that particular temperature?

For some future Signal K version I would like to do away with the hierarchy and just model the different data attributes and associations and non-hierarchical Well Defined Names.

jboynes, 2016-12-10:
In general, it is not the name that conveys the semantic meaning but attributes associated with the value. My proposal is that we add those attributes to the metadata as **Well Known Annotations** for the value so the systems can use them; this makes the name arbitrary.

Anything where there can be multiple instances of a value means the name is no longer unique. For example, on a catamaran with a transducer on each hull you will have two values for STW and if the vessel is turning neither of them corresponds to effective boat STW. A vessel with two GPS sensors presents two positions, and again neither may correspond to the position of the vessel. Depth depends on which transducer is reporting it, and which instance is which is important.

The ability to add multiple annotations allows multiple facets to be added simultaneously: that a temperature is for exhaust, for gas, is associated with engine id 1, which in turn is attributed as being the port engine (or better still, is attributed with its location in the hull and an association with the port drive). Metadata describes speed B as being aligned 3 degrees to centerline, speed C as being aligned at 87 degrees, allowing your fancy performance algorithm to correct for the installation offsets; both are as annotated as being speed of water flow over hull. 

Basically what values **are** matters, what they are **called** does not.

Now a consequence of names not mettering is that the names we currently use (per [keysWithMetadata](https://github.com/SignalK/specification/blob/master/keyswithmetadata.json)) don't need to change. We just need to stop trying to encode semantics _that implementations can rely on_, such as a context hierarchy, into the name.

If we plan to flatten the hierarchy in some future version, why not do it now when less people will be impacted?

tkurki, 2016-12-10:
I think we agree on this, even if your reply sounds like you disagree? I don't care if it is called **Well Known Annotation** or **Well Defined Name**, I think we are talking about the same thing.

I am all for multidimensional metadata/annotations - you have probably seen my comments about not trying to encode everything into the path and not thinking about the data model as a single in memory hierarchy.

I don't think the existence of multiple instances/sources invalidates the Well Known Annotation in any way. That's the way the world is and then as you described additional annotations are needed to make sense of the data, maybe prompt the user to pick a strategy for dealing with the situation (average / prefer a certain source / fallback to others in abcense of the prefered source etc).

What would in your view be the next step in this direction, keeping backwards compatibility?

PS. What do you want to actually call the Well Known Annotation that tells what the value is? Because there could be multiple Well Known Annotations, like the port/starboard/other engine annotation. At least to me _annotation_ is a really generic word - but then again I am not a native english speaker.

sumps, 2016-12-10:
We have a system at the moment that works with minimal human input and that is a very important point and was one of the original key goals that we successfully achieved.

What I do not see in all of these discussions is how this new hierarchy is automatically populated. Annotations, manifests, human readable labels, etc. How are these automatically setup with no human intervention ? If we are expecting Signal K users to spend hours setting all of this stuff up manually then take up is going to be very low.

jboynes, 2016-12-12:
@tkurki I agree we have the same objective. I agree with your comments about the path and not thinking an in-memory hierarchy. The hierarchy is currently still present though, in the way data is modeled by the JSON schema and exposed over the REST API. More, applications are required to understand the hierarchy because the position of a value in the hierarchy (its path) conveys meaning that allows the application (and a human) to tell one value from another.

This presents problems as the "correct" position can be ambiguous (e.g. is livewell temp an environment property or a tank property), needs to be defined up front (e.g. there is no spot in the current hierarchy for "aft cabin temperature") which limits extensibility, or does not achieve the purpose (e.g. use of an "instance" identifier provides no meaning to machines or humans to let them distinguish between instances).

An analogy: if I tell you my pet is "Fido" it does not tell you much about the pet. If I tell you it's a black labrador dog, with a white star on its nose, male, age 8, named "Fido" who sits on the command "Sit" and stays on command "Stay" you get a much better picture - you know to feed it dog food, you can make it sit and stay, you can tell it from another dog. "Fido" becomes an opaque term used only to identify this particular pet. The difference is I told you things about the pet rather than just what it is called.

Based on that I think next steps would be:
* Define an initial set of well known attributes that can be stored in each value's metadata. We already have `unit`, `displayName`, `shortName` and `longName`; suggestions could include `type` (e.g. air temperature, exhaust temperature, distance, length), `authority` to distinguish between duplicate instances (e.g. with 3 GPS sources, which one to use for the location of the vessel); `location` indicating where (physically) the value came from.
* Flatten the schemas to define certain well known names and the type of value they have, based on the names in keyswithMetadata. This would change the schemas to remove the hierarchy of properties without changing the types of the leaf values.
* Expand the REST API to allow search by attribute e.g. find all values whose `unit` is `K`; find all values whose `type` is a temperature.

For backward compatibility, existing clients of the REST API would continue to work. New servers would support the new search operations and additional metadata. New clients would be able to use the additional data to provide more flexible UIs when working with the new server.

I can prototype this, but it will take a couple of weeks (I have a day job).

sumps, 2016-12-12:
What if you could understand "Fido's" bark and from this knew that he was hungry, was aged 8, had a master called Jeremy and was in his basket in the left hand bedroom of the house, would this change the way you approached the flattening of the schema or would you try and optimise it for getting the best performance from "Fido" ? 

A prototype would be a useful next step to see if my conerns are unfounded.

pod909, 2016-12-12:
Not sure it's true that the present model is a hierarchy. Sure it's presented as one by the REST interface but in reality there are between 3 and 6 different independent concepts in the model. Because they are independent their existence in the path is basically arbitrary.

We're dealing with a 2-3 part common noun and a whole bunch of pronouns. The path is just a sentence in which they can be presented in any order.

5 pippin apples in the hold of the good ship Dionysius.
In the hold of the good ship Dionysius are 5 apples of the pippin variety.
On the Dionysius in the hold there are apples of the pippin variety counting 5.
...
or even as we have now...
On the good ship Dionysius there are apples in the hold and of the pinnins there are 5

...

Names are nouns and can appear anywhere in a text. Annotations are a separate note or further description presented separately from the main text.

On^1^ the Dionysius^2^ are 5 apples^3^

1. These apples happened to be in the main hold, a 200sqm room with barely sufficient ventilation to keep the apples from decay.
2. The Dionysius was a good ship with a happy crew of 5.
3. The apples where of the pippin variety. A soft sweet chalky desert apple with a pale body and rose colored crown.

pod909, 2016-12-12:
For my 2c:
- The sentence should identify the unique thing that is measured with out needing to reference an annotation.
- There needs to be a consistent noun ordering in the keys, which there isn't entirely at present.
- The nouns that refer to each other should be together (that horse has bolted I know).

There's more than 1 air temperature in relation to a vessel, so we need to identify them *in the path*.

The challenge in terms of having a fixed lexicon is that it is never going to be exact enough.

pod909, 2016-12-12:
Is livewell temp an environment property or a tank property?

Livewell and tank are pronouns. environment and temp are common. You can put them in any order but for the clearest description put the pronouns and common nouns with each other.

Livewell tank environment temperature: 12 degrees C

pod909, 2016-12-12:
We just need to come up with a noun order:
Vessel identity + general common noun + location + specific common noun(s)

And come up with a dictionary of:
general + specific common noun pairs/triplets
common locations

There's no real need to enumerate out all the possible combinations

sumps, 2016-12-13:
This Issue was raised because of a perception that the current Schema causes confusion. Having spent a lot of time designing a product around the schema, I am not of that opinion and feel that the schema has been designed with a lot of intelligent discussion and consideration, between experienced boaters, developers and product managers.

It may not have the perfect hierarchy, but then we are not working in a perfect world or from a blank sheet of paper, so compromises had to be made. However, the schema does have inherient value and knowledge and it works well in practice, so any proposed change will have to do more than just tidy up the grammar, it needs to bring measurable benefits and the best way to illustrate that is with a prototype that everyone can understand and positively critique.

I am not against change (except if it impacts on V1.0.0) that is what versioning is for, but it has to move us forward not just tidy things up to make them theoretically better.

My final point will make use of the LiveWell Temperature example to try and show the logic behind the current schema that sometimes gets forgotten (usually by me) and results in these discussions. 

What is a Livewell ? Unless you regularly go fishing, you can be forgiven for not knowing and so it would have been easy to stick this in a "catachall" section of the schema with metadata that explains it is the temperature of a Livewell. However, a quick search on Wikipedia and we can learn that...

https://en.wikipedia.org/wiki/Livewell

...a Livewell is a tank used to keep fish and bait alive, so @timmathews is 100% correct in placing this in the Tank group, where there might also be a water level reading for the Livewell and on larger boats there will probably be more than one "tank instance" as well.  

So the current schema supports the intelligent grouping of common data and the key question when deciding how a piece of data is stored in the current schema, is does the data relate to an existing group ? If it does like Engine Pressure (Propulsion), Livewell Temperature (Tanks), Time+Date (Navigation), etc. then place it within that group, if it does not, place it in a generic group like environment and have metadata that explains what the data is. 

This approach to the schema, results in a "marine intelligent" schema that properly groups common data together that makes perfect sense to a boat owner and might help a software developer who has only limited sailing experience to understand the data a bit more.

pod909, 2016-12-13:
Signal k hit he limit of what can be done with a hierarchy with sources and timestamps.

Tim is talking about cross hierarchy relationships. Teppo is talking about moving to a decoupled multi dimension schema (i'm guessing as a result of looking at time based values?).

A lot of transports require the dimensions to be encoded into a single key/path/index. In that case the dimensions better be in the same place or it will get painful to consume. 

The current 'hierachy' is already in effect a multi dimensional key as it identifies things to be measured and adds various different kinds of context. It's just that the dimensions aren't yet   well defined and placed consistently in the key.

For instance it's proposed to move livewell from environment to tanks (good idea) but 'fridge' is left behind in environment.

There is water.level, water.salinity, water.temperature but a property of waterTemperature  (a none divisible entity) is suggested.

rob42, 2016-12-22:
We need to keep focus on simplicity too. While the current hierarchy is not perfect it is quite useable, certainly its pushed the state-of-the-art in marine protocols a looong way. Ive seen a lot of technology which was very powerful for its day (at Sun, and in Java) that was ignored or mis-understood as it was too far ahead of the typical devs and users.

We should complete and stabilize V1 in its current form, and start using it widely to gain more experience. From that we will learn what is actually needed, and what is just 'premature optimization'

No doubt we will find problems and mistakes we have made, but by then we will know where and how to focus our changes.

webmasterkai, 2017-02-24:
@jboynes You might be interested in exploring http://project-haystack.org/doc for inspiration. It's an open-source project where they tackle the challenge of semantic modeling with tagging to make data self-describing, thereby streamlining data interchange among software/devices.