SignalK/specification #305: EXPERIMENTAL RFC 0007 : No Server Signal K

pod909, 2016-12-09:
EXPERIMENATAL RFC 0007: Serverless Signal K

Authors: Ric Morris
Status: EXPERIMENTAL
Version: 0.1
Date: 09/12/2016

This document contains ideas and content first suggested by Teppo Kurki, Paul Sumpner, Sailoog, Toqduj, Jim Boynes, Rob Huitema, Brian P and others.

### Summary

This RFC explores the potential for a subset of Signal K that is capable of being delivered using off the shelf software and cloud services.

This RFC is EXPERIMENTAL and contains some features that may not be compatible with or extend the capabilities of the current Signal K standard. As such this RFC is not intended for inclusion in the Signal K standard AS IS.

### Motivation

Through the discussion on transports it has become clear that there are of the shelf solutions that can provide an MQTT, WS and REST interface and that these solutions may be capable of acting as a broker for Signal K messages delivering something close to full Signal K requirment.

Since there are still instances when messaging needs to be carried over serial, UDP or plane TCP a consistent aproach for these carriers has been considered.

Using an off the shelf broker has obvious attractions in terms of scalability and the availability of services in the cloud.

### Resulting Architecture

The monolythic nature of the Signal K Server hinders the distribution and load balancing of functions amongst devices on a vessel or in the cloud.

By limiting the role of the broker to communication of Signal K messaging using an off the shelf broker or services marine specific services would then hang off the broker to perform specific tasks.

They might include:
- n2k > SK gatway feeding messages from an NMEA2000 network
- A filter servcie taking "latest" values and outputting messages with min, max and mean values
- A navigaton service taking in "latest" values and sending out "target" values for speed and heading
- A "master" service that takes in messages and provides definitive values for the direct attributes of "vessels.self"

### Introduction

### Example Messages

Hello with no context

```json
{"$key":"consumers.my-sk-server-123","type":"any","version":"v1"}
```
Hello with a context
```json
{"$key":"vessels.98493.consumers.my-browserbased-instrument-123","type":"any",
    "version":"v1","timestamp":"12/3/2019T12:34:56Z"}
```

Providing a single value from a network node that does not know its context.

Over WS/TCP
```json
{"$key":"source.achme-thing-1234","type":"signalk-atomic",
    "provides":["navigation.CourseOverGround.latest"]}
14.5
14.7
14.3
```
Over UDP with a source
```json
{"$key":"source.achme-thing-1234","type":"signalk-atomic",
    "provides":["navigation.CourseOverGround.latest"]}}
{"$source":"achme-thing-1234","latest":14.5}
{"$source":"achme-thing-1234","latest":14.7}
{"$source":"achme-thing-1234","latest":14.3}
```
Over TCP or UDP with no source
```json
[... 226 lines cut ...]

APPLICATION: The following conditions apply when messages are sent over UDP.

Any messages received before the sorce SHOULD be ignored. (SHOULD 7.4.1)

Each message MUST be sent in a single UDP packet. (MUST 7.4.2)

IF a source object has been sent message MUST contain a "$source" property that references the source OTHERWISE the "key$" property MUST identify the context (MUST 7.4.3)

#### 7.5 REST

APPLICATION: The following conditions apply when messages are sent over a REST interface.

REST services MUST be made available via URIs with the following format:
A "http://hostname/signalk/vX/apt/" where X is the version of Signal K
B Followed by a partial key including at least "vessels" and a UUID
(MUST 7.5.1)

Messages MUST be requested from a REST service using a GET. The REST service MUST reply with a branch for the partial path in the URI containing all the known values for the full paths the partial path is part of.

Messages MUST be provided to a REST service using a POST.

Messages SHOULD be provided to a REST service as individual values.

... something about context/type and other standard http headers here ...

rob42, 2016-12-22:
I agree the signalk server should not be seen as a monolithic central server. Signal k is always intended to be peer-to-peer, although the way we have built our server implementations thats not obvious.

pod909, 2016-12-29:
Contexts and attributes in the current Signal K standard:

### contexts
registration/national/countryid*
registration/local/countryid*
registration/other/countryid*
environment/outside
environment/inside/engineRoom
environment/inside/mainCabin
environment/inside/refrigerator
environment/inside/freezer
environment/inside/heating
environment/water
propulsion/engine[0-9]
eletrical/batteries/battery[0-9]
eletrical/inverters/inverter[0-9]
eletrical/chargers/charger[0-9]
eletrical/ac/circuit[0-9]
tanks/freshwater[0-9]
tanks/wastewater[0-9]
tanks/blackwater[0-9]
tanks/fuelwater[0-9]
tanks/fuel[0-9]
tanks/lubrication[0-9]
tanks/livewell[0-9]
tanks/baitwell
sails/sail[0-9]
sensors[0-9]

### attributes
url
mmsi
name
flag
port
country
registration
description
communicaiton/callsignVhf
communicaiton/callsignHf
communicaiton/phoneNumber
communicaiton/emailHf
communicaiton/email
communicaiton/satPhoneNumber
communicaiton/skipperName
communicaiton/crewNames
imo
temperature
dewPointTemperature
apparentWindChillTemperature
theoreticalWindChillTemperature
heatindexTemperature
pressure
humidity
water/salinity
depth/belowKeel
depth/belowTransducer
depth/belowSurface
depth/surfaceToTransducer
current
tide/heightHigh
tide/heightLow
tide/timeLow
tide/timeHigh
heave
wind/angleApparent
wind/angleTrueGround
wind/angleTrueWater
wind/directionChangeAlarm