SignalK/specification #307: Values are inconsistent between REST API and Delta messages

jboynes, 2016-12-15:
Working on #304, testing with the node server, I noticed the way the value of a key is sent is different when fetched from the REST API or received in a delta.

For example, a GET to `.../navigation/headingTrue` returns
```json
{
  "value": 0,
  "$source": "nmeaFromFile.II",
  "timestamp": "2016-12-15T03:29:32.000Z",
  "sentence": "HDT"
}
```
but updates to that key are sent as:
```json
{
  "context": "vessels.urn:mrn:signalk:uuid:c0d79334-4e25-4245-8892-54e8ccc8021d",
  "updates": [
    {
      "source": {
        "type": "NMEA0183",
        "sentence": "HDT",
        "label": "nmeaFromFile",
        "talker": "II"
      },
      "timestamp": "2016-12-15T03:20:35.000Z",
      "values": [
        {
          "path": "navigation.headingTrue",
          "value": 0
        }
      ]
    }
  ]
}
```

The difference here is REST call returns an `object` but the `value` property in the delta message is a scalar. The REST result matches the `numberValue` type for headingTrue as [defined by the schema](https://github.com/SignalK/specification/blob/master/schemas/groups/navigation.json#L389), the delta message does not.

However, when the key has an object value, like `navigation.position` then both the REST API and the delta message contain the same object:

```json
{
  "longitude": 23.53145,
  "latitude": 60.0782,
  "$source": "nmeaFromFile.GP",
  "timestamp": "2016-12-15T10:00:37.000Z",
  "sentence": "GLL"
}

{
  "updates": [
    {
      "source": {
        "type": "NMEA0183",
        "sentence": "GLL",
        "label": "nmeaFromFile",
        "talker": "GP"
      },
      "timestamp": "2016-12-15T09:56:07.000Z",
      "values": [
        {
          "path": "navigation.position",
          "value": {
            "longitude": 23.538766666666668,
            "latitude": 60.08436666666667
          }
        }
      ]
    }
  ],
  "context": "vessels.urn:mrn:signalk:uuid:c0d79334-4e25-4245-8892-54e8ccc8021d"
}
```

Also, a `numberValue` like `headingTrue` can contain multiple `values` so the value for this key may at some time be representable as a scalar but for other instances may be a object containing multiple values. The client receiving the update must now handle these special cases.

This can be addressed by specifying that the `value` property in the delta message is always an object of the type as defined by the schema for that key. This would change the `headingTrue` delta message to:
```json
{
  "context": "vessels.urn:mrn:signalk:uuid:c0d79334-4e25-4245-8892-54e8ccc8021d",
  "updates": [
    {
      "source": {
        "type": "NMEA0183",
        "sentence": "HDT",
        "label": "nmeaFromFile",
        "talker": "II"
      },
      "timestamp": "2016-12-15T03:20:35.000Z",
      "values": [
        {
          "path": "navigation.headingTrue",
          "value": { "value": 0 }
        }
      ]
    }
  ]
}
```
where the `updates/values/value` property contains a `numberValue`.

tkurki, 2016-12-15:
The reason for the `value` property in the full model is that it allows
- metadata for the path (`timestamp`, `meta`, `_attr`)
- multiple values handling with `values``

Delta is not meant to convey any of this, only the updated value with minimal metadata about the updated value of the property, so the `value` in delta is not the same `value` as the one in full.

I would fix the inconsistency going in the opposite direction: use `value` always, also in object valued properties, in the full model.

rob42, 2017-03-11:
Closing as its stale, and is covered to some extent by #315 
Pls reopen if still needed

closed 2017-03-11 by rob42