SignalK/specification #310: Use cases for controls

joabakk, 2016-12-20:
Controls from Signal K has been discussed before, and post V1, we could start populating this group.

I'm picturing a `controls` group or something along those lines. At this stage we should be able to set up the structure. I know many have other use cases for controls. I have listed some below:

#293 RFC 0004: Entertainment draft RFC
#203 Signal K to control stereo etc?
#187 Proposal for "Circuits" object for switchable electrical objects

When looking into proprietary protocols such as seatalk and FDX, it is possible to control the light intensity of displays, among other things. It should then be fairly simple to calculate the time of sunset and set the light intensity without fiddling with the individual displays.

If you comment with use cases to this issue, I can propose a structure.

sumps, 2016-12-20:
Conventional Autopilot in "Nav" Mode is an obvious one, drone/ROV (?), electrical variable/dimmable objects, colour changing lights, robotic arm (?), stepper motor, angular movement, engine speed, the list will become quite large.

Not sure a controls group is the right method.

joabakk, 2016-12-20:
Valid point. But without some structure it will be even harder. Autopilot should already be covered in the steering group, but the others are homeless.

sumps, 2016-12-20:
The steering group is sort of strange as it has some interesting keys but in reality you do not really have any current use cases that would actually use them. It appears to be designed with a DIY autopilot system in mind, controlled by Signal K.

The rudder reference angle is useful and populated from NMEA data but everything else is sort of redundant until someone designs an open source Signal K autopilot or steering system.

What I had in mind was the sort of "Nav" mode Autopilot control that has been around for years with the APB and RMB type NMEA0183 sentences and the equivalent NMEA2000 PGNs. As soon as you start thinking about how Signal K could control this simple process it raises a very important issue.

As soon as you start to control something and transmit data in to a system, you have the possibility of "data loops" unless you have a very well thought through control mechanism. For instance, if a Signal K Server sent an APB message to a Raymarine autopilot, that same data will circulate around the Raymarine network on SeaTalk or SeaTalkNG *(NMEA2000) and then might get spat out with the instrument data that is going back in to the Signal K Server.

Suddenly there is duplicated data with no paper trail to tell the server that the bearing to waypoint data it is receiving is the same data it sent out.

This is just one example of what can happen once you start controlling things and I think it is important to get the control mechanism right first, thinking about action, effect and feedback and ensure that the Signal K control mechanism is reliable, safe and fast before getting too bogged down in how to structure the schema.

joabakk, 2016-12-20:
The steering group was populated based on what was in the open domain of values and modes from the major AP manufacturers, and it is used in WilhelmSK. Listen for change in a target or mode, and transmit that to the AP system. 
But I see your point. I did not plan to populate this before new year anyway, just thought a common issue for control would be in place to gather use cases.

closed 2020-03-08 by joabakk