SignalK/specification #311: Non-leaf updates

jboynes, 2016-12-22:
What is the expected behaviour for an update for a non-leaf path e.g.
```json
{
  "context": "vessels.urn:mrn:imo:mmsi:234567890",
  "updates": [{
    "values": [{
      "path": "propulsion",
      "value": {
        "0" : {
          "revolutions": { "value": 16.341667},
          "boostPressure": { "value": 45500.000000},
          "tiltTrim": { "value": 48}
        }
      }
    }]  
  }]
}
```

Does this replace the entire `propulsion` object or is the value a sparse model that is merged? If so, what are the rules for merging?

Or, do we only allow updates for leaf nodes? If so, what are those nodes? Should we modify `keyswithmetadata` to remove non-leaf paths?

rob42, 2016-12-22:
I dont think updating objects was the intent of delta? @tkurki?
If we take the above case to extremes, we would have:
```
{
  "context": "vessels.urn:mrn:imo:mmsi:234567890",
  "updates": [{
    "values": [{
      "path": "navigation",
      "value": {
        ....the whole navigation tree....
      }
    }]  
  }]
}
```
This would then be much better suited to sending the full or sparse format.
But its a good point, we should clarify the intent and rules. As a starter:

delta: `values` will be leaf nodes (edge case, allow arrays) . Complex objects should be decomposed to leaf nodes so we dont have the merge/replace problem. Arrays replace the exiting array.

full: replaces the existing tree.
sparse: merge into exiting tree.

Then we need to identify the sparse/full format somehow I guess?

jboynes, 2016-12-23:
We should continue to allow structures as leaf nodes e.g. position. Otherwise we introduce synchronization problems between components especially where transports don't ensure order of delivery, or where optional parts may be lost in transit and not detected. The behaviour would be the same as for scalar values with the entire leaf node value being replaced. 

Not allowing `context` or `path` to refer to a arbitrary group makes this much more consistent and, would think, be backward compatible with existing clients if they only deal with leaf nodes. 

We should define what contexts are allowed. Suggestion:
* global
* vessel
* device

sumps, 2016-12-23:
A Delta is intended to communicate a value change in the most efficient way. Currently the most common cause of a value change is an NMEA sentence or PGN which may have just one value or multiple values.

The current Delta method works very well for this purpose and values in the Delta replace the values in the client/servers name space.

rob42, 2016-12-24:
Any delta with an object can be decomposed into a delta with a series of values. eg the sample above is equivalent to:
{
  "context": "vessels.urn:mrn:imo:mmsi:234567890",
  "updates": [{
    "values": [{
            "path": "propulsion.0.revolutions",
             "value":  16.341667
         },
        {
            "path": "propulsion.0.boostPressure",
            "value":  45500.000000
        },
        {
            "path": "propulsion.0.tiltTrim",
            "value":  48
        }]  
  }]
}

If we make this the rule, and avoid sending objects as values then we also avoid having to decide between merging or replacing the target object. eg the paths being updated are explicit and precise, if we do want to invalidate a related key we can send value:null etc.

tkurki, 2016-12-26:
Continue at #312

closed 2016-12-26 by tkurki