SignalK/specification #317: Key AIS data not supported

sumps, 2017-01-05:
Whilst testing the latest iKommunicate build, it has become apparent that the schema is still lacking some key AIS data. Currently we have no way of identifying if a vessel is a Class A or Class B AIS target and also what AIS category of vessel it is.

I would suggest that AIS Class should perhaps be stored in the Communications group and the Category of vessel stored in the Design group but I am happy to hear the views of others. It would be nice to get this pushed through in time for V1 though as I think it is an important issue for the correct display and filtering of vessels.

The AIS Class would be automatically populated based on the AIS Message type or PGN, whilst the Category of vessel would be populated from the "Ship/Cargo Type" field in the relevant PGN or NMEA Sentence.

[image]

rob42, 2017-01-06:
Sounds good. Can you make a pull request to suit?

tkurki, 2020-04-22:
AIS class implemented in #562 as sensors.ais.class.

tkurki, 2020-04-22:
AIS ship type is `design.aisShipType`, both as string and as number.

tkurki, 2020-04-22:
https://github.com/SignalK/specification/blob/3ccbd2a904698c6479289a0202311da591f04a68/schemas/groups/design.json#L39-L350

closed 2020-04-22 by tkurki